lovyan03/LovyanGFX #307: sprite.setPaletteColor(No, Color); does not work properly

ESPboy-edu, 2022-11-05:
## Environment ( 実行環境 )

- MCU or Board name: ESP8266
- Panel Driver IC: ST7735
- Bus type: I2C
- LovyanGFX version:  v0.4.17
- FrameWork version:   ArduinoESP8266 - WeMos LOLIN mini
- Build Environment :   ArduinoIDE
- Operating System:   macOS

## Problem Description ( 問題の内容 )
for(int i=0; i<16; i++) screenbuffer.setPaletteColor(i, tft.color888(palette[i*3], palette[i*3+1], palette[i*3+2]));
does not set palette.

but! if we add 3 lines (not 1 not 2, but 3) like "screenbuffer.setPaletteColor(0, 0);"
above mentioned palette in "for" loop sets fine

### Expected Behavior ( 期待される動作 )
palette set as provided

### Actual Behavior ( 実際の動作 )
seemed it sets all colors to 0xFFFF

### Steps to reproduce ( 再現のための前提条件 )
compile below code

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
#define LGFX_ESPBOY
#define LGFX_AUTODETECT
#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>  

const PROGMEM uint8_t palette[16*3] = {
  0, 0, 0, 29, 43, 83, 126, 37, 83, 0, 135, 81, 171, 82, 54, 95, 87, 79, 194, 195, 199, 255, 241, 232,
  255, 0, 77, 255, 163, 0, 255, 236, 39, 0, 228, 54, 41, 173, 255, 131, 118, 156, 255, 119, 168, 255, 204, 170
};

LGFX tft;
LGFX_Sprite screenbuffer (&tft);

void setup(void){ 
 tft.init();
 tft.setRotation(0);
 tft.fillScreen(0);
 
 screenbuffer.setColorDepth(4);
 screenbuffer.createSprite(128, 128); 
 screenbuffer.fillSprite(0);

 for(int i=0; i<16; i++) 
   screenbuffer.setPaletteColor(i, tft.color888(palette[i*3], palette[i*3+1], palette[i*3+2]));

 //screenbuffer.setPaletteColor(0, 0); // IF WE WILL UNCOMMENT THESE 3 LINES, IT STARTS TO WORK FINE
 //screenbuffer.setPaletteColor(0, 0);
 //screenbuffer.setPaletteColor(0, 0);

 for(int i=0; i<16; i++)
   screenbuffer.fillRect(i*128/16, 20, (i+1)*128/16, 100, i);

 screenbuffer.pushSprite(0, 0);
}

void loop(){
  delay(1000);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

m1cr0lab, 2022-11-05:
@ESPboy-edu, have a look at [05-game-of-life.cpp](https://github.com/m1cr0lab-espboy/ESPboy/blob/67d6394814371818f42342bb03559aa9bc76dd0b/examples/5-game-of-life/5-game-of-life.ino#L59-L71) example for ESPboy Library.  😉

ESPboy-edu, 2022-11-05:
using
```cpp
screenbuffer.setPaletteColor(i, palette[i*3], palette[i*3+1], palette[i*3+2]);
```
solved the problem

but previous approach should also work ok, but it does not

closed 2022-11-06 by ESPboy-edu