lovyan03/LovyanGFX #309: Esp32-cam + SPI ili9488 3.5" TFT 565 vs JPEG pixformat performance and ghost effect

Sentinel8000, 2022-11-08:
I have choosed bug report, but im not sure this is issue, but will be good point if somebody working with this library.

Because im not experienced developer, im working with example sketch-es, and i have worked with a toy camera example what use lovyan gfx for display. This sketch using for camera part config.pixel_format = PIXFORMAT_RGB565; settings, so if we want display we need use
the pushimage function with these parameters:

display.pushImage(0, 0, fb->width, fb->height, (lgfx::swap565_t *)fb->buf);

Performance was much better against TFT_espi library what wont support ili9488 DMA. With HVGA resolution 480x320 i get realy good FPS, i can notice sometimes the JPG picture changing, because we know currently esp32 have not enough power to drive mjpeg or any video as stream with usable FPS - so we display from esp32-cam JPG pictures realy fast speed. With QVGA speed was much better, but i have noticed a weird effect - if i put my hands between the camera and the object till few seconds was my hand transparent - maybe a problem with the picture buffer, i dont know.

After comes a idea, what if i will process JPG pictures and i change the camera settings to config.pixel_format = PIXFORMAT_JPEG;
picture was just a colored noise with pushimage, so i have searched the library code and i have find the library have a drawJpg function too however this was marked as deprecated. 

With using display.drawJpg((const uint8_t*)fb->buf, fb->len, 0, 0); i get much much more better FPS than with 565 pixformat + pushimage.

My lovyan settings looks so for SPI 3.5" TFT with Ili9488 + esp32-cam

```
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

lgfx::Panel_ILI9488     _panel_instance;
lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス

public:

  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

// SPIバスの設定
      cfg.spi_host = VSPI_HOST;     // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
      cfg.spi_mode = 0;             // SPI通信モードを設定 (0 ~ 3)

      cfg.freq_write = 79500000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)4
      cfg.freq_read  = 40000000;    // 受信時のSPIクロック 16000000
      cfg.spi_3wire  = true;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock   = true;        // トランザクションロックを使用する場合はtrueを設定
      //cfg.dma_channel = SPI_DMA_CH_AUTO; // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)1
      cfg.dma_channel = SPI_DMA_CH_AUTO;
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 14;            // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 2;            // SPIのMOSIピン番号を設定
      cfg.pin_miso = -1;            // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc   = 13;            // SPIのD/Cピン番号を設定  (-1 = disable)
     // SDカードと共通のSPIバスを使う場合、MISOは省略せず必ず設定してください。

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs           =    15;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    12;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。
      
      cfg.panel_width      =   320;  // 実際に表示可能な幅 240  400x296 480
      cfg.panel_height     =   480;  // 実際に表示可能な高さ 320
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     1;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)0  555555555555555
      //cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_pixel =     1;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  true;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = false;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  false;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};
```

Question why i get lower fps with pushimage + pixformat 565 against drawjpg pixformat JPEG and what can be this ghost effect with lower resolutions?

Also i know esp32-cam have a JPEG compression quality settings, what im sure not working using the 565 pixformat.

tobozo, 2022-11-08:
```cpp
cfg.freq_write = 79500000
```
:thinking:

Sentinel8000, 2022-11-08:
Yeah funny, but with 80000000 wortked not, with this 79500000 value display is realy realy fast.

lovyan03, 2022-11-08:
The clock divider operates at 80 MHz divided by an integer.
Your clock setting is equivalent to 40 MHz.

lovyan03, 2022-11-08:
I really dislike long sentences in English.
I have no idea what you are asking.

Sentinel8000, 2022-11-08:
Ok i try short down.

lovyan03, 2022-11-08:
People say they can see ghosts, but I'm not a psychic, so I have no idea.

lovyan03, 2022-11-08:
[[deprecated("use float scale")]] bool drawJpg
`jpeg_div_t` is deprecated. you can use float scale version.

Sentinel8000, 2022-11-08:
Hehe funny:) 

So, first, what a value can i set for my speed for SPI write, read?

Im using esp32-cam + tft - esp32 cam can work with 565 pixformat, also with JPEG pixformat.

for 565 i have used pushimage function.

for JPEG drawjpg function

pushimage function slower than drawjpg why? why drawjog deprecated?

Using pushimage with QVGA, pixels updating slower, so if i put my hand before the camera, few seconds my hand is transparent, why?

lovyan03, 2022-11-08:
There are multiple versions of drawJpg.
Only some of them are deprecated.
Specifically, only those that take `jpeg_div_t` as an argument.

Sentinel8000, 2022-11-08:
Sure, i saw drawjpg still working from url and from file.

if i want drawjpg using, display image with this way, what is the syntax?

display.drawJpg((const uint8_t*)fb->buf, fb->len, 0, 0);

lovyan03, 2022-11-08:
it's ok

Sentinel8000, 2022-11-08:
There is other alternate way display JPEG format with this way if you remove this function later?

lovyan03, 2022-11-08:
???

lovyan03, 2022-11-08:
There are multiple versions of drawJpg.
Only some of them are deprecated.
Specifically, only those that take jpeg_div_t as an argument.

There are multiple versions of drawJpg.
Only some of them are deprecated.
Specifically, only those that take jpeg_div_t as an argument.

There are multiple versions of drawJpg.
Only some of them are deprecated.
Specifically, only those that take jpeg_div_t as an argument.

lovyan03, 2022-11-08:
ok?

Sentinel8000, 2022-11-08:
Sure PSRAM slow, but im working with examples, i have not try-ed out other ways. Standard esp32-cam examples working with PSRAM buffers.

Questions:

- Normal drawjpg is deprecated - two function what i have found drajpg from url, from SD - 

What is the current, actual way display JPEG from a variable if 
display.**drawJpg**((const uint8_t*)fb->buf, fb->len, 0, 0); currently deprecated ? I hope now clear what is my question.

- So you say this effect is not because library have issue.

- You said my SPI freq settings bad, if i want max speed what i need set for write, read?

lovyan03, 2022-11-08:
drawJpgは非推奨ではありません。

lovyan03, 2022-11-08:
> pushimage unable to handle JPEG format?

不可能ですね

Sentinel8000, 2022-11-08:
Ok ok, i give up, we have language problems:) 

Just answer me to one answer, you can use your langauge, what is the maximum speed settings for SPI write, read? Ok?

lovyan03, 2022-11-08:
私が言いたいのは、79MHzを設定しても40MHzで動作しているという事実だけです。
動作しているなら好きな値を設定すれば良いです。

Sentinel8000, 2022-11-08:
あなたはどこでも良いフィードバックを受け取る非常に高速なライブラリを作成しました。残念なことに、どの言語でも人々と話すのに苦労しています。 私はプログラミングのスキルがあまりありません.2つの値を尋ねました.私の設定が悪かった場合でも、あなたはまだ答えたくない...奇妙な..

lovyan03, 2022-11-08:
そもそもカメラとの連携などは私が知った事ではありません。

それでも情報が欲しいというなら、せめてIssueテンプレートに用意されている項目は書いてください。
https://github.com/lovyan03/LovyanGFX/issues/new?assignees=&labels=&template=bug-report--------.md&title=

あなたの説明には有益な情報が殆どありません。
適切な回答を得たいなら、正確で十分な情報を提供する必要があります。

Sentinel8000, 2022-11-08:
たぶん今、私はレポートコードの例に従っていません。私は熟練したプログラマーではありませんが、あなたとあなたの仕事に敬意を表しました。 最後に SPI 書き込み用と読み取り用の 2 つの値を尋ねましたが、形式が問題ではなく、どの言語でも回答に問題があります。 何か聞きたいことがあり、あなたのライブラリをテストしたので、申し訳ありません。ここで質問したり、書いたり、バグを書いたりすることはもうありません。 ありがとうございました。