lovyan03/LovyanGFX #312: Please give a compilation guide based on ESP-IDF

hero, 2022-11-20:
**Carefully written requests are more likely to be given priority.**
I tried to use this library to drive SSD1351 OLED in my ESP32-S3 project, but the compilation failed!
My compile condition：
- ESP32-S3
- ESP-IDF version: v4.4.3
- Compile ToolChains: ~/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
- Failed Log:
`
-- --------------Compile Info------------
-- IDF_PATH = ~/ESP32/Sdk/esp-idf-v4.4.3
-- IDF_TARGET = esp32s3
-- PROJECT_NAME = SPTC-TEC
-- PROJECT_DIR = ~/ESP32/Demos/SPTC-TEC-esp32-s3
-- BUILD_DIR = ~/ESP32/Demos/SPTC-TEC-esp32-s3/build
-- SDKCONFIG = ~/ESP32/Demos/SPTC-TEC-esp32-s3/build/sdkconfig
-- SDKCONFIG_DEFAULTS =
-- CONFIG_LV_CONF_SKIP = y
-- ---------------------------------------
-- CMAKE_SOURCE_DIR = ~/ESP32/Demos/SPTC-TEC-esp32-s3
-- CMAKE_BINARY_DIR = ~/ESP32/Demos/SPTC-TEC-esp32-s3/build
-- --------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: ~/ESP32/Demos/SPTC-TEC-esp32-s3/build

# Many errors like:
1） LovyanGFX/src/lgfx/v1/platforms/esp32/../../Light.hpp:22:1: error: unknown type name 'namespace'
namespace lgfx
 ^~~~~~~~~
2）error: expected '=', ',', ';', 'asm' or '__attribute__' before 'lgfx'
using namespace lgfx::jpeg_div;
                            ^~~~
3）error: unknown type name 'using'; did you mean 'ulong'?
 using namespace lgfx::colors;
 ^~~~~
4）LovyanGFX/src/lgfx/v1/platforms/esp32/../esp32/../../../utility/../internal/memory.h:13:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.
`

**Describe the solution you'd like**
Please give a compilation guide based on ESP-IDF. 

**Additional context**
I've found the C++ support here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/cplusplus.html
But it still didn't help#~_~!!

lovyan03, 2022-11-20:
This library is written in C++ and cannot be included from `*.c` source.

sukesh-ak, 2022-11-24:
@hero 

Your ESP-IDF main function should look like this and in a `.cpp` file to use `C++`.

```c++
extern "C" void app_main(void)
{

}

```

closed 2022-12-25 by github-actions[bot]