lovyan03/LovyanGFX #316: ILI9481 8-bit parallel + ESP32-S3 issue

JohnTaipei, 2022-11-22:
- MCU ESP32-S3
- Panel Driver IC:  ILI9481 
- Bus type:   Parallel8
- LovyanGFX version:  v0.4.16
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment :   ArduinoIDE
- Operating System:   Windows

Hello Sir :
my pcb esp32-s3 + ili9488 8-bit parallel
use your library and work fine

i changed lcd panel ili9481 8-bit parallel
work failed
============================

but used esp32 + ili9481 8-bit parallel
use your library and work fine
============================

Thanks

lovyan03, 2022-11-24:
v0.4.16 ?  not latest ?
Does it work with the develop branch?

JohnTaipei, 2022-11-24:
Hello Sir :
I updated to v0.4.18
esp32-s3 + ili9481 8-bit parallel still failed

Does " develop branch" means??

lovyan03, 2022-11-24:
Anyway, I do not have enough information.

What do you mean by failed?
I think you have the wrong settings.
I am not clairvoyant, so I can only know the information you have written.

https://github.com/lovyan03/LovyanGFX/issues/new?assignees=&labels=&template=bug-report--------.md&title=

If you need my help, please provide at least the minimum information.
Please fill out all the items in the template.

lovyan03, 2022-11-27:
@JohnTaipei 
An Issue with a configuration similar to this was posted by another user, but that one seems to have been resolved by reviewing the configuration.
 #319

Therefore, it is highly likely that your Issue is not a library deficiency either.

JohnTaipei, 2022-11-28:
@lovyan03 
Hello Sir:
Thanks for your answer
i try again and again
the issue is frequency too high
i forgot to down the frequency

now 8-bit,16-bit work fine @10,16,20MHz 
failed @27,40MHz

ili9481 is too old
ili9488 work fine @40MHz

Thanks again

lovyan03, 2022-11-28:
Glad your problem is resolved.

Next time, please write using the Issue template.

I don't know what your setup is and I find it painful when you raise an Issue without details.
It is very tiring to ask questions over and over again.
Please use the Issue Template next time.

JohnTaipei, 2022-11-28:
@lovyan03 
ok, i got it
i will use the Issue Template

esp32 + 8-bit  parallel supports Max frequency 20MHz
so ili9481 works fine

i have to remember to try lcd panel form low frequency first

Thanks Sir
Have a nice day

closed 2022-12-02 by lovyan03