lovyan03/LovyanGFX #318: Request adaptation to the HX8347D driver support

wdmomoxx, 2022-11-26:
The HX8347 family is also more common and requests for support，thankyou
https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_HX8347D.h
https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_HX8347D.cpp

https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/HX8357D_Init.h
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/HX8357D_Defines.h

tobozo, 2022-11-26:
hey thanks for your feedback :+1: 

I wish I could help but I don't have this display. I found a [cheap version on aliexpress](https://www.aliexpress.com/item/32606589352.html) but it has headers for raspebrry pi, plus it's 90 days shipping...

if you're willing to do some tests and get your hands dirty with C++ code I can guide you though, just let me know

otherwise you'll have to wait one until of the maintainers eventually gets a unit of that display and finds time+motivation to implement it

wdmomoxx, 2022-12-01:
@tobozo
Thank you for your reply.
My screen is 320x240 resolution, I searched and found that the screen I was using has been discontinued, there is not much information, I found it here: https://www.panelook.com/TM024HDZG99_Tianma_2.4_LCM_parameter_26659.html
Maybe it's a bit similar to this one
https://www.waveshare.net/wiki/2.8inch_Resistive_Touch_LCD#.E6.93.8D.E4.BD.9C.E6.AD.A5.E9.AA.A4

Before, I used the above driver and successfully lit up the screen.
https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_HX8347D.h
https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_HX8347D.cpp
Although it showed inversion, but then I fixed it
[image]

tobozo, 2022-12-01:
it looks like that display family is supported, it's part of the ILI948x drivers bundle.

Have you tried using `Panel_HX8357B` or `Panel_HX8357D` ?

wdmomoxx, 2022-12-02:
@tobozo I tried it and it didn't work.

tobozo, 2022-12-02:
okay, in order to create your own panel you'll have to:

- edit [this file](https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx/v1/panel/Panel_ILI948x.hpp#L278)
- duplicate the block `struct Panel_HX8357D : public Panel_ILI948x { ..... }` 
- in your new block rename `Panel_HX8357D` to `Panel_HX8347D`

you are now ready to convert the init sequence from TFT_eSPI to LGFX.

first import all defines into the code block you duplicated earlier, and rename them accordingly:

e.g 
```cpp
#define HX8357D_SETC 0xB9
```

will become:
```cpp
static constexpr uint8_t CMD_SETC = 0xB9;
```

you can then proceed to import the TFT_eSPI init block sequences.

An init block sequence is a `writeCommand` statement followed by one of more `writedata` statements. 

For example this sequence:

```cpp
    // setextc
    writecommand(HX8357D_SETC);
    writedata(0xFF);
    writedata(0x83);
    writedata(0x57);
```

will be inserted as a line in the list0 array, the value '3' is the amount of bytes that follows:

```cpp
      static constexpr uint8_t list0[] =
      {
          // ...
          CMD_SETC , 3, 0xFF, 0x83, 0x57,
          // ...
      };
```

tobozo, 2022-12-02:
I took the existing `Panel_HX8357D` and updated a couple of values, commented out `//CMD_IDMOFF  , 0,` and changed `CMD_SETPANEL, 1, 0x08,` to `CMD_SETPANEL, 1, 0x05,`, apart from that there aren't many differences so I'm not so sure you need to duplicate the Panel block to get your display working.

Here's what I got anyway, let me know if this improved, and if it didn't please be more descriptive than "it doesn't work"

```cpp
  struct Panel_HX8347D : public Panel_ILI948x
  {
  protected:

    static constexpr uint8_t CMD_TEON    = 0x35;
    static constexpr uint8_t CMD_TEARLINE= 0x44;
    static constexpr uint8_t CMD_SETOSC  = 0xB0;
    static constexpr uint8_t CMD_SETPWR1 = 0xB1;
    static constexpr uint8_t CMD_SETRGB  = 0xB3;
    static constexpr uint8_t CMD_SETCYC  = 0xB4;
    static constexpr uint8_t CMD_SETCOM  = 0xB6;
    static constexpr uint8_t CMD_SETC    = 0xB9;
    static constexpr uint8_t CMD_SETSTBA = 0xC0;
    static constexpr uint8_t CMD_SETPANEL= 0xCC;
    static constexpr uint8_t CMD_SETGAMMA= 0xE0;

    const uint8_t* getInitCommands(uint8_t listno) const override
    {
      static constexpr uint8_t list0[] =
      {
          CMD_SWRESET , CMD_INIT_DELAY, 20,
          CMD_SETC    , 3+CMD_INIT_DELAY, 0xFF, 0x83, 0x57, 255,
          CMD_SETRGB  , 4, 0x80, 0x00, 0x06, 0x06,
          CMD_SETCOM  , 1, 0x25,
          CMD_SETOSC  , 1, 0x68,
          CMD_SETPANEL, 1, 0x05,
          CMD_SETPWR1 , 6, 0x00, 0x15, 0x1C, 0x1C, 0x83, 0xAA,
          CMD_SETSTBA , 6, 0x50, 0x50, 0x01, 0x3C, 0x1E, 0x08,
          CMD_SETCYC  , 7, 0x02, 0x40, 0x00, 0x2A, 0x2A, 0x0D, 0x78,

          CMD_SETGAMMA,34,0x02, 0x0A, 0x11, 0x1d, 0x23, 0x35, 0x41, 0x4b,
                          0x4b, 0x42, 0x3A, 0x27, 0x1B, 0x08, 0x09, 0x03,
                          0x02, 0x0A, 0x11, 0x1d, 0x23, 0x35, 0x41, 0x4b,
                          0x4b, 0x42, 0x3A, 0x27, 0x1B, 0x08, 0x09, 0x03,
                          0x00, 0x01,

          CMD_TEON    , 1, 0x00,
          CMD_TEARLINE, 2, 0x00, 0x02,
          CMD_SLPOUT  , 0+CMD_INIT_DELAY, 150,    // Exit sleep mode
          CMD_DISPON  , 0+CMD_INIT_DELAY, 100,
          0xFF,0xFF, // end
      };
      switch (listno)
      {
      case 0: return list0;
      default: return nullptr;
      }
    }
  };

```

wdmomoxx, 2022-12-02:
@tobozo Thank you so much for taking the time to help me. I tried the config above and it still doesn't work. After powering on, just the backlight comes on, and the screen does not respond at all.
[image]

wdmomoxx, 2022-12-03:
@tobozo I tried some more, using the following configuration, the screen has some reaction, but it is still not normal, showing the snowflake screen.I extracted the parameters in this file.
https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_HX8347D.cpp

```
//----------------------------------------------------------------------------

  struct Panel_HX8347D : public Panel_ILI948x
  {
    Panel_HX8347D(void)
    {
      _cfg.memory_width  = _cfg.panel_width  = 240;
      _cfg.memory_height = _cfg.panel_height = 320;
    }

  
  protected:

    const uint8_t* getInitCommands(uint8_t listno) const override
    {
      static constexpr uint8_t list0[] =
      {
          0xEA       , 1, 0x00,
          0xEB       , 1, 0x20,
          0xEC       , 1, 0x0C,
          0xED       , 1, 0xC4,
          0xE8       , 1, 0x38,
          0xE9       , 1, 0x10,
          0xF1       , 1, 0x01,
          0xF2       , 1, 0x10,
          0x40       , 1, 0x01,
          0x41       , 1, 0x00,
          0x42       , 1, 0x00,
          0x43       , 1, 0x10,
          0x44       , 1, 0x0E,
          0x45       , 1, 0x24,
          0x46       , 1, 0x04,
          0x47       , 1, 0x50,
          0x48       , 1, 0x02,
          0x49       , 1, 0x13,
          0x4A       , 1, 0x19,
          0x4B       , 1, 0x19,
          0x4C       , 1, 0x16,
          0x50       , 1, 0x1B,
          0x51       , 1, 0x31,
          0x52       , 1, 0x2F,
          0x53       , 1, 0x3F,
          0x54       , 1, 0x3F,
          0x55       , 1, 0x3E,
          0x56       , 1, 0x2F,
          0x57       , 1, 0x7B,
          0x58       , 1, 0x09,
          0x59       , 1, 0x06,
          0x5A       , 1, 0x06,
          0x5B       , 1, 0x0C,
          0x5C       , 1, 0x1D,
          0x5D       , 1, 0xCC,
          0x1B       , 1, 0x1B,
          0x1A       , 1, 0x01,
          0x24       , 1, 0x2F,
          0x25       , 1, 0x57,
          0x23       , 1, 0x88,
          0x18       , 1, 0x34,
          0x19       , 1, 0x01,
          0x1F       , 4, 0x88, 0x80, 0x90, 0xD0,
          0x17       , 1, 0x05,
          0x28       , 2, 0x38, 0x3F,
          0x16       , 1, 0x18,
          CMD_IDMOFF  , 0,
          CMD_DISPON  , 0+CMD_INIT_DELAY, 100,
          0xFF,0xFF, // end

      };
      switch (listno)
      {
      case 0: return list0;
      default: return nullptr;
      }
    }
  };

//----------------------------------------------------------------------------

```

[image]

tobozo, 2022-12-26:
sorry about the misclosing of this issue, stale bot settings were a bit agressive, reopening

closed 2023-02-01 by github-actions[bot]