lovyan03/LovyanGFX #319: ESP32-s3 and ili9481 not working

magicroomy, 2022-11-27:
## Environment ( 実行環境 )

- ESP32 S3 N16R8
- Panel Driver IC:    ILI9481
- Bus type:           Parallel8
- LovyanGFX version:  v0.4.17
- FrameWork version:   Espressif32 5.2.0 ArduinoESP32 v2.0.5
- Build Environment :   PlatformIO
- Operating System:   Windows

## Display does not show anything. (Black Screen)

### Lovyan example Moving Icons should work

### Just the LEDS of the display are on.

### Steps to reproduce ( 再現のための前提条件 )

Please find attached the src.zip and schematics file.

I designed a dedicated PCB to be used.
I also use the 16 bit bus of the same displays with a different 16 bit PCB and this one works. The difference here is the lgfx_config file to use the 16 bit.

To check out, just try the code.
[image]
[src.zip](https://github.com/lovyan03/LovyanGFX/files/10098196/src.zip)

lovyan03, 2022-11-27:
GPIO33 ~ 37 of ESP32 S3 N16R8 are connected to PSRAM.
[image]

magicroomy, 2022-11-27:
> GPIO33 ~ 37 of ESP32 S3 N16R8 are connected to PSRAM. [image]

Thank you for you immediate answer. I did not have the PS RAM on My Radar.
Nevertheless, the only Pins between 33 and 37 I used was for CS and RS. I changed CS to Pin 3 and RS to pin 8, changed my code.. No effect at all I am afraid.
Any other suggestions for me?

BTW. Great Library !!!

lovyan03, 2022-11-27:
thanks.

```
   cfg.dlen_16bit = true;
```

if you use 8bit parallel. set this parameter to `false`

magicroomy, 2022-11-27:
>    cfg.dlen_16bit = true;
> ```
> 
> if you use 8bit parallel. set this parameter to `false`

Yeah... thank you ... I overlooked this one. Thought the lgfx::Bus_Parallel8 would be all I needed to do.
Now it works !

closed 2022-12-02 by lovyan03