lovyan03/LovyanGFX #324: Push Image to Sprite?

lachlanderrick, 2022-12-05:
Hi,

I have a an unsigned short array that I can push to the display using lcd.pushImage(lcd.pushImage(0,0,320,240,<imageName>[count]);

I'm trying to push the image to a sprite instead, however I can't work out how to do this. 

Can anyone please assist?

Thanks!

tobozo, 2022-12-05:
hi, 

thanks for your feedback.

Have you tried this ?

```cpp
LGFX_Sprite _sprite;
_sprite.setColorDepth( 16 ); // acceptable values: 1, 2, 4, 8, 16, 24
// _sprite.setSwapBytes( true ); // optional: set to true or false if colors are inverted
if( _sprite.createSprite( img_width, img_height ) ) { // allocate sprite, max bytes = 230*230 @ 16bpp
  _sprite.pushImage( 0, 0, img_width, img_height, img_data );
  // do some stuff with the sprite e.g. add some text, border
  // ...
  //
  _sprite.pushSprite( &tft, 0, 0 ); // push sprite to tft
  _sprite.deleteSprite(); // free buffer if the sprite is no longer needed
}

```

lachlanderrick, 2022-12-05:
Hi, yes I tried that and it worked when my image was 96x72 pixels. Thanks!

As a follow up though, I noticed you'd indicated there was a max bye size of 230x230. Originally my image was 320x240 pixels - I scaled it down to make your suggestion work.

Does that mean theres no way to display an image of this size as a sprite? I can use pushImage with the full size image (as indicated in my initial post), but as a sprite it just doesn't work.

For context, im coming from tft_espi, where this worked without any issue.

tobozo, 2022-12-05:
Actually it's 230*230*2 bytes at 16bpp, which means it doubles at 8bpp, quadruples at 4bpp and is 16x more at 1bpp.

you can also use a 160*120 sprite and zoom it at 2.0 scale with `pushImageRotateZoom`

> For context, im coming from tft_espi, where this worked without any issue.

That's because TFT_eSPI auto-selects 8bpp for you whereas LovyanGFX only does what you tell it to do, and only that.

lachlanderrick, 2022-12-06:
Thats very interesting! I didn’t know that. I’ll give that a shot - thanks very much for your help.

closed 2022-12-06 by tobozo

bsramsen, 2025-04-30:
if you have psram you can use 
sprite.setpsram(true);
sprite.createSprite ( bigx,bigy);