lovyan03/LovyanGFX #327: ESP32-C3 12F and RA 8875 (800x480)

osugiw, 2022-12-12:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  AI Thinker ESP32-C3 12F
- Panel Driver IC:    RA8875 
- Bus type:           SPI
- LovyanGFX version:  1.0.0
- FrameWork version:   Arduino IDE 2.0.3
- Build Environment :   ArduinoIDE
- Operating System:   Windows 11

## Problem Description ( 問題の内容 )
The program compiled successfully but display nothing in the RA8875 or print error state in terminal
[image]

### Expected Behavior ( 期待される動作 )
Display text and colour

### Actual Behavior ( 実際の動作 )
Display nothing
[image]

### Steps to reproduce ( 再現のための前提条件 )

1. Download LoyvanGFX library as zip
2. Add zip folder to Arduino IDE as library
3. Open example -> LovyanGFX -> HowToUse -> 2_user_setting
4. Uncomment Panel_RA8875
5. Edit SPI Pins (MISO, MOSI, SCK, CS), RST, INT, and SPI_Host to SPI2_HOST (General Purpose SPI)

// If possible, attach a picture of your setup/wiring here.
RA8875_SCK -> ESP32C3_GPIO4
RA8875_MOSI -> ESP32C3_GPIO6
RA8875_MISO -> ESP32C3_GPIO5
RA8875_INT -> ESP32C3_GPIO19
RA8875_CS -> ESP32C3_GPIO7
RA8875_RST -> ESP32C3_GPIO19
RA8875_LITE -> ESP32C3_GPIO1

### Code to reproduce this issue ( 再現させるためのコード )
https://gist.github.com/osugiw/87b9a042b99295b1686c8a33c28fbc07

lovyan03, 2022-12-12:
Your code is set up to use FT5x06.
The FT5x06 should be an I2C connection, not SPI.
I think the touch panel pin settings are duplicating the LCD pin settings and therefore will not work.

Please remove the touch panel settings first.

osugiw, 2022-12-12:
> Your code is set up to use FT5x06. The FT5x06 should be an I2C connection, not SPI. I think the touch panel pin settings are duplicating the LCD pin settings and therefore will not work.
> 
> Please remove the touch panel settings first.

Thank you for your reply! I already tried your solution, but it still displays nothing.
https://gist.github.com/osugiw/87b9a042b99295b1686c8a33c28fbc07

lovyan03, 2022-12-12:
Are both the RST and INT pins connected to GPIO19?
Perhaps the INT pin needs to be set in `cfg.pin_busy` .

osugiw, 2022-12-12:
> Are both the RST and INT pins connected to GPIO19? Perhaps the INT pin needs to be set in `cfg.pin_busy` .

INT pin connected to GPIO18, while RST pin connected to GPIO19. Do you mean try to assign RST pin to `cfg.pin_busy`?

lovyan03, 2022-12-12:
I mean `cfg.pin_busy = 18;`

lovyan03, 2022-12-12:
You say `INT = GPIO19`
[image]

osugiw, 2022-12-12:
> You say `INT = GPIO19` [image]

I'm sorry my mistake, I wrote typo. It supposed to be GPIO18 for INT

osugiw, 2022-12-12:
> I mean `cfg.pin_busy = 18;`

Should I use WAIT pin also? Because it still not display anything:
https://gist.github.com/osugiw/87b9a042b99295b1686c8a33c28fbc07

lovyan03, 2022-12-12:
Oh, maybe what you need is to set the WAIT pin to `cfg.busy` .
I think the INT pin is for the touch panel.

lovyan03, 2022-12-12:
https://github.com/lovyan03/LovyanGFX/issues/52#issuecomment-1173494377

osugiw, 2022-12-12:
> Oh, maybe what you need is to set the WAIT pin to `cfg.busy` . I think the INT pin is for the touch panel.

Yeah, I already added WAIT pin, it still blank. Is setting `cfg.pin_dc` necessary?
https://gist.github.com/osugiw/87b9a042b99295b1686c8a33c28fbc07 

[image]

lovyan03, 2022-12-12:
The following is an example for ESP32, but please refer to it for the minimum parameters that need to be set.

https://github.com/lovyan03/LovyanGFX/issues/52#issuecomment-1152802280

osugiw, 2022-12-12:
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html
> 
> [image]
> 
> Perhaps GPIO 18 and 19 are used for USB? Try changing them to different pins.

It doesn't matter, because I used another library such as Adafruit_RA8875 it displays properly. By the way, I detached the LITE pin from ESP and now the screen displays white color, but still blank. It sounds weird, but LITE pin (back light pin) affect the display.

[image]

You may look my updated code:
https://gist.github.com/osugiw/87b9a042b99295b1686c8a33c28fbc07

lovyan03, 2022-12-12:
There may be a communication quality problem.
Try lowering the clock to about 5~10 MHz.

```
//      cfg.freq_write = 40000000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_write = 8000000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)

//      cfg.freq_read  = 16000000;    // 受信時のSPIクロック
      cfg.freq_read  =  5000000;    // 受信時のSPIクロック

//      cfg.spi_3wire  = true;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.spi_3wire  = false;        // 受信をMOSIピンで行う場合はtrueを設定
```

Also check that the program is working, including code to output text to the serial monitor.

osugiw, 2022-12-12:
[image]

According to the RA8875 datasheet, the system clock of the RA8875 for 800x480 is like above picture. Do you think this clock setting is for write or read?

lovyan03, 2022-12-12:
I believe that is the pixel clock and not the SPI communication clock.

osugiw, 2022-12-12:
> I believe that is the pixel clock and not the SPI communication clock.

Yeah, I think so. Well, then I'll do trial and error. If you have any idea later about the problem, please let me know, thank you!

tobozo, 2022-12-12:
> What kind of text should be printed to the serial monitor? Because I haven't seen any message appear in my serial monitor.

First set debug output level to `debug` and disable `USB-CDC on boot`, then in your sketch use `Serial.begin(115200); delay(2000);`, and try both `Serial.println("Hello World");` and `log_n("Hello World");` to see which one of JTAG or UART is effectively printed out in the console.

When you know which one of `Serial.print()` or `log_n()` works, use it to debug e.g. print something after `tft.init()`.

osugiw, 2022-12-13:
> First set debug output level to `debug` and disable `USB-CDC on boot`, then in your sketch use `Serial.begin(115200); delay(2000);`, and try both `Serial.println("Hello World");` and `log_n("Hello [...]

It prints no error.
[image]

closed 2022-12-25 by github-actions[bot]