lovyan03/LovyanGFX #328: createPNG() result is a BLACK PNG

PM04290, 2022-12-16:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  WT32-SC01
- Panel Driver IC:    ST7796
- Bus type:           SPI
- LovyanGFX version:  v0.4.18
- FrameWork version:   ArduinoESP32 v1.x
- Build Environment :   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )

using lcd.createPNG() for dump screen (480x320) with 8 parts 120x80, the result is a png of 107 bytes totaly black

### Expected Behavior ( 期待される動作 )

real lcd dump

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

run example savePNG.ino, it's the same thing.

### Code to reproduce this issue ( 再現させるためのコード )

lovyan03, 2022-12-19:
You have not provided any code so I have no idea.

Are you set up to read from the LCD?
Do you have enough memory?

lovyan03, 2022-12-19:
Oh, you are using `WT32-SC01` .
That product does not function to read from the LCD.
If you need to read from the LCD, use another product.

PM04290, 2022-12-19:
Why `WT32-SC01` does not function ? because of WROVER instead of WROOM ?

lovyan03, 2022-12-19:
No, I believe it is because the manufacturer wired the LCD incorrectly.

If you can do a readout, please let me know how.

PM04290, 2022-12-19:
Effectivly, there is no MISO line
[image]
if there a way to know if the panel can readout or not ? so that my program can detect on its own whether the function can be offered or not, depending on the type of panel...
a function like **lcd.isReadOutAvailable()**

lovyan03, 2022-12-28:
Some LCDs can be read out via bidirectional communication using MOSI even without MISO wiring.
However, this is only possible if the LCD driver's IM0 to IM2 pins are set correctly by the board manufacturer.

BTW,  you can use `lcd.isReadable();`

closed 2023-02-03 by github-actions[bot]