lovyan03/LovyanGFX #329: LovyanGFX on Lilygo T-Display S3 or TEmbed

whitegollum, 2022-12-18:
## Environment ( 実行環境 )

- MCU or Board name: 
Lilygo-T-embed: https://github.com/Xinyuan-LilyGO/T-Embed
Lilygo-T-Display: https://github.com/Xinyuan-LilyGO/T-Display-S3

- Panel Driver IC:    ST7789
- Bus type:           Parallel8
- LovyanGFX version:  v0.4.18
- FrameWork version:   ArduinoESP32 v2.0.5
- Build Environment :   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )
Triying to run basic sketch does not work.
The manufacturer provide config examples to TFT_espi:
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/tft/pin_config.h
/*ESP32S3*/
```
#define PIN_LCD_BL                   38

#define PIN_LCD_D0                   39
#define PIN_LCD_D1                   40
#define PIN_LCD_D2                   41
#define PIN_LCD_D3                   42
#define PIN_LCD_D4                   45
#define PIN_LCD_D5                   46
#define PIN_LCD_D6                   47
#define PIN_LCD_D7                   48

#define PIN_POWER_ON                 15

#define PIN_LCD_RES                  5
#define PIN_LCD_CS                   6
#define PIN_LCD_DC                   7
#define PIN_LCD_WR                   8
#define PIN_LCD_RD                   9

#define PIN_BUTTON_1                 0
#define PIN_BUTTON_2                 14
#define PIN_BAT_VOLT                 4

#define PIN_IIC_SCL                  17
#define PIN_IIC_SDA                  18

#define PIN_TOUCH_INT                16
#define PIN_TOUCH_RES                21
```
I just have copy this pin configuration and tried to use it with Lovyan. This is my code:
```
// v1.0.0 を有効にします(v0からの移行期間の特別措置です。これを書かない場合は旧v0系で動作します。)
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

#define PIN_LCD_BL                   38

#define PIN_LCD_D0                   39
#define PIN_LCD_D1                   40
#define PIN_LCD_D2                   41
#define PIN_LCD_D3                   42
#define PIN_LCD_D4                   45
#define PIN_LCD_D5                   46
#define PIN_LCD_D6                   47
#define PIN_LCD_D7                   48

#define PIN_POWER_ON                 15

#define PIN_LCD_RES                  5
#define PIN_LCD_CS                   6
[... 327 lines cut ...]
void loop(void)
{
  Serial.println("loop. changing color..." + String(bucle));
  lcd.startWrite();
  switch (bucle++) {
  case 0:
    lcd.fillScreen(TFT_BLACK);
    lcd.setBrightness(0);
    lcd.drawString("00000", 0, 0);
    break;
  case 1:
    lcd.fillScreen(TFT_GREEN);
    lcd.setBrightness(128);
    lcd.drawString("11111", 0, 0);
    break;
  default:
    bucle = 0;
    break;
}
  delay(1000);
  
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

lovyan03, 2022-12-18:
```
// cfg.dlen_16bit       =  true; // for 16bit parallel
 cfg.dlen_16bit       =  false; // for 8bit parallel
```

whitegollum, 2022-12-19:
thank you Lovyan03, solved de problem.
Sorry for wasting your time. I shoud have seen the solution by myself

closed 2022-12-19 by whitegollum