lovyan03/LovyanGFX #330: drawing Bmp from SPIFFS makes scrampled image

mannasproductions, 2022-12-23:
When I try "drawBmp" with SPIFFS as a source, it prints a scrambled image (see image below, should be a train).
When I try the "drawJpg" with a jpg file it prints no image (not in current code example).
Both images are on the SPIFFS and are both 160x100 pixels.

Can you tell me what is going wrong? Code runs on an esp32 board (esp32-2432s028)
My code is below.

`
#include "FS.h"
#include "SPIFFS.h"
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ILI9341 _panel_instance;
    lgfx::Bus_SPI       _bus_instance;
    lgfx::Light_PWM     _light_instance;
    lgfx::Touch_XPT2046 _touch_instance;
  public:
    LGFX(void)
    {
      {
        auto cfg = _bus_instance.config();

        cfg.spi_host         = VSPI_HOST;
        cfg.spi_mode         = 0;
        cfg.freq_write       = 40000000;
        cfg.freq_read        = 16000000;
        cfg.spi_3wire        = true;
        cfg.use_lock         = true;
        cfg.dma_channel      = 0;
        cfg.pin_sclk         = 14; // change
        cfg.pin_mosi         = 13; // change
        cfg.pin_miso         = 12; // change
        cfg.pin_dc           =  2; // change

        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();

        cfg.pin_cs           =    15; // change
        cfg.pin_rst          =    -1; // change
        cfg.pin_busy         =    -1; // change

        cfg.panel_width      =   240;
        cfg.panel_height     =   320;
        cfg.offset_x         =     0;
        cfg.offset_y         =     0;
        cfg.offset_rotation  =     0;
        cfg.dummy_read_pixel =     8;
        cfg.dummy_read_bits  =     1;
        cfg.readable         =  true;
        cfg.invert           = false;
        cfg.rgb_order        = false;
        cfg.dlen_16bit       = false;
        cfg.bus_shared       = false; // change

        _panel_instance.config(cfg);
      }

      {
        auto cfg = _light_instance.config();

        cfg.pin_bl = 21;              // change
        cfg.invert = false;
        cfg.freq   = 44100;           
        cfg.pwm_channel = 7;

        _light_instance.config(cfg);
        _panel_instance.setLight(&_light_instance);
      }

      {
        auto cfg = _touch_instance.config();

        cfg.x_min      =  300;  // change
        cfg.x_max      = 3900;  // change
        cfg.y_min      = 3700;  // change
        cfg.y_max      =  200;  // change
        cfg.pin_int    = -1;    // change
        cfg.bus_shared = false; // change
        cfg.offset_rotation = 0;

        cfg.spi_host = HSPI_HOST; // change
        cfg.freq = 1000000;
        cfg.pin_sclk = 25;        // change
        cfg.pin_mosi = 32;        // change
        cfg.pin_miso = 39;        // change
        cfg.pin_cs   = 33;        // change

        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);
      }

      setPanel(&_panel_instance);
    }
};

LGFX display;

void setup(void)
{

	Serial.begin(115200);
	display.init();
	
	SPIFFS.begin();	

	display.startWrite();

}

void loop(void)
{
		
	const char * filename="/train_ba.bmp";
	auto file = SPIFFS.open(filename);	
	display.drawBmp(&file, 0, 0);

}
`

[image]

lovyan03, 2022-12-24:
Is the image data in the correct BMP format?
Please attach the file `train_ba.bmp`

mannasproductions, 2022-12-25:
I tested with other files, and now it works perfectly!
I didn't think it would be the file, because i saved the file as different file types, and they all did not work. 
But with the other files it works fine!

Sorry for not testing this in advance, and thanks for your quick response!

closed 2022-12-25 by mannasproductions