lovyan03/LovyanGFX #334: Support T-RGB Esp32-S3

sandmen, 2022-12-30:
Hi,
are there any plans to support the 
https://github.com/Xinyuan-LilyGO/T-RGB

Currently the default repo is using Arduino_GFX...

Thanks

tobozo, 2022-12-30:
Hi, 

The [ST7701S](https://focuslcds.com/content/ST7701S_SPEC_%20V1.2.pdf) module isn't currently supported but panel support could eventually be created from [this example](https://github.com/Xinyuan-LilyGO/T-RGB/blob/d945b06125fdac215225b35f67070ab375e722fe/example/esp-idf-example/T-RGB/main/main.cpp#L51).

I can eventually derivate the default panel and produce a boilerplate but I don't have a T-RGB to play with, so it will highly depend on your feedback and will require fiddling with the code.

I can also guide you into writing your own panel.

Please let us know how you'd prefer to do this.

ihayri, 2022-12-30:
Just for the record, I am available to test code or give feedback as well, as I do have the hardware in question.
Regards, Hayri

sandmen, 2022-12-30:
> 

Hi,
Thanks for the offer.
I think it makes sense, that I try to write a panel for these display.
Is there any documentation for how to do that.
Some hints, templates??

tobozo, 2022-12-30:
what I've understood from T-RGB design:

- XL9535 init is required prior to use both Display and Touch
- after init, FT3267 Touch uses I2C bus
- after init ST7701S Display uses a parallel 16 bits bus for data transfer
- ST7701S Display data transfer requires XL9535 transactions (poking with SDA/SCK/CS/RST pins)

translated in tech debt:

- lgfx/v1/panel/Panel_ST7701S.hpp
- lgfx/v1/touch/Touch_FT3267.hpp
- lgfx/v1/platforms/esp32s2/Bus_XL9535.hpp

which implies the need for:

- writing a basic [XL9535 driver](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/example/factory/XL9535_driver.h) using [Bus_I2C](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32/Bus_I2C.hpp) instead of Wire
- creating Bus_XL9535.hpp by importing code from [Bus_RGB](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp#L77), and using XL9535 driver instead of gpio manipulation when accessing the various non data pins
- creating Touch_FT3267.hpp by integratiing `init()` and `read_pos()` from [FT3267](https://github.com/Xinyuan-LilyGO/T-RGB/blob/d945b06125fdac215225b35f67070ab375e722fe/example/factory/ft3267.h#L61) driver using [Touch_FT5x06](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/touch/Touch_FT5x06.hpp) driver as a template
- creating Panel_ST7701S by integrating the init commands from the [few](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/example/esp-idf-example/T-RGB/main/main.cpp#L51-L94) [examples ](https://github.com/Xinyuan-LilyGO/T-RGB/blob/main/example/GFX/GFX.ino#L24-L67) available using [ST77xx](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/panel/Panel_ST7796.hpp) as a template
- writing a [configuration snippet](https://github.com/lovyan03/LovyanGFX/tree/master/src/lgfx_user)

I have sorted the list by priority but also by complexity, the most complex being at the end.

shauncrain, 2023-01-29:
Hi.   There is now a second model of the T-RGB that uses a different touch panel.  This new version uses FT6856 instead of the FT3267.
screen itseld is still ST7701S, just with a different shape glass and touch panel over it.
I am keen to use these in my devices so will be watching to see if you get it supported in this library.

https://id.aliexpress.com/item/1005004778542414.html?gatewayAdapt=glo2idn

lovyan03, 2023-02-27:
Panel_ST7701 is now supported in the recent version release.
Please refer to here as an example of the configuration and try it out.
https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/LGFX_ESP32S3_RGB_MakerfabsParallelTFTwithTouch40.h

However, I have not yet fully verified the initial setup commands.
There is a possibility that some products may not be displayed correctly.

shauncrain, 2023-02-28:
Thank you.  And well done.   I will try this out.

robertosilvestrigotham, 2023-03-04:
Hi guys!
I have tried this solution with LYLYGO T-Rgb but I cannot get it to work. As far as I understand, the TFT monitor needs to be initialized through commands via i2c passing through XL9535.

closed 2023-04-10 by github-actions[bot]