lovyan03/LovyanGFX #347: color display problem

gtwht, 2023-02-04:
Hello. 

I found a problem with displaying colors for st7789. this problem is described here: [https://forum.arduino.cc/t/problem-with-st7789v-displays-green-value-1-and-every-other-uneven-is-gray/700584/13](url)

 My solution was to add these lines in Panel_ST7789.hpp :

static constexpr uint8_t CMD_RAMCTRL  = 0xB0;
CMD_RAMCTRL, 2, 0x00, 0xF0,  //0xC0
[image]
[image]

lovyan03, 2023-02-05:
Hello, @gtwht 
Thanks for bringing this information to my attention !
I was totally unaware of this.
I have checked the configuration and confirmed that it works, so I have reflected it in the develop branch.

lovyan03, 2023-02-25:
@gtwht 
I have released v1.1.5. I believe this Issue has been resolved.
Please try it.

closed 2023-04-03 by github-actions[bot]

gtwht, 2023-10-20:
Hey!! I have same green problemn with wt32-sc01 plus  ( Panel_ST7796 )
The solution to the problem is exactly the same.