lovyan03/LovyanGFX #349: CVBS in parallel with I2S

zensimilia, 2023-02-06:
## Environment

- MCU or Board name:  Wemos Lolin Lite
- Panel Driver IC: CVBS
- Bus type: I2C
- LovyanGFX version: 1.1.2
- FrameWork version: Espressif 32 (Arduino) 5.1.1
- Build Environment : PlatformIO 6.1.6
- Operating System: Linux

## Problem Description

Hi!

I'm trying to use CVBS panel as video out and in parallel get line in audio signal by I2S (internal ADC for FFT spectrum), but no luck. One by one it works perfect. Serial Monitor says "Errors with Interrupts" and then "Error with i2s driver init". I have tried own i2s driver initialization code and with [arduino-audiokit](https://github.com/pschatzmann/arduino-audiokit) lib, different GPIO and channels - the same.

Functions `analogRead()` and `adc1_get_raw()`/`adc2_get_raw()` still works with CVBS but slow.

Couple questions:

1. Why is not working?
2. **How to make it work?**

### Expected Behavior

- The grass is green, the sun is shining
- CVBS sends video signal to TV
- ADC gets audio signal by I2S, compute FFT and generate audio spectrum on the screen

### Actual Behavior

- Silent
- No video signal
- Couple errors in Serial Monitor
- Kernel still is not in a panic

### Steps to reproduce

1. [Setup CVBS panel](https://gist.github.com/zensimilia/aef32e9d68788d04b4a3f18496b9d986)
2. Setup I2S driver: [witharduino-audiokit](https://gist.github.com/zensimilia/01cb8783c84b22c14ea603c0a900b262) or [without](https://gist.github.com/zensimilia/442b73733d9e2131ab1302564b7f4936)
3. Wiring GPIO26 as video out, GPIO34 (whatever on channels 1 or 2) as audio in

### Code to reproduce this issue

Links to code in above section.

lovyan03, 2023-02-06:
Hello, @zensimilia 
I2S0 - DAC is used for the output of CVBS.
Thus you cannot use I2S0 for any other purpose.

zensimilia, 2023-02-06:
Thanx for quick answer and thanx for this great lib!

So sad that ADC can't work on I2S1 😢 All my ideas always run into strange technical possibilities even though they're not that complex and difficult. Now I'm looking for an external ADC-I2S board. Hope i can connect them by I2S1 channel without any tricks.

zensimilia, 2023-02-07:
Maybe **LCD mode** can help? Referring to the [Technical Reference](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf) it may work at parallel mode.

LCD mode
- Supports multiple LCD modes, including external LCD
- Supports external Camera
- Supports on-chip DAC/ADC modes

zensimilia, 2023-02-08:
I've read in tech manual that LCD mode "supports on-chip DAC/ADC modes" ([page 307](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)) and I thought it might be possible that CVBS can use this mode for transmiting signal to the DAC in parallel. Thereby, do not completely occupy I2S0. If not, ok. Just flight of fancy 😊