lovyan03/LovyanGFX #352: Black screen with sprite full screen

fedyfausto, 2023-02-09:
Hello, I'm using this library to make a little View/Widget manager. To do this I want to manage them with a set of sprites where the main one (the canvas) is full screen and the idea is to draw only the canvas and use it as the destination for the other sprites but if I create a canvas with the exact size of the screen I get a black screen, how could I fix this?

tobozo, 2023-02-09:
hi,

there's a limit to the sprite size depending on the color depth

possible solutions: 

- make your canvas an 8bpp sprite
- divide the canvas into smaller sprites

fedyfausto, 2023-02-09:
Ok but the clock example is my idea, the clock had a canvas and a lot of sprites but there is a Zoom calculated, another solution could be to make the canvas 160 x 120 and then push it with zoom?

tobozo, 2023-02-09:
the clocksample sketch only uses zoom to fit screen dimensions and satisfy the portability of the demo so it's more a workaround than a solution, also it's using 4bpp

```cpp
  canvas.setColorDepth(lgfx::palette_4bit);  // 各部品を４ビットパレットモードで準備する
  clockbase.setColorDepth(lgfx::palette_4bit);
  needle1.setColorDepth(lgfx::palette_4bit);
  shadow1.setColorDepth(lgfx::palette_4bit);
  needle2.setColorDepth(lgfx::palette_4bit);
  shadow2.setColorDepth(lgfx::palette_4bit);

  canvas.createSprite(width, width); // メモリ確保
```

fedyfausto, 2023-02-09:
Ok, I'm trying with 8 bits of depth, then I create two sprites and push the first one in the new position into the canvas and then I push the final canvas, is this right? 
```
void setup()
{
  Serial.begin(115200);
  tft.init();
  tft.setRotation(3);
  int lw = std::min(tft.width(), tft.height());

  zoom = (float)lw / tft.width();
  canvas.setColorDepth(8);
  canvas.createSprite(320, 240);
  sprite.setColorDepth(2);
  sprite.createSprite(40,40);
  sprite.fillCircle(x, 120, 40, 0xff3400U);
  // tft.drawRect(0,0,10,10, 0x00ff00U);
  // tft.drawRect(tft.width() - 10,0,10,10, 0x00ff00U);
  // tft.drawRect(0,tft.height()-10,10,10, 0x00ff00U);
  // tft.drawRect(tft.width() - 10,tft.height()-10,10,10, 0x00ff00U);
  Serial.println("Display initialised");
  tft.startWrite();
}

void loop() {

  //sprite.fillCircle(x, 120, 40, 0x0000ffU);
  x+=1;
  if(x > tft.width()+20) x = -40;
  //sprite.fillCircle(x, 120, 40, 0xff3400U);
  // canvas.drawRect(0,0,10,10, 0x00ff00U);
  // canvas.drawRect(tft.width() - 10,0,10,10, 0x00ff00U);
  // canvas.drawRect(0,tft.height()-10,10,10, 0x00ff00U);
  // canvas.drawRect(tft.width() - 10,tft.height()-10,10,10, 0x00ff00U);
  sprite.pushSprite(&canvas, x,0);
  canvas.pushRotated(&tft, 0,0);
  //tft.display();
}
```

closed 2023-03-18 by github-actions[bot]