lovyan03/LovyanGFX #353: How to Assigned PIN for RGB Panel

uldara1, 2023-02-11:
Dear Sir
My device is RGB Panel model ZX3D95CE01S-AR-4848 with ESP32 S3
can i use this module with your library ? if possible how to assigned pin 
please check attached file for detail pin 
thank
[ZX3D95CE01S-AR-4848.pdf](https://github.com/lovyan03/LovyanGFX/files/10713819/ZX3D95CE01S-AR-4848.pdf)

lovyan03, 2023-02-11:
Hello, @uldara1 

Please refer to this setup.
https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/LGFX_ESP32S3_RGB_ESP32-8048S043.h

uldara1, 2023-02-12:
Thank @lovyan03 
let me try

uldara1, 2023-02-13:
Dear @lovyan03 
i have tested RGB Panel and it work anyway i have other problem with LVGL library 
i tested with 2 library 
- lovyanGFX
- ArduinoGFX
result
- lovyanGFX display seem problem with frame update
- ArduinoGFX work correct

please check code and video below 

[LovyanGFX_RGB_Panel.zip](https://github.com/lovyan03/LovyanGFX/files/10719757/LovyanGFX_RGB_Panel.zip)

[ArduinoGFX_RGB_Panel.zip](https://github.com/lovyan03/LovyanGFX/files/10719760/ArduinoGFX_RGB_Panel.zip)

lovyan03, 2023-02-14:
Perhaps the transmit rate is too high and the PSRAM is not being accessed in time.
Try reducing the speed of freq_write.
```
      cfg.freq_write  = 15000000;
```
Changing this value to perhaps 120000000 or so will probably fix it.

uldara1, 2023-02-14:
Dear @lovyan03 ,
i have update freq_write from high to low result still blink.
just to update you for ArduinoGFX if set auto_flush = 0 screen blink too. if auto_flush = 1 screen work find.
Addiction with your library to work with RGB Panel i need to add some configure with SPI please check 

```
static LGFX lcd;
Arduino_DataBus *bus = new Arduino_SWSPI( GFX_NOT_DEFINED /* DC */, 0 /* CS */, 10 /* SCK */, 9 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

bus->begin();
rgb_driver_init(void);

lcd.init();
lcd.setBrightness(230);

static void rgb_driver_init(void)
{
  bus->sendCommand(0x01); // reset
  delay(120);

	bus->sendCommand (0xF0);
	bus->sendData (0x55);
	bus->sendData (0xAA);
	bus->sendData (0x52);
	bus->sendData (0x08);
	bus->sendData (0x00);

	bus->sendCommand (0xF6);
	bus->sendData (0x5A);
	bus->sendData (0x87);

	bus->sendCommand (0xC1);
	bus->sendData (0x3F);

	bus->sendCommand (0xC2);
	bus->sendData (0x0E);

	bus->sendCommand (0xC6);
	bus->sendData (0xF8);

	bus->sendCommand (0xC9);
	bus->sendData (0x10);

	bus->sendCommand (0xCD);
	bus->sendData (0x25);

	bus->sendCommand (0xF8);
	bus->sendData (0x8A);

	bus->sendCommand (0xAC);
	bus->sendData (0x45);

	bus->sendCommand (0xA0);
	bus->sendData (0xDD);

	bus->sendCommand (0xA7);
	bus->sendData (0x47);

	bus->sendCommand (0xFA);
	bus->sendData (0x00);
	bus->sendData (0x00);
	bus->sendData (0x00);
	bus->sendData (0x04);

	bus->sendCommand (0x86);
	bus->sendData (0x99);
	bus->sendData (0xa3);
	bus->sendData (0xa3);
	bus->sendData (0x51);

[... 507 lines cut ...]
	bus->sendData (0x03);
	bus->sendData (0xba);
	bus->sendData (0x03);
	bus->sendData (0xc5);
	bus->sendData (0x03);
	bus->sendData (0xd0);
	bus->sendData (0x03);
	bus->sendData (0xE0);
	bus->sendData (0x03);
	bus->sendData (0xea);
	bus->sendData (0x03);
	bus->sendData (0xFa);
	bus->sendData (0x03);
	bus->sendData (0xFF);

	bus->sendCommand (0x3a);
	bus->sendData (0x66);

	bus->sendCommand (0x11);
	delay(120);

	bus->sendCommand (0x29);
	delay(20);
}
```

lovyan03, 2023-02-14:
Panel_GC9503 was added to the develop branch.
Additional pin settings are available

```
auto cfg = _panel_instance.config_detail();
cfg.pin_cs = 0;
cfg.pin_sclk = 10;
cfg.pin_mosi = 9;
_panel_instance.config_detail(cfg);
```

uldara1, 2023-02-15:
Hi @lovyan03 
Thank for your update 
Screen start displaying 
But blinking frame still happens 
https://user-images.githubusercontent.com/22394956/218954538-672ecc50-4b4d-4a01-924e-37c9852196a6.MOV

Thank

lovyan03, 2023-02-15:
Currently, this type of display works by reserving a frame buffer in the PSRAM of the ESP32S3. Therefore, there is no solution other than to make freq_write slower.

Of course, in terms of hardware specifications, I suppose it would be possible to implement the transfer to the line buffer repeatedly by interrupts without creating a frame buffer, but that is too far off from the current LovyanGFX specifications to be supported.

uldara1, 2023-02-15:
Thank for your support 🙏