lovyan03/LovyanGFX #355: My project no longer compiles!

Galbi3000, 2023-02-16:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  M5StickC
- Panel Driver IC:    ST7789
- LovyanGFX version:  1.1.2 (Provided in Arduino IDE Libraries, version selected is 0.5.0, the highest available in the drop box)
- Build Environment :   Arduino IDE 2.0.3
- Operating System:   Windows 11

## Problem Description ( 問題の内容 )
I have a project that used to compile fine, I have not made any changes to it since it last compiled other than to enable debugging output to serial (a switch in my code) to prepare for debugging a runtime issue that has cropped up in time due to external dependencies. Now, a few years later, the project fails to compile! All the errors are related to the first one which is:

```
In file included from D:\Documents\Arduino\TeamSpeakMonitorM5StickC\TeamSpeakMonitorM5StickC.ino:31:0:
D:\Documents\Arduino\TeamSpeakMonitorM5StickC\Display.h:11:8: error: 'LGFX' does not name a type
 static LGFX lcd;
        ^
```
This is bugging me as to why the code no longer compiles. Example code from the LovyanGFX library compile fine and they also use the "static LGFX lcd;" line that produces the error for me!
I have tried a complete re-install of Arduino IDE and removed all libraries except those necessary for the project with no success.

### Code to reproduce this issue ( 再現させるためのコード )
You can see the code in full in my GitHub:  [TS3Monitor-M5StickC](https://github.com/Galbi3000/TS3Monitor-M5StickC)

tobozo, 2023-02-16:
hi, 

the project compiles without error if you add `#define LGFX_AUTODETECT` :

```cpp
#define LGFX_M5STICK_C
#define LGFX_AUTODETECT
#include <LovyanGFX.hpp>
```

Galbi3000, 2023-02-16:
> hi,
> 
> the project compiles without error if you add `#define LGFX_AUTODETECT` :
> 
> ```c++
> #define LGFX_M5STICK_C
> #define LGFX_AUTODETECT
> #include <LovyanGFX.hpp>
> ```

Thanks, though that does not clear up the mystery of why it did not compile before.

Today, after the re-install of Arduino IDE and the project failing to compile yet again, I loaded the 1_simple_use.ino example project and uncommented line 10 (#define LGFX_M5STICK_C) plus commented out line 30 (// #define LGFX_AUTODETECT). Essentially the same defines as in my project. It compiled fine.

Now I will go and make sure it still works in the M5StickC with that change.

Galbi3000, 2023-02-16:
Yep, it still works. Thank you very much. Now I can go on with debugging the issue I needed to debug 🙂

closed 2023-02-16 by Galbi3000

tobozo, 2023-02-16:
since that project seems to use both M5 syntax and LGFX syntax, have you tried [M5Unified](https://github.com/m5stack/M5Unified) ? It's a mix of both, with support for many M5 devices. Also it's been updated today.

Galbi3000, 2023-02-16:
I've not looked at it in years. If you look in my issues on my project you will see why I have come back to it! lol
The code works fine but communications with the TeamSpeak server seems to have problems after an update to the server so I needed to enable debugging to see if I could figure out that problem and ran into the compile issue!
I did notice the M5Unified library and installed it, but will need to look into it further before I make any major changes to my project.