lovyan03/LovyanGFX #356: Error compiling with PlatformIO and Arduino+RP2040

PhilippMolitor, 2023-02-22:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  RPi2040
- Panel Driver IC:    GC9A01
- Bus type:           SPI
- LovyanGFX version:  latest (1.1.2)
- FrameWork version:   Arduino
- Build Environment :   PlatformIO
- Operating System:   macOS

platformio.ini:
```ini
[env]
platform = raspberrypi
board = pico
framework = arduino
```

## Problem Description ( 問題の内容 )
I get this error when compiling:
```
.pio/libdeps/debug/LovyanGFX/src/lgfx/v1/platforms/rp2040/Light_PWM.cpp:25:10: fatal error: pico/stdio.h: No such file or directory
   25 | #include <pico/stdio.h>
      |          ^~~~~~~~~~~~~~
```

It seems to include from the `pico` headers, but it is supposed to use the Arduino libraries instead.

### Expected Behavior ( 期待される動作 )

Compilation should succeed without errors.

### Actual Behavior ( 実際の動作 )

As stated above

### Code to reproduce this issue ( 再現させるためのコード )

See this repository:
[https://github.com/PhilippMolitor/waveshare-rp2040-roundlcd-boilerplate](https://github.com/PhilippMolitor/waveshare-rp2040-roundlcd-boilerplate)

The configuration of LovyanGFC happens in `lib/display/display.h`

tobozo, 2023-02-22:
hi, thanks for your feedback :+1: 

Only [earlephilhower/arduino-pico core](https://github.com/earlephilhower/arduino-pico) is supported.

Please check the [platform readme](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst#current-state-of-development), some modifications are required in your ini file:

[image]

> it is supposed to use the Arduino libraries instead.

It's not always true that using Arduino wiring library is the optimal choice; using the lower layer functions often gives performances improvements, and reduce the binary size.

PhilippMolitor, 2023-02-22:
Thank you, that made it work. Do you think it makes sense to document this requirement? (at least I did not find it anywhere when looking through this repo).

Also, if you find it useful, feel free to link the URL to my boilerplate repo as an example if needed. It showcases the use of this library with a round 240x240 GC9A01 panel on a waveshare board. I will update it soon with the required changes.

closed 2023-02-22 by PhilippMolitor

tobozo, 2023-02-22:
Well, the pico board was missing from the platformio CI build (only Arduino build tests were performed for this board), so thanks to you! 

Here's the config sample I've added based on our exchange (also [test run](https://github.com/lovyan03/LovyanGFX/actions/runs/4246273389/jobs/7382880104)):

https://github.com/lovyan03/LovyanGFX/blob/develop/examples/Test/build_test/platformio_rpipico.ini

Not really boilerplates, but some specific config samples are stored in the [lgfx_user](https://github.com/lovyan03/LovyanGFX/tree/develop/src/lgfx_user) folder, so you're welcome to submit yours on the [develop branch](https://github.com/lovyan03/LovyanGFX/tree/develop).