lovyan03/LovyanGFX #360: Set SPI display and xpt2046 touch to use the same SPI bus error restart

xxzling, 2023-02-26:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name: esp32-s3
- Panel Driver IC:    ILI9341
- Bus type:           SPI
- LovyanGFX version:  latest develop
- FrameWork version:   ArduinoESP32 v2.0+
- Build Environment :  PlatformIO
- Operating System:  Windows

## Problem Description ( 問題の内容 )

I tried to use the display screen and touch controller on the same SPI bus(SPI pin, SCK, MISO and MOSI are identical, only CS pin is different), but when loading the LVGL widget sample program, my device repeatedly rebooted, and no error log was generated. When running the lvgl benchmark test program, I found that the touch worked normally, but when I swiped the screen to the image loading test, the program card died and then restarted, because I did not find a similar configuration method, so I might have made a configuration error, Please ask the author to help answer. Thank you very much for your efforts. My setting method is attached below!

## My Test Code

-TFT SPI SETUP:

      auto cfg = _bus_instance.config();

      // SPIバスの設定
      cfg.spi_host = SPI2_HOST;  // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
      cfg.spi_mode = 0;                   // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write =80000000;         
      cfg.freq_read = 20000000;          
      cfg.spi_3wire = true;               // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock = true;             
      cfg.dma_channel = SPI_DMA_CH_AUTO;  SPI_DMA_CH_AUTO=自動設定)
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 40;  // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 42;  // SPIのCLKピン番号を設定
      cfg.pin_miso = -1;  // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc = 48;     // SPIのD/Cピン番号を設定  (-1 = disable)
      cfg.pin_cs = 39;    // CSが接続されているピン番号   (-1 = disable)

      _bus_instance.config(cfg);               // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);  // バスをパネルにセットします

-TOUCH SPI SETUP:

       cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
       cfg.x_max      = 239;  // タッチスクリーンから得られる最大のX値(生の値)
       cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
       cfg.y_max      = 319;  // タッチスクリーンから得られる最大のY値(生の値)
       cfg.pin_int    = 47;   // INTが接続されているピン番号
       cfg.bus_shared = true; // 画面と共通のバスを使用している場合 trueを設定
       cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

      // SPI接続の場合
      cfg.spi_host = SPI2_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST)
      cfg.freq = 1000000;     // SPIクロックを設定
      cfg.pin_sclk = 40;     // SCLKが接続されているピン番号
      cfg.pin_mosi = 42;     // MOSIが接続されているピン番号
      cfg.pin_miso = 41;     // MISOが接続されているピン番号
      cfg.pin_cs   =  3;     //   CSが接続されているピン番号

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。

-Thanks you very much!

xxzling, 2023-02-26:
I'm not sure if LovyanGFX supports this function. I imagine using tft_ The espi library also uses my xpt2046 touch screen, only sets one CS pin, and uses the same SPI bus. Thank you again!

lovyan03, 2023-02-27:
bus_instance config...
```
//  cfg.pin_miso = -1;  // bad setting
    cfg.pin_miso = 41;  // good setting
```

xxzling, 2023-02-27:
thanks for your replay,it's lready solved

closed 2023-02-27 by xxzling