lovyan03/LovyanGFX #367: Touch Calibration

codegrue, 2023-03-01:
Is there support for touch coordinates that don't match the screen coordinates (e.g. touch area extends beyond the visual screen area)? I don't see a way currently to calibrate the values.

tobozo, 2023-03-02:
Yes there is support, see the M5Unified implementation of [M5Core2 touch config](https://github.com/m5stack/M5Unified/blob/master/src/M5Unified.cpp#L896) where the touch panel is physically larger than the display panel.

So it's only a matter of performing a basic calibration and a logic check:  `if (raw.y > 240)` >>> outside display panel area.

[See the autodetect sequence](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32.hpp#L972)

codegrue, 2023-03-02:
I looked through that code and I'm not sure if my scenario is the same. I think in the M5 case the values returned by the touch and the screen don't match. In my case, the touch area is wider than the physical screen, so 0,0 is actually off the visual screen area to the top left, what would effectively be -20, -10.

My screen is 480x320, but when I touch the visible corners I get (30,10) and (460, 300) respectively. 

I use this code currently to recalculate the intended touch point from the returned touch values:

```
    lgfx::touch_point_t tp;
    int touched = lcd.getTouch(&tp);

    int touch_x = tp.x;
    int touch_y = tp.y;

    // calibrate for touch extents
    int calibrated_x = 1.0 * (touch_x - TOUCH_X_MIN) / (TOUCH_X_MAX - TOUCH_X_MIN) * SCREEN_WIDTH;
    calibrated_x = max(0, calibrated_x);
    calibrated_x = min(SCREEN_WIDTH, calibrated_x);
    int calibrated_y = 1.0 * (touch_y - TOUCH_Y_MIN) / (TOUCH_Y_MAX - TOUCH_Y_MIN) * SCREEN_HEIGHT;
    calibrated_y = max(0, calibrated_y);
    calibrated_y = min(SCREEN_HEIGHT, calibrated_y);
```

In your example, I think you are just setting the coordinate values to return so the values match the screen pixels, but this assumes they physically overlay. In case you are wondering, this is the screen I am using:

https://wiki.makerfabs.com/ESP32_S3_Parallel_3.5_TFT_with_Touch.html

And if you look at the image, the class extends to the left and right over a black margins, but picks up touches in this area.

tobozo, 2023-03-02:
It's hard to say which one of the touch driver or the application should be responsible for the coordinates translation and offset, but both are possible.

Maybe you can get some inspiration from the touch section in [this config snippet](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/LGFX_ESP32S2_MakerabsParallelTFTwithTouch.hpp#L41).

closed 2023-04-08 by github-actions[bot]