lovyan03/LovyanGFX #368: Error compiling with PlatformIO and Espressif 32 and ESP32C3

mark8018, 2023-03-02:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  Espressif ESP32-C3-DevKitM-1
- Panel Driver IC: ILI9488
- Bus type:  SPI
- LovyanGFX version:  v1.1.5(latest)
- FrameWork version:   arduinoespressif32 @ 3.20006.221224 (2.0.6)
- Build Environment :   PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )
Compiling .pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32\Bus_SPI.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32\Light_PWM.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32\Panel_CVBS.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32\common.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32c3\Bus_Parallel8.cpp.o
.pio/libdeps/esp32-c3-devkitm-1/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:32:10: fatal error: soc/dport_reg.h: No such file or directory
 #include <soc/dport_reg.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32-c3-devkitm-1\libbf3\LovyanGFX\lgfx\v1\platforms\esp32\Bus_SPI.cpp.o] Error 1

### Expected Behavior ( 期待される動作 )
Compilation should succeed without errors.

### Actual Behavior ( 実際の動作 )
Compile error.

### Steps to reproduce ( 再現のための前提条件 )
may be select chip set of ESP32C3.

tobozo, 2023-03-02:
hi, thanks for reporting this :+1: 

I'm not sure why ESP32-C3 wasn't part of the CI tests, the profile seems to exist in the platformio.ini so I've [restored the entries](https://github.com/lovyan03/LovyanGFX/commit/36bca4e0d482db8e242e21f3d9b4104272b827a1).

The automated build produces a similar result indeed.

[image]

quickfix: edit `lgfx/v1/platforms/esp32/Bus_SPI.cpp` and replace `#include <soc/dport_reg.h>` by this:

```cpp
#if !defined ( CONFIG_IDF_TARGET_ESP32C3 )
  #include <soc/dport_reg.h>
#endif
```

mark8018, 2023-03-03:
Thank you for the solution.
Compilation has been successfully completed.

commit f1a51391 (2023-03-02): ESP32-C3 doesn't have dport_reg, fixes #368
closed 2023-04-10 by github-actions[bot]