lovyan03/LovyanGFX #374: Problem with ESP32s3 and 480x272 RGB display (module ESP32-4827S043)

Emmanuele75, 2023-03-21:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32-4827S043) [ESP32s3 + 4.3" RGB display]
- Panel Driver IC:    ILI6485
- Bus type:           RGB
- LovyanGFX version:  1.1.5
- FrameWork version:   ArduinoESP32 v2.0.6
- Build Environment :   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )

I encountered a serious bug with the ESP32-4827S043 module, which houses an ESP32S3 and a 4.3" RGB display.
In particular there seems to be problems with the LVGL library, under some conditions the display is not drawn completely.
This happens when there are few animations, on the contrary it seems to work well with complex drawings and lots of active animations.
From what I understand the problem lies with writePixels.
I tried changing the flush_cb from:

void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p ){
    uint32_t w = ( area->x2 - area->x1 + 1 );
   uint32_t h = ( area->y2 - area->y1 + 1 );
   tft.startWrite();
   tft.setAddrWindow( area->x1, area->y1, w, h );
   tft.writePixels((lgfx::rgb565_t *)&color_p->full, w * h);
   tft.endWrite();
   lv_disp_flush_ready(disp);
}

to:

void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p){\n
  uint16_t c;
  uint32_t w = ( area->x2 - area->x1 + 1 );
  uint32_t h = ( area->y2 - area->y1 + 1 );
  tft.startWrite(); 
  tft.setAddrWindow(area->x1, area->y1, w, h);
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.writeColor(c, 1);
      color_p++;
    }
  }
  tft.endWrite();
  lv_disp_flush_ready(disp); 
}

Now the display works fine but it's terribly slow and the animations aren't smooth.
How to solve this problem?
Thanks!
[image]

tobozo, 2023-03-21:
hi, thanks for your feedback :+1: 

this is untested code, but this loop may be slightly faster:

```cpp
  for (int y = area->y1; y <= area->y2; y++) {
    tft.setAddrWindow(area->x1, y, w, 1);
    tft.writePixels((lgfx::rgb565_t *)&color_p->full[y*w], w);
  }
```

Emmanuele75, 2023-03-21:
`I can't compile, I get this error:

C:\Users\emman\Desktop\KEY\COD\LvglWidgets\LvglWidgets.ino: In function 'void my_disp_flush(lv_disp_drv_t*, const lv_area_t*, lv_color_t*)':
LvglWidgets:78:57: error: invalid types 'uint16_t {aka short unsigned int}[int]' for array subscript
     tft.writePixels((lgfx::rgb565_t *)&color_p->full[y*w], w);
                                                         ^
exit status 1
invalid types 'uint16_t {aka short unsigned int}[int]' for array subscript

`

tobozo, 2023-03-21:
yep, untested code means it probably has errors you have to fix

I can't check now but the idea is to improve the performance by using `writePixels()` to send a full line instead of one pixel per iteration.

maybe try one of those:

```cpp
    tft.writePixels((lgfx::rgb565_t*)(&color_p->full)[y*w], w);
```

```cpp
    uint16_t* framebufptr = (uint16_t*)&color_p->full;
    uint16_t* scanlineptr = framebufptr + (y*w);
    tft.writePixels((lgfx::rgb565_t*)scanlineptr, w);
```

Emmanuele75, 2023-03-21:
okay. I'll try)

Emmanuele75, 2023-03-21:
Following your suggestion I compiled this code:
`void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p ){
  int c = 0;
  uint32_t w = ( area->x2 - area->x1 + 1 );
  uint32_t h = ( area->y2 - area->y1 + 1 );
  tft.startWrite();
  tft.setAddrWindow(area->x1, area->y1, w, h);
  for (int y = area->y1; y <= area->y2; y++) {
    uint16_t* framebufptr = (uint16_t*)&color_p->full;
    uint16_t* scanlineptr = framebufptr + (c*w);
    tft.writePixels((lgfx::rgb565_t*)scanlineptr, w);
    c++;
  }     
   tft.endWrite();
   lv_disp_flush_ready(disp);
}`
Unfortunately the result does not change, the display is not drawn well.
So at the moment the only solution is to draw point by point.
Do you have any other suggestions for me to get rid of this bug?
Thanks!

tobozo, 2023-03-21:
> Do you have any other suggestions for me to get rid of this bug?

not without seeing your LGFX configuration

Emmanuele75, 2023-03-21:
My lgfx_4827S043.h:

```cpp
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

class LGFX : public lgfx::LGFX_Device{
  lgfx::Bus_RGB     _bus_instance;
  lgfx::Panel_RGB   _panel_instance;
  lgfx::Light_PWM   _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

public:LGFX(void){
  auto cfg    = _bus_instance.config();
  cfg.panel   = &_panel_instance;
  cfg.pin_d0  = GPIO_NUM_8;  // B0
  cfg.pin_d1  = GPIO_NUM_3;  // B1
  cfg.pin_d2  = GPIO_NUM_46; // B2
  cfg.pin_d3  = GPIO_NUM_9;  // B3
  cfg.pin_d4  = GPIO_NUM_1;  // B4
  cfg.pin_d5  = GPIO_NUM_5;  // G0
  cfg.pin_d6  = GPIO_NUM_6;  // G1
  cfg.pin_d7  = GPIO_NUM_7;  // G2
  cfg.pin_d8  = GPIO_NUM_15; // G3
  cfg.pin_d9  = GPIO_NUM_16; // G4
  cfg.pin_d10 = GPIO_NUM_4;  // G5
  cfg.pin_d11 = GPIO_NUM_45; // R0
  cfg.pin_d12 = GPIO_NUM_48; // R1
  cfg.pin_d13 = GPIO_NUM_47; // R2
  cfg.pin_d14 = GPIO_NUM_21; // R3
  cfg.pin_d15 = GPIO_NUM_14; // R4
  cfg.pin_henable       = GPIO_NUM_40;
  cfg.pin_vsync         = GPIO_NUM_41;
  cfg.pin_hsync         = GPIO_NUM_39;
  cfg.pin_pclk          = GPIO_NUM_42;
  cfg.freq_write        = 9000000;   //16000000
  cfg.hsync_polarity    = 0; //0
  cfg.hsync_front_porch = 8; //8
  cfg.hsync_pulse_width = 1; //4
  cfg.hsync_back_porch  = 43; //43
  cfg.vsync_polarity    = 0;  //0
  cfg.vsync_front_porch = 8; //8
  cfg.vsync_pulse_width = 4; //4
  cfg.vsync_back_porch  = 12; //12
  cfg.pclk_idle_high    = 1; //1
  _bus_instance.config(cfg);
  _panel_instance.setBus(&_bus_instance);
  
  { auto cfg = _panel_instance.config();
  cfg.memory_width  = 480;
  cfg.memory_height = 272;
  cfg.panel_width   = 480;
  cfg.panel_height  = 272;
  cfg.offset_x      = 0;
  cfg.offset_y      = 0;
  _panel_instance.config(cfg);
  }
  
  { auto cfg = _panel_instance.config_detail();
  cfg.use_psram = 1;
  _panel_instance.config_detail(cfg);
  }

  { auto cfg = _light_instance.config();
  cfg.pin_bl = GPIO_NUM_2;
  _light_instance.config(cfg);
  }
  _panel_instance.light(&_light_instance);

{ 
      auto cfg = _touch_instance.config();
      cfg.x_min      = 100;    
      cfg.x_max      = 4000;   
      cfg.y_min      = 100;     
      cfg.y_max      = 4000;    
      cfg.pin_int    = -1;    
      cfg.bus_shared = true;  
      cfg.offset_rotation = 0;// 0-7
      cfg.spi_host = 1; // 
      cfg.freq = 1000000;    
      cfg.pin_sclk = 12;     
      cfg.pin_mosi = 11;     
      cfg.pin_miso = 13;     
      cfg.pin_cs   = 38;     
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  
    }
  setPanel(&_panel_instance); 
  }
};
static LGFX tft; 
```

tobozo, 2023-03-21:
`cfg.freq_write        = 9000000;` seems strange, did you try other values ? 2MHz should be a reasonable start

also I'm not sure about `cfg.bus_shared = true;` as it doesn't seem like panel and touch are using the same bus

Emmanuele75, 2023-03-21:
As for `cfg.freq_write` it works from 8MHz to 12MHz other values cause the display to flicker and not solve the problem.
For `cfg.bus_shared = true` or `false` nothing changes, but still tough works fine.

Emmanuele75, 2023-03-21:
I just did the test by changing the parameters` hsync_polarity` and `vsync_polarity` with 1, the result doesn't change.
0 or 1 the display is also drawn badly

tobozo, 2023-03-21:
`hsync_pulse_width = 1` seems under the minimal value 

[image]

Emmanuele75, 2023-03-21:
I will do the tests with the new parameters and I will send you the photos.

Emmanuele75, 2023-03-21:
[image]
Image 1 with the new parameters, without CPU and memory usage.
Image 2 is about point to point display draw.

Emmanuele75, 2023-03-21:
I tried the library Arduino_GFX_Library.h and the display seems to work fine, it draws everything and fast...the problem is that the display flickers :(

tobozo, 2023-03-21:
comparing image1 with the image in the first post shows that it had *some* effect, so maybe it's worth poking with the values.

It would be interesting to know if that glitch always happens at the same line (e.g is that 240th or 256th?).

If so then it could be possible to just split the buffer in two separate writes, e.g. write the first chunk, use tft.endWrite() + tft.startWrite() and write the second chunk? 

again, this is untested pseudo-code:

```cpp

void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p ){
  int c = 0;
  uint32_t w = ( area->x2 - area->x1 + 1 );
  uint32_t h = ( area->y2 - area->y1 + 1 );

  uint32_t max_height = 240; // adjust this

  tft.setAddrWindow(area->x1, area->y1, w, h);

  tft.startWrite();

  if( h > max_height ) {
      tft.writePixels((lgfx::rgb565_t*)&color_p->full, w*max_height); // write first chunk
      tft.endWrite();
      tft.startWrite();
      uint16_t* framebufptr = (uint16_t*)&color_p->full;
      uint16_t* scanlineptr = framebufptr + (max_height*w);
      tft.writePixels((lgfx::rgb565_t*)scanlineptr, w*(h%max_height)); // write second chunk
  } else {
     tft.writePixels((lgfx::rgb565_t*)&color_p->full, w*h);  // write full buffer
  }

  tft.endWrite();
  lv_disp_flush_ready(disp);
}

```

Emmanuele75, 2023-03-21:
I tried the new code, nothing changes, 240/256 same result.
As for the images I sent you, the difference between the first and last is that the CPU USAGE is activated in the first.

Emmanuele75, 2023-03-21:
With active cpu usage the display works better!

Emmanuele75, 2023-03-21:
Video

https://user-images.githubusercontent.com/128136690/226758261-7cb16802-394d-480d-aed6-bd8362bc9cc0.mp4

Emmanuele75, 2023-03-21:
draw point to point:

https://user-images.githubusercontent.com/128136690/226759187-9bb0179a-4cc4-4ea8-833f-d511e36ada1b.mp4

lovyan03, 2023-03-21:
@Emmanuele75 @tobozo 
Sorry, I have an idea of the cause and will now work on a fix in the develop branch.
The cause is that you are not calling the write back function properly when writing to PSRAM.

lovyan03, 2023-03-21:
Incidentally, the following settings on the user side would improve performance and solve the problem.

 - Enable byte swap in LVGL settings

lv_conf.h
```
#define LV_COLOR_16_SWAP 1
```

 - Make the output function as follows
```
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p){
  tft.pushImageDMA( area->x1
                  , area->y1
                  , area->x2 - area->x1 + 1
                  , area->y2 - area->y1 + 1
                  , ( lgfx::swap565_t* )&color_p->full );
  lv_disp_flush_ready(disp); 
}
```

Emmanuele75, 2023-03-22:
I did as you said, but unfortunately nothing changes!

Emmanuele75, 2023-03-22:
ok, i will let you know soon

Emmanuele75, 2023-03-22:
I forgot... This is my display configuration:
```cpp
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

class LGFX : public lgfx::LGFX_Device{
  lgfx::Bus_RGB     _bus_instance;
  lgfx::Panel_RGB   _panel_instance;
  lgfx::Light_PWM   _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

public:LGFX(void){
  auto cfg    = _bus_instance.config();
  cfg.panel   = &_panel_instance;
  cfg.pin_d0  = GPIO_NUM_8;  // B0
  cfg.pin_d1  = GPIO_NUM_3;  // B1
  cfg.pin_d2  = GPIO_NUM_46; // B2
  cfg.pin_d3  = GPIO_NUM_9;  // B3
  cfg.pin_d4  = GPIO_NUM_1;  // B4
  cfg.pin_d5  = GPIO_NUM_5;  // G0
  cfg.pin_d6  = GPIO_NUM_6;  // G1
  cfg.pin_d7  = GPIO_NUM_7;  // G2
  cfg.pin_d8  = GPIO_NUM_15; // G3
  cfg.pin_d9  = GPIO_NUM_16; // G4
  cfg.pin_d10 = GPIO_NUM_4;  // G5
  cfg.pin_d11 = GPIO_NUM_45; // R0
  cfg.pin_d12 = GPIO_NUM_48; // R1
  cfg.pin_d13 = GPIO_NUM_47; // R2
  cfg.pin_d14 = GPIO_NUM_21; // R3
  cfg.pin_d15 = GPIO_NUM_14; // R4
  cfg.pin_henable       = GPIO_NUM_40;
  cfg.pin_vsync         = GPIO_NUM_41;
  cfg.pin_hsync         = GPIO_NUM_39;
  cfg.pin_pclk          = GPIO_NUM_42;
  cfg.freq_write        = 9000000;   //16000000
  cfg.hsync_polarity    = 0; //0
  cfg.hsync_front_porch = 8; //8
  cfg.hsync_pulse_width = 4; //4
  cfg.hsync_back_porch  = 43; //43
  cfg.vsync_polarity    = 0;  //0
  cfg.vsync_front_porch = 8; //8
  cfg.vsync_pulse_width = 4; //4
  cfg.vsync_back_porch  = 12; //12
  cfg.pclk_idle_high    = 1; //1
  _bus_instance.config(cfg);
  _panel_instance.setBus(&_bus_instance);
  
  { auto cfg = _panel_instance.config();
  cfg.memory_width  = 480;
  cfg.memory_height = 272;
  cfg.panel_width   = 480;
  cfg.panel_height  = 272;
  cfg.offset_x      = 0;
  cfg.offset_y      = 0;
  _panel_instance.config(cfg);
  }
  
  { auto cfg = _panel_instance.config_detail();
  cfg.use_psram = 1;
  _panel_instance.config_detail(cfg);
  }

  { auto cfg = _light_instance.config();
  cfg.pin_bl = GPIO_NUM_2;
  _light_instance.config(cfg);
  }
  _panel_instance.light(&_light_instance);

{ 
      auto cfg = _touch_instance.config();
      cfg.x_min      = 100;    
      cfg.x_max      = 4000;   
      cfg.y_min      = 100;     
      cfg.y_max      = 4000;    
      cfg.pin_int    = -1;    
      cfg.bus_shared = true;  
      cfg.offset_rotation = 0;// 0-7
      cfg.spi_host = 1; // 
      cfg.freq = 1000000;    
      cfg.pin_sclk = 12;     
      cfg.pin_mosi = 11;     
      cfg.pin_miso = 13;     
      cfg.pin_cs   = 38;     
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  
    }
  setPanel(&_panel_instance); 
  }
};
static LGFX tft; 
```

ruihe1117, 2023-03-26:
@Emmanuele75 
Hi, I am trying to drive the RGB565 screen with ESP32-S3, but I am not familiar with the GFX library and do not know how to use the LVGL example, can you share a copy of your just fixed lvgl example code? I would like to learn an example of how gfx uses lvgl
thank you.

Emmanuele75, 2023-03-26:
@ruihe1117  
```ccp
#pragma GCC optimize ("Ofast")
#include <arduino.h>
#include <lvgl.h>
#define LGFX_USE_V1
#include <LovyanGFX.h>
#include "ui.h"

#include "lgfx_4827S043C.h"
//#include "lgfx_4827S043R.h"

/*Change to your screen resolution*/
static const uint32_t screenWidth  = 480;
static const uint32_t screenHeight = 272;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t disp_draw_buf[screenWidth * 10];

void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p){
  tft.pushImageDMA( area->x1
                  , area->y1
                  , area->x2 - area->x1 + 1
                  , area->y2 - area->y1 + 1
                  , ( lgfx::swap565_t* )&color_p->full );
  lv_disp_flush_ready(disp); 
}

/*Read the touchpad*/
void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data){
  int32_t x, y; 
  if (tft.getTouch(&x, &y)) {
      data->state = LV_INDEV_STATE_PR;
      data->point.x = x;
      data->point.y = y;
  }else{
    data->state = LV_INDEV_STATE_REL;
  }
}

void setup(){
  Serial.begin(115200);  
  tft.begin();        
  tft.setRotation(0);
  tft.setBrightness(255);  
  lv_init(); 

  lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, NULL, screenWidth * 10);
  static lv_disp_drv_t disp_drv;

/* Initialize the display */
  lv_disp_drv_init(&disp_drv);
  /* Change the following line to your display resolution */
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register(&disp_drv);

  /* Initialize the (dummy) input device driver */
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);

  ui_init(); 
}

void loop(){     
  lv_timer_handler();  
  delay(5); 
}
```

tobozo, 2023-03-26:
@ruihe1117  I've edited the email address (changed @ to 📧) in your post to avoid drawing attention from spambots

gamma2, 2023-03-26:
HI, i'm trying to test your code, but what include is ui.h(ui_init)? can you share a copy of ui.h?
Thanks

gamma2, 2023-03-26:
i just bought the esp32-4827S043 board and i am trying to test your code can you share the ui.h and ui.c files?

Thanks

Emmanuele75, 2023-03-26:
download squareline studio and export the sample file to your project folder and you're done.

ruihe1117, 2023-03-27:
@Emmanuele75 @lovyan03 

First of all, I really feel your help;

The screen I am using is an 800x480RGB screen. With the help of the above, I can already display it normally and have enabled "dual buffering full pixel refresh" (when not enabled, the display is blocked). "But my screen is only half displayed, everything else is normal, and I tested it on two screens with completely different hardware designs, and the results were only half displayed.".

Have you ever encountered a situation where only half of the display is displayed? Or are there any possible problems?
[image]

https://user-images.githubusercontent.com/116372394/227892671-fe117345-966c-4e71-8716-1a049b502c1b.mp4

Emmanuele75, 2023-03-27:
@ruihe1117 
I'll do some tests with 800x480 displays and then get back to you

Zer0-bit, 2023-04-01:
I tested and everything renders fine on the 800x480 LCD variant but effects are very sluggish, feels like it's an issue only with full page rendering though as localised transitions like scrolling and swiping elements seem fine.
@lovyan03 maybe you have an idea of why is this happening ?

Emmanuele75, 2023-04-01:
@Zer0-bit 
I too have experienced a lot of slowness with the 800x480 display, it seems that ESP32 is not up to par.

Zer0-bit, 2023-04-01:
@Emmanuele75 it's 100% not the mcu not being able to run smoother, the same LVGL code runs quite smooth and transitions look like transitions if using Arduino GFX.

Emmanuele75, 2023-04-01:
@Zer0-bit 
could you post the configuration for Arduino_GFX?

Zer0-bit, 2023-04-02:
@Emmanuele75 - i mean i can, not sure if it's not against certain rules around here ?

tobozo, 2023-04-02:
please share your working configurations, every GFX driver eventually benefits from the others, the more the merrier

Zer0-bit, 2023-04-02:
@tobozo sure no problems, i can add my project's working config later but just to mention the arduino gfx guys provide a working config by default for these displays on this link https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration
Look for the **ESP32-4827S043** and **ESP32-8048S070** displays in there.

Zer0-bit, 2023-04-03:
Okay so here's the behaviour using the standard widgets demo and LovyanGFX:

https://user-images.githubusercontent.com/42692077/229564555-ce2e7dab-11e6-4497-93ba-a0507e783b9d.mp4

And here's the same demo but running under Arduino GFX:

https://user-images.githubusercontent.com/42692077/229564718-0601d954-a89c-48c8-84a1-1718779a1376.mp4

gamma2, 2023-04-04:
God morning, i need help and i apologize if i'm not expert in this thing, but when i try to compile emmanuele75's example i get the following errors and the compilation aborts:

---------------------------------------------------------------------------------------------------------------------
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...

................................................................................................... other lines

src/ui/ui.c: In function 'Particle1_Animation':
src/ui/ui.c:189:5: warning: implicit declaration of function 'lv_anim_set_deleted_cb'; did you mean 'lv_anim_set_exec_cb'? [-Wimplicit-function-declaration]
     lv_anim_set_deleted_cb(&PropertyAnimation_0, _ui_anim_callback_free_user_data);
     ^~~~~~~~~~~~~~~~~~~~~~
     lv_anim_set_exec_cb

lib/Lvgl/src/core/lv_obj.c: In function 'lv_init':
lib/Lvgl/src/core/lv_obj.c:139:18: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     char * txt = "Á";
	            ^
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c: In function 'lv_tabview_add_tab':
Compiling .pio\build\esp32-s3-devkitc-1-n16r8v\libc67\Lvgl\font\lv_font_montserrat_12_subpx.c.o
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c:89:25: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         new_map[tab_id] = "";
                                     ^
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c:98:24: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[1] = "";
                                 ^
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c:101:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[tab_id * 2 - 3] = "\n";
                                                    ^
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c:103:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[tab_id * 2 - 1] = "";
                                                    ^
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c: In function 'lv_tabview_constructor':
lib/Lvgl/src/extra/widgets/tabview/lv_tabview.c:206:21: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     tabview->map[0] = "";
                                  ^
	 
lib/Lvgl/src/widgets/lv_checkbox.c: In function 'lv_checkbox_constructor':
lib/Lvgl/src/widgets/lv_checkbox.c:125:13: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     cb->txt = "Check box";
                  ^
-------------------------------------------------------------------------------------------------------
My platformio.ini is :
------------------------------------------------------
[env:esp32-s3-devkitc-1-n16r8v]
platform = espressif32  
board = esp32-s3-devkitc-1-n16r8v
framework = arduino

build_type = debug
build_flags = -D CONFIG_ESP32_SPIRAM_SUPPORT=y
			  -D CORE_DEBUG_LEVEL=5
monitor_speed = 115200
monitor_filters = log2file, esp32_exception_decoder, default
monitor_raw = yes
-----------------------------------------------------------------------------
Tanks in advance for the help

closed 2023-05-13 by github-actions[bot]