lovyan03/LovyanGFX #375: Transparency problem with pushImage

PM04290, 2023-03-24:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-DevKit]
- Panel Driver IC:    [ST7796]
- Bus type:           [SPI]
- LovyanGFX version:  [v0.4.17]
- FrameWork version:   [ArduinoESP32 v2.0.5]
- Build Environment :   [ArduinoIDE]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )

When using pushImage on background color not black, i can't have transparency color. background stay black.

### Expected Behavior ( 期待される動作 )

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1/ have memory icon (original png coded with image2cpp with 565 2 bytes)
```
constexpr unsigned short icon_volumedown [] PROGMEM = {
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 
  0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,
etc.

```
2/ have defined colors
```
#define COLOR_BLACK lcd.color565(0, 0, 0)
#define SEGMENT_ON  lcd.color565(64, 64, 255)
#define SEGMENT_OFF lcd.color565(10, 10, 10)
#define SEGMENT_SEL lcd.color565(255, 64, 64)
#define COLOR_ITEM_BACK lcd.color565(24, 24, 24)
#define COLOR_ITEM_TEXT lcd.color565(253, 126, 20)
#define COLOR_ITEM_WHITE lcd.color565(253, 253, 253)

```
### Code to reproduce this issue ( 再現させるためのコード )

```cpp
        item.createSprite(_w, _h);
        item.setSwapBytes(true);
        item.setFont(&font_XL);
        item.setTextDatum(lgfx::middle_center);
        item.setTextColor(COLOR_ITEM_TEXT);
        item.fillScreen(COLOR_BLACK);
        item.fillRoundRect(0, 0, _w, _h, 8, COLOR_ITEM_BACK);
        item.pushImage((_w / 2) - 24, (_h / 2) - 24, 48,   48, icon_volumedown , COLOR_ITEM_BACK);
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2023-03-24:
hi, 

this is normal behaviour, the transparency color refers the the `source` image

```cpp
item.pushImage((_w / 2) - 24, (_h / 2) - 24, 48,   48, icon_volumedown , COLOR_ITEM_BACK);
```

"any pixels with COLOR_ITEM_BACK will NOT be printed and considered as transparent."

tobozo, 2023-03-24:
> without set the tranparency color ?

but the transparency color is set:

```cpp
icons[a->img].pushRotateZoom(&sprites[flip], a->x, a->y - y, a->r, a->z, a->z, 0); 
```

the last argument `0` is the transparency color

PNG uses zlib, which alone needs 32Kb for its dictionary so it's already at its best in terms of memory consumption.

QIO format may be a better compromise, and even be smaller than PNG depending on the size and entropy of the images.

closed 2023-03-24 by PM04290