lovyan03/LovyanGFX #377: Random noise issue with MKS-TFT35R during power up.

astorun, 2023-03-26:
Hi,

I am using MKS-DLC32 esp32 board(board schematic is here: https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware) with MKS-TFT35R display. (schematics for display can be found here: https://github.com/makerbase-mks/MKS-TFT-Hardware/tree/master/MKS%20TS35-R/MKS%20TS35-R%20V2.0_002).

It works without any issue. Only problem is when tft.begin is called there is a colored noise on display that lasts for a few milliseconds until it is flushed.

It happens only when tft.begin is called. No other function seems to be affecting.

I tried several methods like setting the tft.setbrightness to zero during begin process on several places, changing the values of cfg instances, adding delays and setting the display pins to either low or high manually but nothing solves it.

It only happens if i power down the board and start it again. it does not happen when it is restarted with reset command or reset pin.

What might be the issue and i wonder if there is any way to eliminate it via proper code.

Here is the code i am using.

Thanks.

```
/////////////////////////////////////////////////////////////////
/*
  ESP32 | LVGL8 | Ep 0. GFX Setup (ft. LovyanGFX)
  Video Tutorial: https://youtu.be/IPCvQ4o_WP8
  Created by Eric N. (ThatProject)
*/
/////////////////////////////////////////////////////////////////

#include <lvgl.h>
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include "ui.h"

boolean tftBright = false;

class LGFX : public lgfx::LGFX_Device
{
lgfx::Panel_ST7796      _panel_instance;
lgfx::Bus_SPI           _bus_instance;   // SPI
lgfx::Light_PWM         _light_instance;
lgfx::Touch_XPT2046     _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = VSPI_HOST;     // (VSPI_HOST or HSPI_HOST)
      cfg.spi_mode = 0;             // SPI (0 ~ 3)
      cfg.freq_write = 80000000;    // SPI (80MHz)
      cfg.freq_read  = 16000000;    // 
      cfg.spi_3wire  = false;       //
      cfg.use_lock   = true;        //
      cfg.dma_channel = 1;          // Set the DMA channel (1 or 2. 0=disable)
      cfg.pin_sclk = 18;            // SPI SCLK
      cfg.pin_mosi = 23;            // SPI MOSI
      cfg.pin_miso = 19;            // SPI MISO (-1 = disable)
      cfg.pin_dc   = 33;            // SPI (-1 = disable) Data Command control pin
      _bus_instance.config(cfg);    //
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();
      cfg.pin_cs           =    25;  // CS    (-1 = disable)
      cfg.pin_rst          =    27;  // RST   (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSY  (-1 = disable)
      cfg.memory_width     =   320;
      cfg.memory_height    =   480;
      cfg.panel_width      =   320;
      cfg.panel_height     =   480;
      cfg.offset_x         =     0;
      cfg.offset_y         =     0;
      cfg.offset_rotation  =     0;
      cfg.dummy_read_pixel =     8;
      cfg.dummy_read_bits  =     1;
      cfg.readable         =  true;
      cfg.invert           = false;
      cfg.rgb_order        = false;
      cfg.dlen_16bit       = false;
      cfg.bus_shared       =  true;

      _panel_instance.config(cfg);
    }
    
    {
      auto cfg = _light_instance.config();

      cfg.pin_bl = 5;
      cfg.invert = true;
      cfg.freq   = 44100;
      cfg.pwm_channel = 7;

      _light_instance.setBrightness(0);
      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);
    }

    {
      auto cfg = _touch_instance.config();
      cfg.x_min      = 0;
      cfg.x_max      = 319;
      cfg.y_min      = 0;
      cfg.y_max      = 479;
      cfg.pin_int    = -1;      // INT pin number to which is connected
      cfg.bus_shared = true;    // Set to true when using a common bus with the screen
      cfg.offset_rotation = 0;  // Adjustment when the display and touch direction do not match Set with a value of 0 to 7
      cfg.spi_host = VSPI_HOST;
      cfg.freq = 1000000;       // SPI set clock
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 23;
      cfg.pin_miso = 19;
      cfg.pin_cs   = 26;
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }

    setPanel(&_panel_instance);
  }
};

LGFX tft;

//Change to your screen resolution
static const uint32_t screenWidth  = 480;
static const uint32_t screenHeight = 320;
static lv_disp_draw_buf_t draw_buf;

/*Static or global buffer(s). The second buffer is optional*/
static lv_color_t buf_1[screenWidth * 10];
//static lv_color_t buf_2[screenWidth * 10];

//Display flushing 
void disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
   uint32_t w = ( area->x2 - area->x1 + 1 );
   uint32_t h = ( area->y2 - area->y1 + 1 );

   tft.startWrite();
   tft.setAddrWindow( area->x1, area->y1, w, h );
   //tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
   tft.writePixels((lgfx::rgb565_t *)&color_p->full, w * h);
   tft.endWrite();

   lv_disp_flush_ready( disp );
}

//Read the touchpad
void touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
   uint16_t touchX, touchY;
   bool touched = tft.getTouch( &touchX, &touchY);
   if( !touched )
   {
      data->state = LV_INDEV_STATE_REL;
   }
   else
   {
      data->state = LV_INDEV_STATE_PR;

      //Set the coordinates
      data->point.x = touchX;
      data->point.y = touchY;

   }
}

void display_setup_init()
{
  tft.begin();
  tft.setBrightness(0);
  tft.setRotation(1);

  lv_init();
  uint16_t calData[] = { 3787, 3868, 3835, 200, 205, 3885, 168, 193};
  tft.setTouchCalibrate(calData);

  /*
  std::uint16_t fg = TFT_WHITE;
  std::uint16_t bg = TFT_BLACK;
  if (tft.isEPD()) std::swap(fg, bg);
  uint16_t calibrationData[8];
  tft.calibrateTouch(calibrationData, fg, bg, std::max(tft.width(), tft.height()) >> 3);

  for(int i=0; i<8; i++){
    Serial.print(calibrationData[i]);
    Serial.print(", ");
  }
  */

  delay(100);

  /*Initialize `disp_buf` with the buffer(s). With only one buffer use NULL instead buf_2 */
  lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, screenWidth * 10);

  //Initialize the display
  static lv_disp_drv_t disp_drv;
  lv_disp_drv_init( &disp_drv );

  //Change the following line to your display resolution
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register( &disp_drv );

  //Initialize the input device driver
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);                                 //Basic initialization
  indev_drv.type = LV_INDEV_TYPE_POINTER;                        //Touch pad is a pointer-like device
  indev_drv.read_cb = touchpad_read;                             //Set your driver function
  lv_indev_t * disp_indev = lv_indev_drv_register(&indev_drv);   //Register the driver in LVGL and save the created input device object

  delay(100);

  ui_init();
}

void displayInit()
{
    lv_timer_handler();
}
```

lovyan03, 2023-03-27:
In the ST7796, the contents of the frame buffer are filled with random noise immediately after the power is turned on.
This means that if the backlight is on before clearing the screen, noise will be visible.
It may be possible to avoid this problem by energizing the system with the backlight off.

astorun, 2023-03-27:
Any suggestions what can i try besides the solutions i already tried?

So far I already tried pulling the pin 5 low and high during setup with digitalWrite, setting the backlight "0" during tft.begin sequence but no luck on both configurations.

As soon as tft.begin starts it will turn on the backlight until those garbled noise appears and dissappears and afterwards it sets the brightness back to zero.

not sure if it is caused by the circuitry itself. strangely it only happens when machine is powered off and on but not between soft restarts.

lovyan03, 2023-03-27:
```
  tft.setBrightness(0);
  tft.begin();
```

astorun, 2023-03-27:
yeah i already tried that also. i put several of tft.setBrightness(0); between each code line but no effect.
not sure if it is a specific bug or hardware issue.

lovyan03, 2023-03-27:
By the way, I think GPIO5 is the pin that is pulled up at ESP32 startup.
I think GPIO2 and GPIO12 are pulled down at startup, so how about changing the pins used for backlighting?

astorun, 2023-03-27:
no effect.
i believe it is caused by board circuitry.

i will look into another solution for this.

thanks.

lovyan03, 2023-03-27:
What does that mean?
Do you see noise before you run tft.begin? If so, please close this Issue.
If the noise appears only at the moment of executing begin, I will look into the cause.

astorun, 2023-03-27:
I see noise when tft.begin executes. Before that there is no noise or any light. It only happens during tft.begin execution.
Also I cannot change or use other pins, it is how this board is designed.

https://i.ebayimg.com/images/g/cdsAAOSwJhBjQoYb/s-l1600.jpg

lovyan03, 2023-03-29:
Hello, @astorun 
The develop branch has been updated, please try it out.

Changes to the user code are probably not necessary.
I would expect the problem to go away without using the setBrightness function.

astorun, 2023-03-29:
Yep it fixed it!

Thank you so much!

closed 2023-03-30 by astorun