lovyan03/LovyanGFX #379: Save memory with mass storage cache

PM04290, 2023-03-28:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**
On application like dialog menu or other, we need very often the same image/text, and on large application we can't store it in cache (buffer) because of memory foot print.

**Is your feature request related to a problem? Please describe.**
I am allways in problem with heap memory on large application, so it's difficult to use PNG file who needs more than 47k memory.

**Describe the solution you'd like**
I have imagine a solution to ease the memory is to store Sprite on mass storage like SPIFFS, to re-use the same image at each time we need without memory usage (or less); I dream of a function that does this : 
sprite.dumpToDisk(SPIFFS,"filename.dump");
and the read one
sprite.createFromDump(SPIFFS,"filename.dump", width, height); // like createFromBmp

**Describe alternatives you've considered**
with my memory problem, i have consider many solutions, like using another image format, but i d'ont know what format use less memory.

**Additional context**
on U.I, the time to refresh screen is not very important, so we can use SPIFFS or SDcard to store/load image, and the time to do that is very low.

tobozo, 2023-03-28:
hi,

if you don't mind the bloat, feel free to vampirize the logic from [ESP32-Chimera-Core's ScreenShot Service](https://github.com/UT2UH/ESP32-Chimera-Core/tree/master/src/helpers/ScreenShotService), the `snapBMP()` function sounds like what you're looking for.

PM04290, 2023-03-28:
i don't want scpecifically a BMP format, just a memory dump off a Sprite, to re-use later.
i don't see anay Lovyan call in the project Chimere and  i prefer to use lovyan in production version.

PM04290, 2023-03-28:
@tobozo are you confirm that **readRectRGB** function read buffer of Sprite ant do a SPI reading panel ?

PM04290, 2023-03-28:
i have just tested readRectRBG() with a clone off Chimera SnapShoot BMP.... the BMP is totaly black.
i'ill try somting with getBuffer()

tobozo, 2023-03-28:
I made an example to [load/save sprite from filesystem or stream](https://github.com/tobozo/ESP32-Chimera-Core/tree/1.4.7/examples/Tests/spriteFS).

The saved file has more than a copy of the buffer, it also holds width/height/buffer_size so that the bit depth can be guessed when reloading the sprite, and the stream can loaded directly into the sprite buffer.

PM04290, 2023-03-29:
great, i'll test it today.

closed 2023-05-05 by github-actions[bot]