lovyan03/LovyanGFX #387: Smaller areas starting at x=0 are drawn cluttered

moinmoin-sh, 2023-04-13:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  WT32S3-86V / mcu esp32s3
- Panel Driver IC:    GC9503
- Bus type:           Parallel16
- LovyanGFX version: latest develop
- FrameWork version:  framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
- Build Environment :   PlatformIO
- Operating System:   Windows
## Problem Description ( 問題の内容 )
I have an issue with your latest develop branch.
I have an WT32S3-86V panel with a GC9503 LCD driver chip and an ESP32S3 mcu. 
Everything worked fine after I followed your instructions here https://github.com/lovyan03/LovyanGFX/issues/374#issuecomment-1478718157  
I run LVGL which means all display write access goes via 
```CPP
    tft.pushImageDMA( area->x1
                    , area->y1
                    , area->x2 - area->x1 + 1
                    , area->y2 - area->y1 + 1
                    , ( lgfx::swap565_t* )&color_p->full );
```
In my case areas smaller than the screen width are refreshed.  When an area starts at the left (x = 0) the output is cluttered, parts of the output is shifted to the right by the width of the area to be drawn. 
If the area starts further to the right then drawing this area is ok.
I searched for what was changed the last weeks and found this piece of code in `Panel_FrameBufferBase.cpp` at line `282` which was added.
```cpp
      if (sw == w && x == 0) {
        memcpy(_lines_buffer[y], src, w * h);
        cacheWriteBack(_lines_buffer[y], w * h);
        return;
      }
```
When I comment out these lines everything works fine again.

Without knowing too much about your lib I assume that this modification doesn't work if the drawn area starts at x=0 and is smaller than a whole line (just guessing)

### Expected Behavior ( 期待される動作 )

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

### Code to reproduce this issue ( 再現させるためのコード )

lovyan03, 2023-04-14:
@moinmoin-sh 
Thanks for letting us know about this Issue !

moinmoin-sh, 2023-04-14:
Works well now, thank's for taking care of it.

@lovyan03 
Do you have plans to cut a new realease in near future?

commit 41c7181e (2023-04-14): fix #387
closed 2023-04-27 by lovyan03