lovyan03/LovyanGFX #389: Issues with GPIO 0 button on TTGO_TDISPLAY

neilpas23, 2023-04-16:
I'm currently using LovyanGFX because its the only library that I can scale/flip sprites in for a game. My issue is that it seems like the GPIO-0 Button on the TTGO T-Display doesn't work while using this library? Is it possible that this library could be effecting that button in some way?

neilpas23, 2023-04-16:
I've trouble shot it down to the lcd.init(); function.
When you call lcd.init(); in setup gpio-0 gets shorted to LOW, it will only read out 0.

lovyan03, 2023-04-16:
I hate people who do not attach code to Issues.
Please provide proof first.

lovyan03, 2023-04-16:
One possibility is that you are using autodetect and have changed the GPIO settings prior to that.
You should set the pin mode later than lcd.init.

neilpas23, 2023-04-16:
#define LGFX_TTGO_TDISPLAY
#define LGFX_AUTODETECT
#include <Arduino.h>
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

static LGFX lcd;
int buttonVal;

void setup() {

    Serial.begin(115200);
    pinMode(0, INPUT_PULLUP);
    lcd.init();

}

void loop() {
  buttonVal = digitalRead(0);
    Serial.println(buttonVal);

}

If you comment out lcd.init(); button on gpio-0 goes back to working as normal
Sorry for not adding code.

lovyan03, 2023-04-16:
You should set the pin mode later than lcd.init.

neilpas23, 2023-04-16:
#define LGFX_TTGO_TDISPLAY
#define LGFX_AUTODETECT
#include <Arduino.h>
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

static LGFX lcd;
int buttonVal;

void setup() {

    Serial.begin(115200);
    lcd.init();
    pinMode(0, INPUT_PULLUP);

}

void loop() {
  buttonVal = digitalRead(0);
    Serial.println(buttonVal);

}

The result with this is the same, still all 0s

lovyan03, 2023-04-16:
Are you use latest version ?

lovyan03, 2023-04-16:
I have lost my T-Display and cannot verify.

neilpas23, 2023-04-16:
Just downloaded and replaced to 1.1.5 off the github instead of loading it thought the Arduino IDE. Reset and recompiled. Can confirm that this is still occurring. Maybe someone else on the GIT can verify if they own a T-display.

lovyan03, 2023-04-16:
umm,

please comment out this line...
```
 #define LGFX_AUTODETECT
```

lovyan03, 2023-04-16:
And, please set the debug log level to verbose and present the contents of the serial output.

neilpas23, 2023-04-16:
Using board 'ttgo-lora32-v1' from platform in folder: C:\Users\pasca\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\pasca\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Users\\pasca\\OneDrive\\Documents\\Arduino\\buttonTestScript\\partitions.csv" copy /y "C:\\Users\\pasca\\OneDrive\\Documents\\Arduino\\buttonTestScript\\partitions.csv" "C:\\Users\\pasca\\AppData\\Local\\Temp\\arduino\\sketches\\C3064446FEC9712FA85987DBDE1CD7A6\\partitions.csv"
cmd /c if not exist "C:\\Users\\pasca\\AppData\\Local\\Temp\\arduino\\sketches\\C3064446FEC9712FA85987DBDE1CD7A6\\partitions.csv" copy "C:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\pasca\\AppData\\Local\\Temp\\arduino\\sketches\\C3064446FEC9712FA85987DBDE1CD7A6\\partitions.csv"
Detecting libraries used...
"C:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/libsodium" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/log" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/lwip" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mbedtls" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mdns" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/micro-ecc" "-IC:\\Users\\pasca\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mqtt" "-IC:\\Use [... cut ...]
Writing at 0x00020000... (55 %)
Writing at 0x00024000... (66 %)
Writing at 0x00028000... (77 %)
Writing at 0x0002c000... (88 %)
Writing at 0x00030000... (100 %)
Wrote 306224 bytes (144993 compressed) at 0x00010000 in 2.7 seconds (effective 892.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 8191.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

lovyan03, 2023-04-16:
no, it's build log
I mean runtime output log.

neilpas23, 2023-04-16:
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0
23:42:38.348 -> 0

The Serial Monitor just updates 0s on loop because thats all thats printing.

lovyan03, 2023-04-16:
boot log...

lovyan03, 2023-04-16:
I mean this... 

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[     5][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[  3035][I][LGFX_AutoDetect_ESP32_all.hpp:1008] init_impl(): [LovyanGFX] [Autodetect] load from NVS : board:9
[  3036][V][LGFX_AutoDetect_ESP32_all.hpp:3350] autodetect(): [LGFX] pkg: 0
[  3041][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  3063][I][LGFX_AutoDetect_ESP32_all.hpp:2353] setup(): [LovyanGFX] [Autodetect] TTGO_TDisplay
```

neilpas23, 2023-04-16:
23:51:46.998 -> ets Jul 29 2019 12:21:46
23:51:46.999 -> 
23:51:46.999 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
23:51:46.999 -> configsip: 0, SPIWP:0xee
23:51:46.999 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:51:46.999 -> mode:DIO, clock div:1
23:51:46.999 -> load:0x3fff0018,len:4
23:51:46.999 -> load:0x3fff001c,len:1044
23:51:46.999 -> load:0x40078000,len:10124
23:51:46.999 -> load:0x40080400,len:5856
23:51:46.999 -> entry 0x400806a8

lovyan03, 2023-04-16:
ログレベルを上げてください

neilpas23, 2023-04-16:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[I][LGFX_AutoDetect_ESP32_all.hpp:1008] init_impl(): [Autodetect] load from NVS : board:9
[V][LGFX_AutoDetect_ESP32_all.hpp:3304] autodetect(): pkg: 0
[V][LGFX_AutoDetect_ESP32_all.hpp:714] _read_panel_id(): [Autodetect] read cmd:04 = ff528585
[I][LGFX_AutoDetect_ESP32_all.hpp:2308] setup(): [Autodetect] TTGO_TDisplay

lovyan03, 2023-04-16:
ArduinoESP32 v1.0.6を使っていますか？
もしそうであれば、v2.0.xに変更してみてください。

neilpas23, 2023-04-16:
I am using Arduino IDE 2.0.4. 
[image]

lovyan03, 2023-04-16:
no...  it's  IDE version...
I mean Arduino core for ESP32 ( board manager ) version

lovyan03, 2023-04-16:
I have found areas that may be affected and made adjustments.
You can grab the code from the develop branch and try it out.

neilpas23, 2023-04-16:
I though I had latest for Arduino ESP32 but I guess I didn't. Updated to 2.0.7 and I believe it works now. Should I still try the change you just made?

lovyan03, 2023-04-16:
If it has been resolved, that's fine.

closed 2023-04-16 by neilpas23