lovyan03/LovyanGFX #396: Touch bug on WT32-SC01 "Plus"

PM04290, 2023-05-06:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  WT32-SC01 **Plus**
- Panel Driver IC:    ST7796
- Bus type:           Parallel8
- LovyanGFX version:  1.1.16
- FrameWork version:   Arduino ESP32 v2.0.6
- Build Environment :   Arduino IDE 1.8.19
- Operating System:   Windows

## Problem Description ( 問題の内容 )
Touch not run.

Config 
```
        cfg.x_min      = 0;
        cfg.x_max      = 319;
        cfg.y_min      = 0;
        cfg.y_max      = 479;
        cfg.pin_int    = 7;
        cfg.bus_shared = true;
        cfg.offset_rotation = 0;

        cfg.i2c_port = 1;
        cfg.i2c_addr = 0x38;
        cfg.pin_sda  = 6;
        cfg.pin_scl  = 5;
        cfg.freq = 400000;

```

log displayed with Debug mode
[  6569][D][common.cpp:1131] writeBytes(): [LGFX] i2c write error : ack wait

PM04290, 2023-05-13:
The example 2_user_setting.ino run a sometimes with the previous configuration and generate a lot of i2d write error.
Is there any dedicated processors speed needed ? or just timing bug in library ?

nochkin, 2023-05-18:
I have the same exact config on my own WT32-SC01 Plus and it works great with no issues.
Try to make a simple I2C program without using LovyanGFX just to query the touch controller and see how it responds. This way you can rule out LovyanGFX lib and maybe check for other sources. That could be just a hardware issue.

PM04290, 2023-05-19:
after 2 tests on 2 differents modules which did not work, the first test was on a module where I had to have a program bug and the second was indeed defective (randomly). I just had a 3rd module loaned to me and it actually works well.
How unlucky.

closed 2023-05-19 by PM04290