lovyan03/LovyanGFX #401: -1 pin generates error

mmar22, 2023-05-24:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**
i use 
`lgfx::Bus_Parallel8  _bus_instance;`

## Environment ( 実行環境 )

- LovyanGFX version: lovyan03/LovyanGFX@^1.1.6
- FrameWork version:   PIO v2.0.7
- Build Environment :   PlatformIO

## Problem Description ( 問題の内容 )
class init produce actual behaviour
### Expected Behavior ( 期待される動作 )
no error
### Actual Behavior ( 実際の動作 )
E (625) gpio: gpio_set_direction(274): GPIO number error
E (905) gpio: gpio_set_direction(274): GPIO number error

### Steps to reproduce ( 再現のための前提条件 )
cfg.pin_rd =  -1;

cfg.pin_cs           =    -1;

lovyan03, 2023-05-24:
Not enough information at all.

mmar22, 2023-05-24:
How not enough , your lib code call gpio_set_direction with pin -1

lovyan03, 2023-05-24:
ah, ok. I see. thanks.

closed 2023-05-25 by lovyan03