lovyan03/LovyanGFX #402: Two displays pixels artifact

mmar22, 2023-05-25:
Hi i use two displays parallel8 switched with CS (one inverted)
My cb code
`/* Display flushing */
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
int16_t yl1;

  if(area->y1 >= screenHeight) {
//    lv_disp_flush_ready( disp );return;
     yl1 = area->y1 - screenHeight;
    digitalWrite(DSPS_CS, HIGH);
  }
  else {
    yl1 = area->y1;
    digitalWrite(DSPS_CS, LOW);
  }
  int16_t w = ( area->x2 - area->x1 + 1 );
  int16_t h = ( area->y2 - area->y1 + 1 );

    if (gfx.getStartCount() == 0)
    {   // Processing if not yet started
        gfx.startWrite();
    }
    gfx.pushImageDMA( area->x1
                    , yl1
                    , w
                    , h
                    , ( lgfx::swap565_t* )&color_p->full);
     lv_disp_flush_ready( disp );
}
`
both displays work . Image is 17FPS and good, but sometimes randomly some pixels from D2 apear on D1 or D1 on D2, exactly only line of random count 1-50px. 
This ocurs ofcourse when widgets changed on both displays.
I try wait for DMA or other , but seems nothing helps. I mean maybe this exist on one display too.

lovyan03, 2023-05-25:
you need gfx.endWrite()

closed 2023-05-25 by lovyan03

mmar22, 2023-05-25:
Ok try show example, with don’t lost offload effectivity