lovyan03/LovyanGFX #415: stack layers in certain way

Forairaaaaa, 2023-06-14:
Hi!
I'm trying to draw a simple `menu` with a `selector`.
I'm not sure if we have some APIs to `stack layers` in certain way, like "`difference`" in PS,
to make the `text` under the selector more clear, like this:
[image]

So, I try to [add "difference" in the fill rect api](https://github.com/Forairaaaaa/LovyanGFX/commit/a1aa196f02ca9dc0f7a6855d517e55b8d2cce72e) :

[image]
[image]
[image]

It sure meets my need so far, but the code becomes quite dirty.
I wander if there's a better way to achieve it?

Thanks!

tobozo, 2023-06-14:
Hi, thanks for your contribution :+1: 

Some thoughts:

- `readPixelRGB()` can be limited to sprites depending on the panel, `tft.isReadable()` is your friend for detecting that.
- `drawPixel` is the slowest function for drawing, storing pixels in one horizontal line buffer may be more efficient

Forairaaaaa, 2023-06-14:
> Hi, thanks for your contribution 👍
> 
> Some thoughts:
> 
> * `readPixelRGB()` can be limited to sprites depending on the panel, `tft.isReadable()` is your friend for detecting that.
> * `drawPixel` is the slowest function for drawing, storing pixels in one horizontal line buffer may be more efficient

Thanks! This cound be quite helpful

closed 2023-07-21 by github-actions[bot]