lovyan03/LovyanGFX #416: USB_CAMERA_LCD_DISPLAY

Nadeemranm, 2023-06-16:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-S3-WROOM|
- Panel Driver IC:    [ILI9486|
- Bus type:           Parallel8BIT
- LovyanGFX version:  [v1.4.17|latest|
- FrameWork version:   [ESP-IDF v5.0]
- Build Environment :   [VS CODE WITH ESPRESSIF -IDF]
- Operating System:   |Linux|

## Problem Description ( 問題の内容 )
LoyanGFX could not be found .this could be becouse : comonent name was misspelled , the component was not added to the build the component has been moved to the IDF component manger or has been removed and refactored into some other compenent  

### Expected Behavior ( 期待される動作
code run will be properly and start the live streaming on lcd display

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

closed 2023-07-23 by github-actions[bot]

Nadeemranm, 2023-07-25:
why are you closing this  issue !
This issue not resolved till now

Nadeemranm, 2023-07-25:
LovyanGFX library  and  uvc_stream  both library not compiling  together why!?

lovyan03, 2023-07-31:
https://github.com/lovyan03/LovyanGFX/issues/417#issuecomment-1628489624

あなたがCMakeLists.txtを正しく構成していないだけではないですか？
既に回答しているのでそれを試してください。

Nadeemranm, 2023-08-02:
> #417 (comment)

/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:278:5:
error: designator order for field 'uvc_config::frame_index' does not match
declaration order in 'uvc_config_t' {aka 'uvc_config'}
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:261:18:
warning: unused variable 'uvc_config' [-Wunused-variable]
  261 |     uvc_config_t uvc_config = {
      |                  ^~~~~~~~~~
What is the designator order for failed type error?

Nadeemranm, 2023-08-02:
cmake_minimum_required(VERSION 3.5)

#If IOT_SOLUTION_PATH is not defined, use relative path as default value
if(NOT DEFINED ENV{IOT_SOLUTION_PATH})
    get_filename_component(IOT_SOLUTION_PATH
"${CMAKE_SOURCE_DIR}/../../../.." ABSOLUTE)
    set(ENV{IOT_SOLUTION_PATH} ${IOT_SOLUTION_PATH})
    message(WARNING "Can't detect IOT_SOLUTION_PATH in your environment, we
infer it is $ENV{IOT_SOLUTION_PATH}")
endif()

# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi
and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}
/examples/common_components/protocol_examples_common
                            $ENV{IOT_SOLUTION_PATH}/components/bus
                            $ENV{IOT_SOLUTION_PATH}/components/display
                            $ENV{IOT_SOLUTION_PATH}
/components/usb/uvc_stream)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(uvc_mjpeg_lcd_lovyangfx)

Nadeemranm, 2023-08-04:
> idf_component_register(SRCS "main.cpp"
>     INCLUDE_DIRS "."
>     REQUIRES driver LovyanGFX-1.1.7 spiffs bus uvc_stream decoder_ijg
> )

Hello  sir!
Can u help me?