lovyan03/LovyanGFX #418: Tearing/flickering issue

FSDavideMaggioni, 2023-06-26:
We are currently building a product with a round LCD powered by an ESP32 clocked at 240MHz and we noticed a weird tearing/flicker issue when updating the display.

## Environment

- MCU or Board name:  ESP32
- Panel Driver IC:    GC9A01
- Bus type:           SPI - 80MHz
- LovyanGFX version:  latest
- Framework version:   ESP-IDF / Mongoose OS
- Build Environment :  VSCode
- Operating System:  macOS

## Problem Description

When using a framebuffer and drawing the whole display, tearing occurs. Our setup is as follow:
- We have a task that runs 10 times per second that draws a framebuffer to the display.
- The framebuffer is an LGFX_Sprite with the same size of the display itself but the color depth reduced to 8 bit to same some memory.
- The actual updating of the framebuffer occurs atomically before the draw to the actual lcd.

### Expected Behavior

The display should be updated all at the same time.

### Actual Behavior

Flickering/tearing.

### Steps to reproduce

Reproducing the issue should be pretty easy, the minimal reproducible example shows the issue by just painting the `fb` white and black repeatedly.
With the GC9A01 display running at 80MHz SPI, create a framebuffer with bit depth of 8 bits, fill it white, draw on the screen, and then black and repeat.

What is weird to me is that, at that SPI speed, this should not happen. I tired to dig into the inner workings of the library to find if it was a batching or transaction issue but other than doubling `WRITEPIXELS_MAXLEN` in `Panel_LCD.cpp` or increasing the limit in `void Bus_SPI::writePixels(pixelcopy_t* param, uint32_t length)` is `Bus_SPI.cpp` I found nothing helped.

https://github.com/lovyan03/LovyanGFX/assets/107125764/cb3ed8b5-cc44-4430-92ab-a3f31611983c

### Pseudocode

Clear the `fb`
Draw something on the `fb`.
Draw the actual fb to the lcd via `fb.pushSprite(&lcd, 0, 0)`
Wait the required time and repeat.

I'm sorry to not include an actual reproducible code, if any clarification is needed I'll be happy to provide it! Thank you so much for the wonderful library, we are looking forward to sponsor it! 😄

lovyan03, 2023-07-10:
First, if the colorDepth of the screen and the colorDepth of the frame buffer do not match, transmission is performed while converting the colorDepth.
Communication is performed at 80 MHz, but the average speed should be slow because of repeated transmission and interruptions.

Preventing tearing is very difficult.
Tearing usually occurs because drawing to the screen and panel refresh are done asynchronously.

If you can read from the LCD, try using the getScanLine() function to get the current scan line position.
You may be able to control the tearing position by waiting for the scan line position to return to the top of the screen before starting to draw.

FSDavideMaggioni, 2023-07-10:
Thank you for the reply @lovyan03!

I tried the function `getScanLine` as you suggested but it seems like it always returns `0`. More on that, it creates some weird color glitches on the display. Maybe I'm doing something wrong?

lovyan03, 2023-07-10:
Perhaps the pin is not configured correctly.
I do not know your config.