lovyan03/LovyanGFX #425: usb_camera_lcd

nadeem268, 2023-07-13:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-S3-DevKitC-1]
- Panel Driver IC:    [ILI9488|
- Bus type:           Parallel8|
- LovyanGFX version:  [latest develop|]
- FrameWork version:   [ESP-IDF v5.0.0]
- Build Environment :   [other]
- Operating System:   [Linux]

## Problem Description ( 問題の内容 )

### Expected Behavior ( 期待される動作 )
LovyanGFX library not compiling

/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_wifi_transfer/components/LovyanGFX/src/lgfx/v1/platforms/esp32/common.hpp:28:10: fatal error: driver/gpio.h: No such file or directory
   28 | #include <driver/gpio.h>
      |          ^~~~~~~~~~~~~~~

### Actual Behavior ( 実際の動作 )
#define LGFX_USE_V1
#include "LovyanGFX.hpp"
### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
 tft.init();
    tft.clear(TFT_WHITE);
    tft.setBrightness(255);
    tft.setRotation(2);
    tft.setFreeFont(&Orbitron_Light_24);
    tft.setTextColor(0x0000FFU, 0xFFFFFF);
    tft.drawString("UVC Stream To LCD", 10, 10);
    tft.setTextColor(0x000000U, 0xFFFFFF);
    tft.drawString("Resolution: 320x240", 10, tft.height()/2 + 60);
    tft.drawString("Camera: USB 2.0", 10, tft.height() / 2 + 90);
    tft.drawString("ESP32-S2", 10, tft.height() / 2 + 120);
    tft.drawString("USB 1.1: 12Mbps", 10, tft.height() / 2 + 150);
    tft.setFont(&FreeSansBoldOblique24pt7b);
    tft.setTextColor(0xFF0000U, 0xFFFFFF);
    //tft.setTextDatum(bottom_left);

    uint8_t *lcd_buffer = (uint8_t *)heap_caps_malloc(DEMO_SPI_MAX_TRANFER_SIZE, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    assert(lcd_buffer != NULL);

    /* malloc double buffer for usb payload, xfer_buffer_size >= frame_buffer_size*/
    uint8_t *xfer_buffer_a = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(xfer_buffer_a != NULL);
    uint8_t *xfer_buffer_b = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(xfer_buffer_b != NULL);

    /* malloc frame buffer for a jpeg frame*/
    uint8_t *frame_buffer = (uint8_t *)_malloc(DEMO_XFER_BUFFER_SIZE);
    assert(frame_buffer != NULL);

    /* the quick demo skip the standard get descriptors process,
    users need to get params from camera descriptors,
    run the example first to printf the log with descriptors */
    uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL,
        //.xfer_type = DEMO_XFER_MODE,
        .configuration = DESCRIPTOR_CONFIGURATION_INDEX,
        .format_index = DESCRIPTOR_FORMAT_MJPEG_INDEX,
        .frame_width = DEMO_FRAME_WIDTH,
        .frame_height = DEMO_FRAME_HEIGHT,
        .frame_index = DEMO_FRAME_INDEX,
        .frame_interval = DEMO_FRAME_INTERVAL,
        .interface = DESCRIPTOR_STREAM_INTERFACE_INDEX,
        .interface_alt = DEMO_ISOC_INTERFACE_ALT,
        .isoc_ep_addr = DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR,
        .isoc_ep_mps = DEMO_ISOC_EP_MPS,
        .xfer_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .xfer_buffer_a = xfer_buffer_a,
        .xfer_buffer_b = xfer_buffer_b,
        .frame_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .frame_buffer = frame_buffer,
    };

    /* pre-config UVC driver with params from known USB Camera Descriptors*/
    esp_err_t ret = uvc_streaming_config(&uvc_config);

    /* Start camera IN stream with pre-configs, uvc driver will create multi-tasks internal
    to handle usb data from different pipes, and user's callback will be called after new frame ready. */
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "uvc streaming config failed");
    } else {
        uvc_streaming_start(frame_cb, NULL);
    }

    while (1) {
        /* task monitor code if necessary */
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

Nadeemranm, 2023-07-25:
??

lovyan03, 2023-08-04:
簡単なサンプルコードから順番に試していくことをお勧めします。
使い方を知らない状態で同時に複数の事をやろうとすると混乱するだけです。

それから、同様のIssueを何個も作成するのはやめて貰えますか。

closed 2023-08-04 by lovyan03