lovyan03/LovyanGFX #431: Read touch screen in an interrupt

elengr, 2023-07-28:
**Read touch screen in an interrupt**

- MCU or Board name: Esp32 WT32-SC01
- LovyanGFX version:  v 1.1.7
- FrameWork version:   Arduino 2.1.1
- Build Environment :   ArduinoIDE
- Operating System:   Windows

Hey there I hope you are having a great day, 
I am working on a project with the LovyanGFX library. After finishing the project we want to add interrupts to our code.
My issue is that I need to read the x and y coordinates for touch.
I notice that lcd.getTouch(&x, &y); is frequently used. However, I cannot use this command within an interrupt because, as far as I can tell, it is an interrupt.
Do you have any suggestions on what I should do?

lovyan03, 2023-07-29:
In general, you should set some flag in the interrupt function and terminate processing immediately.
The actual communication should be done in the main loop. You can do getTouch when the flag is set.

This is the job of the user code, not the GFX.

lovyan03, 2023-07-29:
I mentioned flags earlier, but the recommended method for ESP32 would be to use the FreeRTOS functionality to handle notifications between tasks.

elengr, 2023-07-31:
Thanks for your advise :) Have a nice day.

closed 2023-08-04 by lovyan03