lovyan03/LovyanGFX #432: Use a .otf file as a set of icons

jeanvr-iff, 2023-07-31:
Hi, i'm trying to use a .otf font file from FontAwesome to render icons on screen. I've converted this file using Adafruit's "fontconvert" program to obtain an array that i can use on "setFont" function. The problem is that i only get an empty rectangle on screen, meaning that the character was not able to be rendered correctly.  The code is shown below:

```
#define LGFX_WT32_SC01
#define LGFX_USE_V1

#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

#include "./FA6_Regular.h"

LGFX lcd;

auto icon = u8"\uf015";
int x_pos, y_pos;

void setup() 
{
    lcd.init();
    lcd.setRotation(2);
    lcd.setBrightness(128);
    lcd.fillScreen(TFT_BLACK);
    lcd.setFont(&Font_Awesome_6_Brands_Regular_40012pt7b);
    lcd.setColor(TFT_WHITE);

    x_pos = (lcd.width() - lcd.textWidth(icon))/2;
    y_pos = (lcd.height() - lcd.fontHeight())/2;
    
    lcd.setCursor(x_pos, y_pos);
    lcd.print(icon);
}

void loop() 
{

}
```
NOTE: "f015" is the unicode shown here: https://fontawesome.com/icons/house?f=classic&s=solid

Here is the object exported on "FA6_Regular.h":
```
const GFXfont Font_Awesome_6_Brands_Regular_40012pt7b PROGMEM = {
  (uint8_t  *)Font_Awesome_6_Brands_Regular_40012pt7bBitmaps,
  (GFXglyph *)Font_Awesome_6_Brands_Regular_40012pt7bGlyphs,
  0x20, 0x7E, 24 };
``

lovyan03, 2023-08-01:
検証するためのデータを提供してください

jeanvr-iff, 2023-08-01:
Thanks for your attention. Here are the files i'm using.
[Test_Icon_Font.zip](https://github.com/lovyan03/LovyanGFX/files/12232511/Test_Icon_Font.zip)

lovyan03, 2023-08-05:
こちらのツールが使えるかも知れません。

https://github.com/m5stack/FontCreator

jeanvr-iff, 2023-08-24:
Thanks for the tips. I've got this to work using Remix Icon's font file using either fontconvert or FontCreator.

closed 2023-08-24 by jeanvr-iff