lovyan03/LovyanGFX #434: ESP32-S3-N16R8 reset loop when connecting to WiFi if LCD is initialized first

benlye, 2023-08-01:
## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3-DevKit C N16R8
- Panel Driver IC:    ILI9488
- Bus type:           [SPI|I2C|Parallel8|Parallel16|other]
- LovyanGFX version:  1.1.7
- FrameWork version:   ArduinoESP32 v2.0.11
- Build Environment :   ArduinoIDE / PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )
I am trying to use a simple 4.3" ILI9488 + XPT2046 touchscreen with an ESP32-S3-N16R8 dev board.  I have had this setup working with different libraries (TFT_eSPI) and the screen working with an ESP32.

To port my code from the ESP32 to the ESP32-S3 I had to change `cfg.spi_host` from `VHOST` (on the ESP32) to `SPI2_HOST` or `SPI3_HOST` for the ESP32-S3.

When I try to use `cfg.spi_host = SPI3_HOST` the touch panel does not work at all.  If I use `cfg.spi_host = SPI2_HOST` the board resets when it connects to WiFi if the LCD was initialized first, but works fine if the WiFi connection comes before the LCD initialization.

I want to initialize the LCD first so that I can show WiFi connection progress on the display.

### Expected Behavior ( 期待される動作 )
It should not matter what order the LCD initialization and WiFi connection are performed.

### Actual Behavior ( 実際の動作 )
Initializing the LCD before connecting to WiFi causes the board to reset when WiFi is started.  Connecting to WiFi and then initializing the LCD works fine.

### Steps to reproduce ( 再現のための前提条件 )

1. Create a simple sketch which will initialize the LCD before connecting to WiFi
2. Run the sketch on an ESP32-S3 (N16R8)

// If possible, attach a picture of your setup/wiring here.
| LCD Pin | ESP32S3 GPIO |
| - | - |
| DC/RS | GPIO6 |
| CS | GPIO10 |
| SCK | GPIO12 |
| SDI (MOSI) | GPIO11 |
| SDO (MISO)| N/C |
| LED | GPIO7 |
| RST | GPIO5 |
| T_CLK | GPIO12 |
| T_DO | GPIO13 |
| T_DIN | GPIO11 |
| T_CS | GPIO15 |
| T_IRQ | GPIO16 (but disabled in pin assignments)| 

### Code to reproduce this issue ( 再現させるためのコード )
https://gist.github.com/benlye/33c3feaa97e412a661fec82803874c8b

benlye, 2023-08-01:
To be clear, I would also be happy to have the XPT2046 touch working with `cfg.spi_host = SPI3_HOST`, if that's possible.

benlye, 2023-08-01:
I found my mistake. 

I was defining `cfg.pin_miso` as `-1` when it needed to be defined as the `TOUCH_MISO` (`T_DO`) pin.  

With that done I am able to use either `cfg.spi_host = SPI3_HOST` or `cfg.spi_host = SPI3_HOST` freely with fully working WiFi and touch.

closed 2023-08-01 by benlye

FatherMarco1971, 2024-07-19:
@benlye i'm trying to use the Es32-S3-N16R8 with ili9431 XPT2046 and SD too since on the display there is a reader.
If i init display and touch only they work perfectly (shared SPI).
If i enable sd only on a different SPI it work perfectly, no way to use alla togheter.
Have you a piece of code use the board loyvagfx libray and all the 3 peripheral?

benlye, 2024-07-19:
> @benlye i'm trying to use the Es32-S3-N16R8 with ili9431 XPT2046 and SD too since on the display there is a reader. If i init display and touch only they work perfectly (shared SPI). If i enable sd [...]

I have not tried to use SD, sorry.

FatherMarco1971, 2024-07-19:
This thing is driving me crazy...
Thank you anyway.