lovyan03/LovyanGFX #439: LILYGO® T-HMI ESP32-S3 Touch Display 2.8 inch ST7789

taoloo897, 2023-08-16:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

### Device Name (デバイスの名称・型番等)
LILYGO® T-HMI ESP32-S3 Touch Display 2.8 inch ST7789 

### URL of Device Specifications document (仕様書等のURL)
https://github.com/Xinyuan-LilyGO/T-HMI

### URL of the store where we can purchase (商品を購入できるURL)
https://vi.aliexpress.com/item/1005005187357655.html?spm=a2g0o.order_list.order_list_main.5.71921802yxoX0U&gatewayAdapt=glo2vnm

tobozo, 2023-08-16:
hi, 

sorry if this code is untested but it should be a good start as a profile for T-HMI

pins configuration was grabbed from [here](https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/examples/touch/pins.h) and [here](https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/lib/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h) as you suggested.

```C++

class LGFX : public lgfx::LGFX_Device {

  lgfx::Panel_ST7789 _panel_instance;
  lgfx::Bus_Parallel8 _bus_instance;
  lgfx::Light_PWM   _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

  public:

    LGFX(void)
    {
      {
        auto cfg = _bus_instance.config();
        cfg.freq_write = 30000000; // used 16MHz and 20MHz during testing
        cfg.pin_wr = 8;
        cfg.pin_rd = -1;
        cfg.pin_rs = 7;

        cfg.pin_d0 = 48;  // D0
        cfg.pin_d1 = 47;  // D1
        cfg.pin_d2 = 39;  // D2
        cfg.pin_d3 = 40;  // D3
        cfg.pin_d4 = 41;  // D4
        cfg.pin_d5 = 42; // D5
        cfg.pin_d6 = 45;  // D6
        cfg.pin_d7 = 46; // D7
        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();
        cfg.pin_cs   = 6;
        cfg.pin_rst  = -1;
        cfg.pin_busy = -1;

        cfg.panel_width  = 240;
        cfg.panel_height = 320;
        
        cfg.readable   = true;
        cfg.invert     = false;
        cfg.rgb_order  = false;
        cfg.dlen_16bit = false;
        cfg.bus_shared = true; // SD = true
        _panel_instance.config(cfg);
      }
      setPanel(&_panel_instance);

      {
        auto cfg = _light_instance.config();
        cfg.pin_bl = 38;
        _light_instance.config(cfg);
      }
      _panel_instance.light(&_light_instance);

      {
        auto cfg = _touch_instance.config();
        cfg.spi_mosi   = 3;
        cfg.spi_miso   = 4;
        cfg.spi_sclk   = 1;
        cfg.spi_cs     = 2;
        cfg.pin_int    = 9;
        cfg.spi_host   = VSPI_HOST;
        cfg.bus_shared = false;
        cfg.freq       = 2500000;
      }
      touch(&_touch_instance);

    }

};

```

taoloo897, 2023-08-17:
Hi and Thanks.
I made a new LGFX_ESP32S3_T-HMI.h in libraries\LovyanGFX\src\lgfx_user, but after please tell me how can i use.

tobozo, 2023-08-17:
see the [user config example](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino), there's a lot of code but it's straightforward:

```cpp

#include <LovyanGFX.hpp>

// use the config snippet inline **or** #include "lgfx_user/LGFX_ESP32S3_T-HMI.h" 

// class LGFX : public lgfx::LGFX_Device
// {
//  (.....)
// }

// create the display object
LGFX display;

void setup(void)
{
  // init the display
  display.init();
}
```

taoloo897, 2023-08-17:
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino: In constructor 'LGFX::LGFX()':
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:61:13: error: 'struct lgfx::v1::ITouch::config_t' has no member named 'spi_mosi'
         cfg.spi_mosi   = 3;
             ^~~~~~~~
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:62:13: error: 'struct lgfx::v1::ITouch::config_t' has no member named 'spi_miso'
         cfg.spi_miso   = 4;
             ^~~~~~~~
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:63:13: error: 'struct lgfx::v1::ITouch::config_t' has no member named 'spi_sclk'
         cfg.spi_sclk   = 1;
             ^~~~~~~~
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:64:13: error: 'struct lgfx::v1::ITouch::config_t' has no member named 'spi_cs'; did you mean 'pin_cs'?
         cfg.spi_cs     = 2;
             ^~~~~~
             pin_cs
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:66:26: error: 'VSPI_HOST' was not declared in this scope
         cfg.spi_host   = VSPI_HOST;
                          ^~~~~~~~~
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:66:26: note: suggested alternative: 'SPI3_HOST'
         cfg.spi_host   = VSPI_HOST;
                          ^~~~~~~~~
                          SPI3_HOST
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:70:29: error: no matching function for call to 'LGFX::touch(lgfx::v1::Touch_XPT2046*)'
       touch(&_touch_instance);
                             ^
In file included from c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1_init.hpp:25,
                 from c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/LovyanGFX.hpp:31,
                 from C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-3uge2g.mhj84\2_user_setting\2_user_setting.ino:2:
c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1/LGFXBase.hpp:1303:20: note: candidate: 'lgfx::v1::ITouch* lgfx::v1::LGFX_Device::touch() const'
     inline ITouch* touch(void) const { return _panel ? panel()->touch() : nullptr; }
                    ^~~~~
c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1/LGFXBase.hpp:1303:20: note:   candidate expects 0 arguments, 1 provided

exit status 1

Compilation error: 'struct lgfx::v1::ITouch::config_t' has no member named 'spi_mosi'

tobozo, 2023-08-17:
I'm sorry I can't test code for the meantime so I obviously created some problems in the code snippet 😊 

The mistake here was the properties prefixed with 'spi_' whereas it should be 'pin_', also the error message was hinting at this, so you should also read it before pasting it and expecting assistance.

```cpp
      {
        auto cfg = _touch_instance.config();
        cfg.pin_mosi   = 3;
        cfg.pin_miso   = 4;
        cfg.pin_sclk   = 1;
        cfg.pin_cs     = 2;
        cfg.pin_int    = 9;
        // cfg.spi_host   = SPI3_HOST; // try to uncomment this after the display init test succeeds
        cfg.bus_shared = false;
        cfg.freq       = 2500000;
      }
      touch(&_touch_instance);
```

taoloo897, 2023-08-17:
thanks again

C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-1hyif8u.cyon\2_user_setting\2_user_setting.ino: In constructor 'LGFX::LGFX()':
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-1hyif8u.cyon\2_user_setting\2_user_setting.ino:70:29: error: no matching function for call to 'LGFX::touch(lgfx::v1::Touch_XPT2046*)'
       touch(&_touch_instance);
                             ^
In file included from c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1_init.hpp:25,
                 from c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/LovyanGFX.hpp:31,
                 from C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2023717-49988-1hyif8u.cyon\2_user_setting\2_user_setting.ino:2:
c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1/LGFXBase.hpp:1303:20: note: candidate: 'lgfx::v1::ITouch* lgfx::v1::LGFX_Device::touch() const'
     inline ITouch* touch(void) const { return _panel ? panel()->touch() : nullptr; }
                    ^~~~~
c:\Users\taolo\Documents\Arduino\libraries\LovyanGFX\src/lgfx/v1/LGFXBase.hpp:1303:20: note:   candidate expects 0 arguments, 1 provided

exit status 1

Compilation error: no matching function for call to 'LGFX::touch(lgfx::v1::Touch_XPT2046*)'

[2_user_setting.zip](https://github.com/lovyan03/LovyanGFX/files/12368278/2_user_setting.zip)

tobozo, 2023-08-19:
more mistakes fixed by looking at this [config sample](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/LGFX_ESP32_sample.hpp):

```cpp
class LGFX : public lgfx::LGFX_Device {

  lgfx::Panel_ST7789 _panel_instance;
  lgfx::Bus_Parallel8 _bus_instance;
  lgfx::Light_PWM   _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

  public:

    LGFX(void)
    {
      {
        auto cfg = _bus_instance.config();
        cfg.freq_write = 30000000; // used 16MHz and 20MHz during testing
        cfg.pin_wr = 8;
        cfg.pin_rd = -1;
        cfg.pin_rs = 7;

        cfg.pin_d0 = 48;  // D0
        cfg.pin_d1 = 47;  // D1
        cfg.pin_d2 = 39;  // D2
        cfg.pin_d3 = 40;  // D3
        cfg.pin_d4 = 41;  // D4
        cfg.pin_d5 = 42; // D5
        cfg.pin_d6 = 45;  // D6
        cfg.pin_d7 = 46; // D7
        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();
        cfg.pin_cs   = 6;
        cfg.pin_rst  = -1;
        cfg.pin_busy = -1;

        cfg.panel_width  = 240;
        cfg.panel_height = 320;
        
        cfg.readable   = true;
        cfg.invert     = false;
        cfg.rgb_order  = false;
        cfg.dlen_16bit = false;
        cfg.bus_shared = true; // SD = true
        _panel_instance.config(cfg);
      }

      {
        auto cfg = _light_instance.config();
        cfg.pin_bl = 38;
        _light_instance.config(cfg);
        _panel_instance.setLight(&_light_instance);
      }

      {
        auto cfg = _touch_instance.config();
        cfg.pin_mosi   = 3;
        cfg.pin_miso   = 4;
        cfg.pin_sclk   = 1;
        cfg.pin_cs     = 2;
        cfg.pin_int    = 9;
        cfg.spi_host   = SPI3_HOST; // try to uncomment this after the display init test succeeds
        cfg.bus_shared = false;
        cfg.freq       = 2500000;
        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
      }

      setPanel(&_panel_instance);

    }

};

```

taoloo897, 2023-08-19:
Thanks again for all your time.
The code compiled and uploaded ok.
I made some define and setup because lcd not opened , but with all of this nothing show on the lcd.
#define PWR_EN_PIN  (10)

in setup()
pinMode(PWR_EN_PIN, OUTPUT);
  digitalWrite(PWR_EN_PIN, HIGH);

tobozo, 2023-08-19:
- ` cfg.freq_write = 30000000;` use a lower value, it's even okay to start with 2MHz (2000000)
- use `Serial.begin(115200)` in your setup()
- set Debug Output Level to "debug", flash the sketch and look in the serial console for any message
- verify pins numbers, I may have made some mistakes there too :-)

taoloo897, 2023-08-19:
thanks
this was a problem  cfg.freq_write = 30000000; use a lower value, it's even okay to start with 2MHz (2000000)
i made it 
cfg.freq_write = 20000000
but must remaining
#define PWR_EN_PIN (10)

in setup()
pinMode(PWR_EN_PIN, OUTPUT);
digitalWrite(PWR_EN_PIN, HIGH);

i will try in my project to see if it work faster than vendors library , there i have a big problem when i read on canbus at 500000 it will make white screen if not put a little delay (2) .

tobozo, 2023-08-19:
I'm not sure what this PWR_EN_PIN does, commenting it out and doing a full power cycle will provide the answer :-)

> big problem when i read on canbus

Try using `tft.endWrite()` before accessing the CAN bus, and `tft.startWrite()` before drawing on the display.

Otherwise there's a config setting `cfg.bus_shared = false;` both for panel and touch, maybe changing the values will show improvements?

taoloo897, 2023-08-19:
I have a big problem with this , canbus and display inreal time.
I get this esp s3 lilygo because first i tried esp32 wroomer 32u wioth nextion, i have the same problem, with delay
If i put delay i skip some canbus id.s if not.,.....the diusp[lay works bad.
I have other 2 protocols LINbus and kline.....it is a boooooomb. Not work all in the same time.

tobozo, 2023-08-19:
maybe use an atmega for the CAN bus and set it up as a slave I2C with the ESP ?

otherwise you can play with freeRTOS tasks and use mutexes/semaphores to prevent concurrent access to the SPI bus but that may affect performances

can you please confirm that the LGFX configuration works with the LilyGo T-HMI so this issue can be closed?

taoloo897, 2023-08-19:
yes with your last example lgfx works ok with lilygo 2.8 t-hmi
cfg.freq_write = 20000000
but must remaining
#define PWR_EN_PIN (10)
in setup()
pinMode(PWR_EN_PIN, OUTPUT);
digitalWrite(PWR_EN_PIN, HIGH);
If you can update the library will be awsome!
THANKS a lot!

tobozo, 2023-08-19:
thanks for confirming

The LGFX configuration works externally and the [PWR_EN_PIN is not display-related](https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/examples/adc/adc.ino) (it seems to be for the battery) so there is no reason to update LovyanGFX.

closing this as solved, have fun !

closed 2023-08-19 by tobozo

taoloo897, 2023-08-19:
Can you help me a little more , when i try to display a string , number....
with drawString , prin , printf , but what is show is write over the old text.

tobozo, 2023-08-19:
the second argument of setTextColor() is the background color e.g. `lcd.setTextColor(0x00FFFFU, 0xFF0000U);`

when only one argument is provided to setTextColor(), background color is assumed to be transparent

taoloo897, 2023-08-20:
THANKS A LOT! Incredible job from your part.