lovyan03/LovyanGFX #448: CST816s

Jswood33, 2023-09-23:
I can't find this touch driver under the lfgx_user sample even though it says it's supported.  Do I need to set it custom under v1 or something? Appreciate the help.

Jswood33, 2023-09-25:
Figured it out, sorry for the bother.

Jswood33, 2023-09-26:
Trying to run some simple code kept causing a Guru Meditation Error: storeprohibited.  
[image]
[image]

this is altered lgfx_user file where set the CST816S as my touch instance.
[LGFX_ESP32_test.txt](https://github.com/lovyan03/LovyanGFX/files/12721989/LGFX_ESP32_test.txt)

When using lcd.getTouch() I get the storeprohibited error a bunch, but the program still works and I receive my touch data printed on screen like intended.  It just resets the program and doesn't stop resetting seemingly at random time intervals.

When using lcd.getTouchRaw() the storeprohibited error went away but was replaced by receiving (-01, -01) again seemingly at random time intervals.  Just like before the data still reads out correctly and I'm able to track my way around the touchpad, but I'm very positive I've done something wrong, maybe a variable I've missed, or maybe this was why CST816S wasn't originally on the lgfx_user template? 

Thanks for anyone who looks at this and helps!

lovyan03, 2023-09-27:
Thanks for letting me know. I will add a line for CST816S to the document.
In the meantime, please try the develop branch version and see.

If you still think there is a problem, please let us know the details of your environment.

Jswood33, 2023-09-27:
From preliminary tests, the dev branch seems to be working great! I'm unsure what was done to fix it, but I appreciate the speedy response and fix! will close this issue now, thank you so much.

closed 2023-09-27 by Jswood33