lovyan03/LovyanGFX #454: Possible improving pushImageRotateZoom() render

ESPboy-edu, 2023-10-07:
## Environment ( 実行環境 )

- MCU or Board name:  ESPboy/ESP8266
- Panel Driver IC:    ST7735
- Bus type:           SPI
- LovyanGFX version:  v1.1.9
- FrameWork version:   Arduino LOLIN/WeMos D1/R2 and mini
- Build Environment:   ArduinoIDE
- Operating System:   macOS

## Problem Description ( 問題の内容 )
Function: 
```cpp
void pushImageRotateZoom(float dst_x, float dst_y, float src_x, float src_y, float angle, float zoom_x, float zoom_y, int32_t w, int32_t h, const void* data, color_depth_t depth, const T* palette);
```
with parameter "color_depth_t depth" = "lgfx::color_depth_t::palette_4bit"
Stops render all image if rendered coordinates "float dst_x, float dst_y" of the image are less than 0 coordinate of the display and it is not good.
Anyway if we push the image beyond the maximum coordinates of the display, we still see part of the image until it is completely outside the display and it is fine.

### Expected Behavior ( 期待される動作 )
"pushImageRotateZoom()" render part of the pushed image which is still in the display's boundaries even if X or Y coordinates of pushed picture less then display's X=0 or Y=0 the same as we see image rendering near the maximum displays coordinates (when the part of image is out of the display).

### Actual Behavior ( 実際の動作 )
"pushImageRotateZoom()" do not render part of the pushed image if X or Y coordinates of pushed picture less then display's X=0 or Y=0.

### Steps to reproduce ( 再現のための前提条件 )

1. Do "pushImageRotateZoom()" with size 16x16 for example to display's coordinates (-10; -5) and we will see nothing but i supposer we should see part of the image 6x11 pixels in the upper left corner of the display

### Code to reproduce this issue ( 再現させるためのコード )

https://gist.github.com/ESPboy-edu/f57c7d3dda8680aff5fe4e99380cd9c6

lovyan03, 2023-10-07:
You are using `uint8_t`.
Thus I assume you are not specifying negative coordinates.
Change the coordinate variable to `int8_t` and see how it works.

ESPboy-edu, 2023-10-07:
I am even ashamed for such inattention on my part and taking up your time. I definitely need to be more careful before writing such nonsense. Thank you.

closed 2023-10-12 by lovyan03