lovyan03/LovyanGFX #455: compile issue with esp idf v5

chaabanihoussem, 2023-10-09:
Hi

trying to build with esp32s3 esp idf v5 and doesn't build, it build with esp idf v4 but I need esp idf v5 since other libraries are only supported esp idf 5 

here is logs for terminal output:

-----------------------------------------------------------------------------------------------------------------------------------------------------------------

screen/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp: In constructor 'lgfx::v1::pixelcopy_t::pixelcopy_t(const void*, lgfx::v1::color_depth_t, lgfx::v1::color_depth_t, bool, const void*, uint32_t)':
/home/hch/esp/screen/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:39:25: error: member 'lgfx::v1::pixelcopy_t::<anonymous>' is used uninitialized [-Werror=uninitialized]
   39 |     , src_mask  ( (1 << src_bits) - 1 )
      |                         ^~~~~~~~
/home/hch/esp/screen/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:40:25: error: member 'lgfx::v1::pixelcopy_t::<anonymous>' is used uninitialized [-Werror=uninitialized]
   40 |     , dst_mask  ( (1 << dst_bits) - 1 )
      |                         ^~~~~~~~
cc1plus: some warnings being treated as errors
[606/654] Building CXX object esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/LGFXBase.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/hch/esp/screen/build/log/idf_py_stderr_output_21084 and /home/hch/esp/screen/build/log/idf_py_stdout_output_21084

-------------------------------------------------------------------------------------------------------------------------------------------------------------------

any idea how to get this building? 

thanks

lovyan03, 2023-10-12:
最新版を使っていますか？

chaabanihoussem, 2023-10-12:
Hi, yes I'm using the last esp idf and the last version from your master branch,
Did you manage to compile it with v5.x idf?

tobozo, 2023-10-12:
hi,

[compilation tests are fine with esp-idf v5.x](https://github.com/lovyan03/LovyanGFX/actions/runs/6494087790)

I'm not sure why you get this error, but I'd focus on the ` cc1plus: some warnings being treated as errors` and try to play with `target_compile_options` in CMakeLists.txt  or `CXXFLAGS` in component.mk.

The `-Wuninitialized` error could be a false positive though, if you believe so you can still use `#pragma GCC diagnostic ignored "-Wuninitialized"` until this is fixed.

chaabanihoussem, 2023-10-13:
thanks, that hepled to complie that file but now I'm getting linking issue

--------------------------------------------------------------------------------------------------------------------

undefined reference to `_ZN4lgfx2v112Panel_Device11rst_controlEb'

-----------------------------------------------------------------------------------

any ideas?

tobozo, 2023-10-13:
maybe there are some leftovers from a previously failed compilation?

or it could be a problem in your project folder, hard to say without seeing the actual project source :-)

closed 2023-10-13 by tobozo

Py-Jinx, 2024-01-30:
I'm still gettig the same error(s) as described above.  Errors start with
[815/1224] Building CXX object esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/misc/pixelcopy.cpp.obj**FAILED:** esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/misc/pixelcopy.cpp.obj.

I don't understand the solution of chaabanihoussem. Can someone please explain the esp-idf path issue? 
I'm using ESP-IDF v5.1.2