lovyan03/LovyanGFX #456: drawBmp() displays cropped image

TheRegularDX, 2023-10-11:
## Environment ( 実行環境 )

- MCU or Board name:  ESP32-2432S028
- Panel Driver IC:  ILI9341
- Bus type:  SPI
- LovyanGFX version:  latest
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   PlatformIO
- Operating System:   macOS

## Problem Description ( 問題の内容 )

loading an image from an sd card and displaying it with drawBmp() results in a cropped image being displayed on the lcd, specifically only the bottom portion of the image. ive tried other pictures but got the same result.

### Expected Behavior ( 期待される動作 )

a full image gets displayed on the lcd

### Actual Behavior ( 実際の動作 )

cropped image gets displayed.

### Steps to reproduce ( 再現のための前提条件 )

1. init display
2. init sd card
3. load an image and display it like this:
```
auto file = SD.open("/1.bmp");	
lcd.drawBmp(&file, 100, 50);
``` 

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
// frankenstein code from many examples and tutorials. 
// i am a beginner and this was a bit hard to set up. 
// loyvan if possible please improve the documentation

#include <Arduino.h>
#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>  
#include <SPI.h>
#include <SD.h>
#include <FS.h>

#define LGFX_ESP32_2432S028                // Sunton ESP32 2432S028

#define SD_MOSI 23
#define SD_MISO 19
#define SD_SCK 18
#define SD_CS 5

static LGFX lcd;                 
SPIClass SD_SPI;

int SD_init()
{
  SD_SPI.begin(SD_SCK, SD_MISO, SD_MOSI);
  if (!SD.begin(SD_CS, SD_SPI, 40000000))
  {
    Serial.println("Card Mount Failed");
    return 1;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE)
  {
    Serial.println("No TF card attached");
    return 1;
  }
  return 0;
}

void setup(void)
{
  Serial.begin(115200);
  lcd.init();
  lcd.setRotation(1);
  lcd.setBrightness(128);

  if (SD_init() == 0)
  {
    Serial.println("TF card initialized successfully");

  } else {
    Serial.println("TF card initialization failed");

  }

}

void loop(void)
{
	auto file = SD.open("/1.bmp");	
	lcd.drawBmp(&file, 100, 50);
}
```

TheRegularDX, 2023-10-11:
This is the result:

[image]

Unfortunately i dont have access to 1.bmp right now. Ill upload it when possible

lovyan03, 2023-10-12:
no need `SPIClass SD_SPI;`
you can use `SPI` instance

lovyan03, 2023-10-12:
Do the following
```
  SPI.begin(SD_SCK, SD_MISO, SD_MOSI);
  if (!SD.begin(SD_CS, SPI, 40000000))
```

TheRegularDX, 2023-10-12:
thank you! noob mistake. now it works fine

closed 2023-10-12 by tobozo