lovyan03/LovyanGFX #457: Incorrect using of new ledc with core 3.0-alpha

PM04290, 2023-10-16:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  all ESP
- Panel Driver IC:    *
- Bus type:           *
- LovyanGFX version:  last version
- FrameWork version:   ArduinoESP32 v3.0-alpha
- Build Environment:   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )
compile error with new Espressif prototype with core 3.0-alpha

```
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/defines" "-ID:\\HomeAssistant\\DIY\\MesProjets\\SmartMonitor" -iprefix "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/include/" "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/includes" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/qio_qspi/include" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\cores\\esp32" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\variants\\esp32s3" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\WiFi\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\FS\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\SPIFFS\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\HAintegration\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\LovyanGFX\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\ArduinoJson\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\ESPAsyncWebServer\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\AsyncTCP\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\ESPmDNS\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\EEPROM\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\Update\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\PubSubClient\\src" "@C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_771901/build_opt.h" "D:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\LovyanGFX\\src\\lgfx\\v1\\platforms\\esp32\\Light_PWM.cpp" -o "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_771901\\libraries\\LovyanGFX\\lgfx\\v1\\platforms\\esp32\\Light_PWM.cpp.o"
D:\Users\Philippe-ASUS\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32\Light_PWM.cpp: In member function 'virtual bool lgfx::v1::Light_PWM::init(uint8_t)':
D:\Users\Philippe-ASUS\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32\Light_PWM.cpp:43:5: error: 'ledcSetup' was not declared in this scope
   43 |     ledcSetup(_cfg.pwm_channel, _cfg.freq, PWM_BITS);
      |     ^~~~~~~~~
D:\Users\Philippe-ASUS\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32\Light_PWM.cpp:44:5: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   44 |     ledcAttachPin(_cfg.pin_bl, _cfg.pwm_channel);
      |     ^~~~~~~~~~~~~
      |     ledcAttach
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/defines" "-ID:\\HomeAssistant\\DIY\\MesProjets\\SmartMonitor" -iprefix "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/include/" "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/includes" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/qio_qspi/include" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\cores\\esp32" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\variants\\esp32s3" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\WiFi\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\FS\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\SPIFFS\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\HAintegration\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\LovyanGFX\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\ArduinoJson\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\ESPAsyncWebServer\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\AsyncTCP\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\ESPmDNS\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\EEPROM\\src" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\libraries\\Update\\src" "-ID:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\PubSubClient\\src" "@C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_771901/build_opt.h" "D:\\Users\\Philippe-ASUS\\Documents\\Arduino\\libraries\\LovyanGFX\\src\\lgfx\\v1\\platforms\\esp32\\Panel_CVBS.cpp" -o "C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino_build_771901\\libraries\\LovyanGFX\\lgfx\\v1\\platforms\\esp32\\Panel_CVBS.cpp.o"
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCo [... cut ...]
### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2023-10-16:
Hi,

please try [this branch](https://github.com/lovyan03/LovyanGFX/tree/esp32-3.x.x), it has a commit fixing issues with arduino core 3.x.x

https://github.com/lovyan03/LovyanGFX/commit/f3ad7c45c4e84ab128fb275948009a065bdcb7ed

PM04290, 2023-10-16:
my remark was just to point out the error due to the ledc change, when testing the new core 3.0.
I've seen the code and it seems to fix the problem; I will wait for the release of the public version of the library.
For my projects I am currently staying in version 2.x of the core.
well seen the #define on SOC_DAC_SUPPORTED rather than ESP_ARDUINO_VERSION_MAJOR

tobozo, 2023-10-16:
although it does the work, I'm not sure about adding `SOC_DAC_SUPPORTED` in the library, I believe it should inherit from esp32 core instead but haven't been able to find where it's declared from

closed 2023-11-22 by github-actions[bot]