lovyan03/LovyanGFX #463: T-Display-S3

powersoft, 2023-11-01:
Is there a driver for the `Lilygo T-Display S3 available?

Thanks for any help.

Jan Kromhout
Hellevoetsluis-NL

tobozo, 2023-11-01:
hi,

have you tried adding this **before** including LovyanGFX.h ? 

```cpp
#define LGFX_TTGO_TDISPLAY
```

powersoft, 2023-11-01:
Have trie to do the 1_simple_use, no results.
This is my addapted code for it.
`// 基本的な使い方

// ※  もし対応機種を ArduinoIDE以外の環境で使用する場合や、
// 対応機種がボードマネージャに無い場合 ( TTGO T-Wristband や ESP-WROVER-KIT等 ) は、
// LovyanGFX.hppのincludeより前に、define LGFX_～ の定義を記述してください。

#define LGFX_TTGO_TDISPLAY                 // TTGO T-Display

// #define LGFX_AUTODETECT // 自動認識 (D-duino-32 XS, WT32-SC01, PyBadge はパネルID読取りが出来ないため自動認識の対象から外れています)

// 複数機種の定義を行うか、LGFX_AUTODETECTを定義することで、実行時にボードを自動認識します。

// ヘッダをincludeします。
#include <LovyanGFX.hpp>

#include <LGFX_AUTODETECT.hpp>  // クラス"LGFX"を準備します
//#include <lgfx_user/LGFX_ESP32_sample.hpp> // またはユーザ自身が用意したLGFXクラスを準備します

static LGFX lcd;                 // LGFXのインスタンスを作成。
static LGFX_Sprite sprite(&lcd); // スプライトを使う場合はLGFX_Spriteのインスタンスを作成。

// もし現在 TFT_eSPI を使用中で、ソースをなるべく変更したくない場合は、こちらのヘッダを利用できます。
// #include <LGFX_TFT_eSPI.hpp>
// static TFT_eSPI lcd;               // TFT_eSPIがLGFXの別名として定義されます。
// static TFT_eSprite sprite(&lcd);   // TFT_eSpriteがLGFX_Spriteの別名として定義されます。

// 対応機種に無い構成で使う場合は、 examples/HowToUse/2_user_setting.ino を参照してください。
// また設定例はsrc/lgfx_userフォルダにもあります。

void setup(void)
{
// 最初に初期化関数を呼び出します。
  lcd.init();

// 回転方向を 0～3 の4方向から設定します。(4～7を使用すると上下反転になります。)
  lcd.setRotation(1);

// バックライトの輝度を 0～255 の範囲で設定します。
  lcd.setBrightness(128);

// 必要に応じてカラーモードを設定します。（初期値は16）
// 16の方がSPI通信量が少なく高速に動作しますが、赤と青の諧調が5bitになります。
// 24の方がSPI通信量が多くなりますが、諧調表現が綺麗になります。
//lcd.setColorDepth(16);  // RGB565の16ビットに設定
  lcd.setColorDepth(24);  // RGB888の24ビットに設定(表示される色数はパネル性能によりRGB666の18ビットになります)

// 基本的な図形の描画関数は以下の通りです。
/*
  fillScreen    (                color);  // 画面全体の塗り潰し
  drawPixel     ( x, y         , color);  // 点
  drawFastVLine ( x, y   , h   , color);  // 垂直線
  drawFastHLine ( x, y, w      , color);  // 水平線
  drawRect      ( x, y, w, h   , color);  // 矩形の外周
  fillRect      ( x, y, w, h   , color);  // 矩形の塗り
  drawRoundRect ( x, y, w, h, r, color);  // 角丸の矩形の外周
  fillRoundRect ( x, y, w, h, r, color);  // 角丸の矩形の塗り
  drawCircle    ( x, y      , r, color);  // 円の外周
  fillCircle    ( x, y      , r, color);  // 円の塗り
  drawEllipse   ( x, y, rx, ry , color);  // 楕円の外周
  fillEllipse   ( x, y, rx, ry , color);  // 楕円の塗り
  drawLine      ( x0, y0, x1, y1        , color); // ２点間の直線
  drawTriangle  ( x0, y0, x1, y1, x2, y2, color); // ３点間の三角形の外周
  fillTriangle  ( x0, y0, x1, y1, x2, y2, color); // ３点間の三角形の塗り
  drawBezier    ( x0, y0, x1, y1, x2, y2, color); // ３点間のベジエ曲線
  drawBezier    ( x0, y0, x1, y1, x2, y2, x3, y3, color); // ４点間のベジエ曲線
  drawArc       ( x, y, r0, r1, angle0, angle1, color);   // 円弧の外周
  fillArc       ( x, y, r0, r1, angle0, angle1, color);   // 円弧の塗り
*/
[... 202 lines cut ...]
  delay(5000);

  lcd.startWrite(); // ここでstartWrite()することで、SPIバスを占有したままにする。
}

void loop(void)
{
  static int count = 0;
  static int a = 0;
  static int x = 0;
  static int y = 0;
  static float zoom = 3;
  ++count;
  if ((a += 1) >= 360) a -= 360;
  if ((x += 2) >= lcd.width()) x -= lcd.width();
  if ((y += 1) >= lcd.height()) y -= lcd.height();
  sprite.setPaletteColor(1, lcd.color888( 0, 0, count & 0xFF));
  sprite.setPaletteColor(2, lcd.color888( 0,~count & 0xFF, 0));
  sprite.setPaletteColor(3, lcd.color888( count & 0xFF, 0, 0));

  sprite.pushRotateZoom(x, y, a, zoom, zoom, 0);

  if ((count % 100) == 0) lcd.display(); // 電子ペーパーの場合の表示更新を 100回に一度行う
}
`

powersoft, 2023-11-01:
sorry for this mess, have put the code between the code.
Try it again.
`#define LGFX_TTGO_TDISPLAY                
#include <LovyanGFX.hpp>

#include <LGFX_AUTODETECT.hpp>

static LGFX lcd; 

void setup() {
  lcd.init();
  lcd.setRotation(0); 
  lcd.fillScreen(TFT_BLUE);
}

void loop() {
  // put your main code here, to run repeatedly:

}`

tobozo, 2023-11-01:
you should use three backticks ``` to open and close formatted code, append the language name to enable the highlighter

```
    ```cpp

     // some cpp code

    ```
```

powersoft, 2023-11-01:
This is my test code for the T3 display. When running this the screen is stay black.
What to do to get this simple code running.

Cheers,
Jan

```
#define LGFX_TTGO_TDISPLAY
#include <LovyanGFX.hpp>

#include <LGFX_AUTODETECT.hpp>

static LGFX lcd;

void setup() {
lcd.init();
lcd.setRotation(0);
lcd.fillScreen(TFT_BLUE);
}

void loop() {
// put your main code here, to run repeatedly:

}
```

tobozo, 2023-11-01:
see this [previous discussion](https://github.com/lovyan03/LovyanGFX/discussions/315)

powersoft, 2023-11-01:
Thanks, you are my lifesaver.!!!! Now  can go ahead with LovyanGFX.
Cheers,
Jan

powersoft, 2023-11-01:
Do you know if there is a LovyanGFX driver for the new 
LilyGO T-Display S3 AMOLED ESP32-S3 - with 1.91 inch AMOLED Display?

Cheers,
Jan

tobozo, 2023-11-01:
nope I didn't know this product existed

zeroomega, 2023-11-06:
@powersoft I am using T-Display-S3 in my hobbyist project with LovyanGFX. My configuration files look like this:
https://github.com/nostreetracing/S3Dash/blob/165f7ac334fe66c848cde3aa8673fb135b6c95c1/main/lcd.h
And it works well.

The AMOLED one uses QSPI instead of single lane SPI or 8 bit 8080 parallel so you currently cannot get the AMOLED one working with LovyanGFX. I submit a feature request for QSPI in https://github.com/lovyan03/LovyanGFX/issues/467 .

tobozo, 2023-11-06:
@zeroomega maybe join forces with @Forairaaaaa ?

https://github.com/lovyan03/LovyanGFX/pull/388

closed 2023-12-13 by github-actions[bot]

sauloverissimo, 2025-02-07:
> [@powersoft](https://github.com/powersoft) I am using T-Display-S3 in my hobbyist project with LovyanGFX. My configuration files look like this: [...]

Nice here! Thanks, man!

tobozo, 2025-04-25:
@powersoft support for [Lilygo T-Display-S3-AMOLED](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T_Display_S3_AMOLED.hpp) was added to the [QSPI branch](https://github.com/lovyan03/LovyanGFX/tree/QSPI)

there's also a config sample for [Lyligo T4-S3 AMOLED](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T4_S3.hpp)