lovyan03/LovyanGFX #467: Add support for QSPI bus in ESP32-S3?

zeroomega, 2023-11-06:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

As the title suggested, is it feasible to support the QSPI bus in ESP32-S3 platform? Right now under "src/lgfx/v1/platforms/esp32" there is only regular SPI bus support. It won't work with a QSPI bus.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I have been using LovyanGFX on Lilygo T-Display-S3 with great success. However their newer display boards like "T-Display-S3 AMOLED"(https://www.lilygo.cc/products/t-display-s3-amoled) and "Lilygo T-Display-S3-Long"(https://www.lilygo.cc/products/t-display-s3-long) started to use QSPI bus between the MCU and display IC. I was trying to add support of QSPI bus and their display ICs to my local LovyanGFX fork but I discovered that the SPI bus code for ESP32 in your code base is using the low level ESP32 HAL functions instead of the regular ESP-IDF APIs. I don't have a lot of knowledge about low level HAL of ESP32 platform so I cannot add QSPI support by my own. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A supported QSPI bus implementation under "src/lgfx/v1/platforms/esp32" or "src/lgfx/v1/platforms/esp32S3"

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Maybe provide a common interface that allows the bus implementation to be done in user's code (outside of LovyanGFX) so ESP-IDF APIs can be easily used.

**Additional context**
Add any other context or screenshots about the feature request here.

tobozo, 2025-04-26:
@zeroomega @Kalrkloss please try the [QSPI branch](https://github.com/lovyan03/LovyanGFX/tree/QSPI)

Implementing the T-Display-S3-Long config:

- Copy the config from [T-Display-S3-Amoled](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T_Display_S3_AMOLED.hpp)
- Edit pinout to match qspi io/sck/cs/rst pins
- Use Panel_SH8601Z instead of Panel_RM67162
- Add a Light_PWM config section for the backlight

d4rkmen, 2025-12-04:
Hi, guys. I am also trying to make a driver for QSPI ST77916 (but for ESP-IDF) using esp_driver_spi with queued DMA transactions. It looks very promising. I had to add Bus_QSPI in separate struct to make it more clear. When working on the driver I have found pixelcopy issues for large sprites, but figured out how to solve them. One thing I cant make to work is Compressed Memory Write. It behaves 1:1 as normal write, and not like described in the datasheet.