lovyan03/LovyanGFX #468: examples_for_PC/CMake_WASM: compile LGFX_WASM.cpp fails with  error: no member named 'sdl_event_handler' in 'lgfx::Panel_sdl'

mhaberler, 2023-11-08:
## Environment ( 実行環境 )

- MCU or Board name:  [SDL2 webassembly example]
- LovyanGFX version:  [v1.1.9]
- FrameWork version:   [ArduinoESP32 v2.0.3|ESP-IDF v4.4|other]
- Build Environment:   [bash]
- Operating System:   [Linux Debian 6.1.55-1 and macOS 14.0 (23A344), M1 - identical results]

## Problem Description ( 問題の内容 )
````
/Users/mah/graphics/LovyanGFX/examples_for_PC/CMake_WASM/LGFX_WASM.cpp:12:20: error: no member named 'sdl_event_handler' in 'lgfx::Panel_sdl'
   12 |   lgfx::Panel_sdl::sdl_event_handler();
      |   ~~~~~~~~~~~~~~~~~^
````
### Expected Behavior ( 期待される動作 )
compiles and builds WASM output

### Actual Behavior ( 実際の動作 )
make fails

### Steps to reproduce ( 再現のための前提条件 )

1. Install emscripten as per https://emscripten.org/docs/getting_started/downloads.html#installation-instructions-using-the-emsdk-recommended
2. import emscripten shell environment
3. checkout Lovyan v1.1.9 302169
4. cd examples_for_PC/CMake_WASM
5. mkdir build
6. cd build
7. emcmake cmake ..
8. emmake make

### build log
````
BigM1:build mah$ emcmake cmake ..
configure: cmake .. -DCMAKE_TOOLCHAIN_FILE=/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake -DCMAKE_CROSSCOMPILING_EMULATOR=/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/node/16.20.0_64bit/bin/node
-- Configuring done (1.6s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/examples_for_PC/CMake_WASM/build
BigM1:build mah$ emmake make
make: make
[  4%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/Fonts/IPA/lgfx_font_japan.c.o
[  8%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_cn.c.o
[ 12%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_ja.c.o
[ 16%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_kr.c.o
[ 20%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_tw.c.o
[ 24%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/utility/lgfx_pngle.c.o
[ 28%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/utility/lgfx_qoi.c.o
[ 32%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/utility/lgfx_qrcode.c.o
[ 36%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/utility/lgfx_tjpgd.c.o
[ 40%] Building C object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/utility/miniz.c.o
[ 44%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/LGFXBase.cpp.o
ports:INFO: retrieving port: sdl2 from https://github.com/libsdl-org/SDL/archive/release-2.24.2.zip
ports:INFO: unpacking port: sdl2
cache:INFO: generating port: sysroot/lib/wasm32-emscripten/libSDL2.a... (this will be cached in "/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/libSDL2.a" for subsequent builds)
system_libs:INFO: compiled 115 inputs in 2.52s
cache:INFO:  - ok
[ 48%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/LGFX_Button.cpp.o
[ 52%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/LGFX_Sprite.cpp.o
[ 56%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/lgfx_fonts.cpp.o
[ 60%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/misc/DividedFrameBuffer.cpp.o
[ 64%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/misc/SpriteBuffer.cpp.o
[ 68%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/misc/common_function.cpp.o
[ 72%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp.o
[ 76%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/panel/Panel_Device.cpp.o
[ 80%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/panel/Panel_FrameBufferBase.cpp.o
[ 84%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/platforms/sdl/Panel_sdl.cpp.o
[ 88%] Building CXX object CMakeFiles/index.dir/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/src/lgfx/v1/platforms/sdl/common.cpp.o
[ 92%] Building CXX object CMakeFiles/index.dir/LGFX_WASM.cpp.o
/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/examples_for_PC/CMake_WASM/LGFX_WASM.cpp:12:20: error: no member named 'sdl_event_handler' in 'lgfx::Panel_sdl'
   12 |   lgfx::Panel_sdl::sdl_event_handler();
      |   ~~~~~~~~~~~~~~~~~^
1 error generated.
em++: error: '/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/upstream/bin/clang++ -target wasm32-unknown-emscripten -fignore-exceptions -fvisibility=default -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr -DEMSCRIPTEN -I/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/upstream/emscripten/cache/sysroot/include/SDL2 --sysroot=/Users/mah/Ballon/src/BalloonWare/graphics/emsdk/upstream/emscripten/cache/sysroot -Xclang -iwithsysroot/include/compat -DLGFX_SDL -I/Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/examples_for_PC/CMake_WASM/../../../LovyanGFX/src -O2 -MD -MT CMakeFiles/index.dir/LGFX_WASM.cpp.o -MF CMakeFiles/index.dir/LGFX_WASM.cpp.o.d -c /Users/mah/Ballon/src/BalloonWare/graphics/LovyanGFX/examples_for_PC/CMake_WASM/LGFX_WASM.cpp -o CMakeFiles/index.dir/LGFX_WASM.cpp.o' failed (returned 1)
make[2]: *** [CMakeFiles/index.dir/LGFX_WASM.cpp.o] Error 1
make[1]: *** [CMakeFiles/index.dir/all] Error 2
make: *** [all] Error 2
emmake: error: 'make' failed (returned 2)
````

I assume this is an issue with either the SDL (release-2.24.2) and/or the LovyanGFX example/SDL support code @imliubo have a suggestion?

thanks in advance,
Michael

tobozo, 2023-11-08:
hi,

`lgfx::Panel_sdl::sdl_event_handler()` was removed and replaced by `lgfx::Panel_sdl::main()` which accepts a callback

https://github.com/lovyan03/LovyanGFX/blob/master/examples_for_PC/PlatformIO_SDL/src/sdl_main.cpp

the WASM example sure needs to be updated

mhaberler, 2023-11-08:
thanks! will post a PR

mhaberler, 2023-11-10:
see https://github.com/lovyan03/LovyanGFX/pull/470 

warning: complete emscripten idiot here

mhaberler, 2023-11-12:
resolved

closed 2023-11-12 by mhaberler