lovyan03/LovyanGFX #485: setSwapBytes issue. cut top row pixels?

lefty01, 2023-12-06:
not sure if this is a bug or some usage/config error but I really do not currently understand this ...

## Environment 
platform = espressif32
board = esp32dev
framework = arduino
- MCU or Board name:  **TTGO-T-Display** ESP32
- Panel Driver IC:    ST7789
- Bus type:           SPI
- LovyanGFX version:  LovyanGFX@^1.1.9
- Build Environment:  PlatformIO
- Operating System:   Linux
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

## Problem Description ( 問題の内容 )
using Sprites and pushImage to sprite (see example code below) to get the color right need to use swapBytes.
the  two images are  50x50 pixle rectangle with solid colors (blue-ish and purple-ish color created in gimp and converted online to c-array)
with setSwapBytes(false) colors are not correctly displayed but the position / dimension is ok.
when setSwapBytes(true) (I toggle via button to show the effect) then the top pixel row or rows (not sure how many pixel) gets "cleared" out.
You can see in the images that with swap=false the color is like red/orange and the vert red lines go to top.
when swap=true color is blue/purple and you see lines and top-left blue rectangle are reduced.

### Expected Behavior ( 期待される動作 )
toggle swapBytes shall change color only not remove pixels from top of tft screen.

### Actual Behavior ( 実際の動作 )
row of pixels not show on top of screen

### Steps to reproduce ( 再現のための前提条件 )
load code below, I 

// If possible, attach a picture of your setup/wiring here.
[image]
[image]

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

https://gist.github.com/lefty01/9ff3d09073d182cd136f68f9564fdced

lovyan03, 2023-12-07:
あなたのデータは４ピクセル足りないように見える。

あなたが想定しているデータサイズは 50×50 = 2500 pixel
実際に配列に入っているデータサイズは 16 × 156 = 2496 pixel

４ピクセル足りないのはあなたのデータが足りないからだと思います。

closed 2023-12-07 by lovyan03

lefty01, 2023-12-07:
thanks for your fast response and the hint to the number of pixels.
now when experimenting further I noticed that without changing (or fixing) the actual image data array, when I add
`  tft.setRotation(1);
` 
also to the loop()  then this again behaves as expected. that means I do need setRotation in setup (if createSprite is done in setup()) **and** loop()

I need to learn how does tft.setRotation effect the sprite(s)

lovyan03, 2023-12-07:
As an additional note, when startWrite is used, LovyanGFX occupies the SPI bus and operates.

If startWrite is not used, each function will not proceed until SPI communication is complete.

When startWrite is used, each function does not wait for SPI communication to complete before proceeding. In other words, the DMA transfer communication is executed in the background.

For this reason, fillSprite(TFT_WHITE); rewrites the data being transferred.

The sample code should have displayed the image well by using two sprites alternately, but your code uses only one sprite, which rewrites the data being transferred, resulting in an incomplete image.

lefty01, 2023-12-07:
now the MovingCircles example makes more sense (flipping beween those two sprites)