lovyan03/LovyanGFX #489: Compile errors

ammaree, 2023-12-14:
**Carefully written Issues are more likely to be given priority.**

## Environment
- MCU or Board name:  ESP-WROVER-DEV v4.1
- Panel Driver IC:    ILI9341
- Bus type:           SPI
- LovyanGFX version:  master
- FrameWork version:   ESP-IDF v5.3.0 (master)
- Build Environment:   Eclipse
- Operating System:   macOS

## Problem Description
Errors and warnings when building small text only demo

### Expected Behavior
Build to complete without errors

### Actual Behavior

> /Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/panel/Panel_M5HDMI.cpp:308:8: error: 'uint8_t' in namespace 'std' does not name a type; did you mean 'wint_t'?
  308 |   std::uint8_t Panel_M5HDMI::HDMI_Trans::readRegister(std::uint8_t register_address)
      |        ^~~~~~~
      |        wint_t
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/panel/Panel_M5HDMI.cpp:315:8: error: 'uint16_t' in namespace 'std' does not name a type; did you mean 'wint_t'?
  315 |   std::uint16_t Panel_M5HDMI::HDMI_Trans::readRegister16(std::uint8_t register_address)
      |        ^~~~~~~~
      |        wint_t
In file included from /Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp:43:
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/soc/esp32/include/soc/apb_ctrl_reg.h:9:2: warning: #warning "apb_ctrl_reg is deprecated due to duplicated with syscon_reg, please use syscon_reg instead, they are same" [-Wcpp]
    9 | #warning "apb_ctrl_reg is deprecated due to duplicated with syscon_reg, please use syscon_reg instead, they are same"
      |  ^~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp: In function 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::spi::init(int, int, int, int, int)':
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp:487:29: error: 'INTR_CPU_ID_AUTO' was not declared in this scope
  487 |         buscfg.isr_cpu_id = INTR_CPU_ID_AUTO;
      |                             ^~~~~~~~~~~~~~~~
In file included from /Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp:31:
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/driver/deprecated/driver/dac.h:16:2: warning: #warning "The legacy DAC driver is deprecated, please use `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead" [-Wcpp]
   16 | #warning "The legacy DAC driver is deprecated, please use `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead"
      |  ^~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp: In member function 'void lgfx::v1::Panel_CVBS::deinit()':
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp:1905:28: warning: 'DAC_CHANNEL_1' is deprecated: please use 'DAC_CHAN_0' instead [-Wdeprecated-declarations]
 1905 |         dac_output_disable(DAC_CHANNEL_1); // for GPIO 25
      |                            ^~~~~~~~~~~~~
In file included from /Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/driver/deprecated/driver/dac_types_legacy.h:14,
                 from /Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/driver/deprecated/driver/dac.h:13:
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/hal/include/hal/dac_types.h:10:5: note: declared here
   10 |     DAC_CHANNEL_1 __attribute__((deprecated("please use 'DAC_CHAN_0' instead"))) = 0,    /*!< Alias of 'DAC_CHAN_0', now the channel index start from '0' */
      |     ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp:1908:28: warning: 'DAC_CHANNEL_2' is deprecated: please use 'DAC_CHAN_1' instead [-Wdeprecated-declarations]
 1908 |         dac_output_disable(DAC_CHANNEL_2); // for GPIO 26
      |                            ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/hal/include/hal/dac_types.h:11:5: note: declared here
   11 |     DAC_CHANNEL_2 __attribute__((deprecated("please use 'DAC_CHAN_1' instead"))) = 1,    /*!< Alias of 'DAC_CHAN_1', now the channel index start from '0' */
      |     ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp: In member function 'virtual bool lgfx::v1::Panel_CVBS::init(bool)':
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp:1955:25: warning: 'DAC_CHANNEL_1' is deprecated: please use 'DAC_CHAN_0' instead [-Wdeprecated-declarations]
 1955 |       dac_output_enable(DAC_CHANNEL_1); // for GPIO 25
      |                         ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/hal/include/hal/dac_types.h:10:5: note: declared here
   10 |     DAC_CHANNEL_1 __attribute__((deprecated("please use 'DAC_CHAN_0' instead"))) = 0,    /*!< Alias of 'DAC_CHAN_0', now the channel index start from '0' */
      |     ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/Panel_CVBS.cpp:1958:25: warning: 'DAC_CHANNEL_2' is deprecated: please use 'DAC_CHAN_1' instead [-Wdeprecated-declarations]
 1958 |       dac_output_enable(DAC_CHANNEL_2); // for GPIO 26
      |                         ^~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/hal/include/hal/dac_types.h:11:5: note: declared here
   11 |     DAC_CHANNEL_2 __attribute__((deprecated("please use 'DAC_CHAN_1' instead"))) = 1,    /*!< Alias of 'DAC_CHAN_1', now the channel index start from '0' */
      |     ^~~~~~~~~~~~~
`

### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
```cpp
#define LGFX_ESP_WROVER_KIT
#define SD_SUPPORTED
#include <LovyanGFX.hpp>

static LGFX lcd;
static StaticTask_t ttsGUI;
static StackType_t tsbGUI[guiSTACK_SIZE] = { 0 };
char DispBuf[halLCD_BUF_SIZE];
u8_t Page = 0, lcdLevel = 0;

void vGuiInit(void) {
	lcd.init();
	lcd.clear();
	if (lcd.width() < lcd.height())
		lcd.setRotation(lcd.getRotation() ^ 1);
	lcd.setBrightness(128);
	lcd.setColorDepth(halLCD_BITS_PX);
	lcd.cp437(true);
	lcd.setTextWrap(false);
}

void vGuiBrightness(void) { }
void vGuiUpdate(void) { }
void vGuiRefresh(void) { }
void vGuiDeInit(void) { }

int xGuiPageTasks(void) {
	report_t sRprt = { .pcBuf=DispBuf, .Size=halLCD_MAX_CHAR, .sFM=(fm_t)makeMASK09x23(0,0,1,1,1,1,1,0,1,0x03FFFFF) };
	return xRtosReportTasks(&sRprt);
}

int xGuiPageMemory(void) {
	report_t sRprt = { .pcBuf = DispBuf, .Size = halLCD_MAX_CHAR, .sFM = (fm_t) makeMASK11x21(1,1,1,1,1,1,1,0,1,1,1,0) };
	return xRtosReportMemory(&sRprt);
}

int xGuiPageFlags(void) {
	report_t sRprt = { .pcBuf = DispBuf, .Size = halLCD_MAX_CHAR, .sFM = (fm_t)makeMASK08x24(0,1,0,0,0,0,0,0,0) };
	int iRV = xBitMapDecodeChanges(&sRprt, EFprv, EFcur, 0x0000FF00, EFnames);
	iRV += xBitMapDecodeChanges(&sRprt, EFprv, EFcur, 0x00000FF, EFnames);
	EFprv = EFcur;
	iRV += wprintfx(&sRprt, "System Flags : 0x%08X\r\n", SFcur);
	return iRV;
}

int xGuiPageSensors(void) {
	report_t sRprt={ .pcBuf=DispBuf, .Size=halLCD_MAX_CHAR, .sFM=(fm_t)makeMASK12x20(1,1,0,0,0,0,0,0,0,0,0,1,0x000FFFFF) };
	return xTaskSensorsReport(&sRprt);
}

// ###################################### GUI Task support #########################################

void vTaskGUI(void * pvPara) {
	vGuiInit();
    while(bRtosTaskWaitOK(taskGUI_MASK, portMAX_DELAY)) {
    	vGuiBrightness();
    	if (pReqBuf == NULL) vGuiUpdate();
    	vGuiRefresh();
        vTaskDelay(pdMS_TO_TICKS((ioB4GET(ioGUIintval)+1) * MILLIS_IN_SECOND));
    }
    vGuiDeInit();
    vRtosTaskDelete(NULL);
}

extern "C" void vTaskGUI_Start(void * pvPara) {
	xRtosTaskCreateStatic(vTaskGUI, guiNAME, guiSTACK_SIZE, pvPara, guiTASK_PRIORITY, tsbGUI, &ttsGUI, tskNO_AFFINITY);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2023-12-18:
hi, 

have you tried a release version of esp-idf (e.g 5.1.2) along with the develop branch of this repo ?

ammaree, 2023-12-18:
At a glance the same problems with release v5.2 and release v5.1, messages just in different order.

ammaree, 2023-12-18:
With v5.1, other than the warnings related to the DAC the only errors that remain are:
```cpp
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:50:8: error: 'LGFX' does not name a type
   50 | static LGFX lcd;
      |        ^~~~
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp: In function 'void vGuiInit()':
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:63:9: error: 'lcd' was not declared in this scope
   63 |         lcd.init();
      |         ^~~
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp: In function 'void vGuiRefresh()':
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:113:5: error: 'lcd' was not declared in this scope
  113 |     lcd.clear(0x0000);
      |     ^~~
```

So effectively the 1st error....

ammaree, 2023-12-18:
With v5.2, other than the DAC related warnings (all still there) the errors that remain are:
```cpp
In file included from /Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp:43:
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/soc/esp32/include/soc/apb_ctrl_reg.h:9:2: warning: #warning "apb_ctrl_reg is deprecated due to duplicated with syscon_reg, please use syscon_reg instead, they are same" [-Wcpp]
    9 | #warning "apb_ctrl_reg is deprecated due to duplicated with syscon_reg, please use syscon_reg instead, they are same"
      |  ^~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp: In function 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::spi::init(int, int, int, int, int)':
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/platforms/esp32/common.cpp:487:29: error: 'INTR_CPU_ID_AUTO' was not declared in this scope
  487 |         buscfg.isr_cpu_id = INTR_CPU_ID_AUTO;
      |                             ^~~~~~~~~~~~~~~~
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/panel/Panel_M5HDMI.cpp:308:8: error: 'uint8_t' in namespace 'std' does not name a type; did you mean 'wint_t'?
  308 |   std::uint8_t Panel_M5HDMI::HDMI_Trans::readRegister(std::uint8_t register_address)
      |        ^~~~~~~
      |        wint_t
/Users/andremaree/DevSpace/z-repo/lovyan/src/lgfx/v1/panel/Panel_M5HDMI.cpp:315:8: error: 'uint16_t' in namespace 'std' does not name a type; did you mean 'wint_t'?
  315 |   std::uint16_t Panel_M5HDMI::HDMI_Trans::readRegister16(std::uint8_t register_address)
      |        ^~~~~~~~
      |        wint_t
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:50:8: error: 'LGFX' does not name a type
   50 | static LGFX lcd;
      |        ^~~~
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp: In function 'void vGuiInit()':
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:63:9: error: 'lcd' was not declared in this scope
   63 |         lcd.init();
      |         ^~~
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp: In function 'void vGuiRefresh()':
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_3.cpp:113:5: error: 'lcd' was not declared in this scope
  113 |     lcd.clear(0x0000);
      |     ^~~
```

So effectively the exact same errors and warnings as master/v5.3

ammaree, 2023-12-18:
@tobozo @lovyan03 

Cannot say for sure but seems all the errors are related to missing include files.
"esp_intr_types.h"
maybe "inttypes.h"

tobozo, 2023-12-18:
I'm not sure about the compatibility matrix, maybe try one of the versions listed in this workflow, they all seem to work

https://github.com/lovyan03/LovyanGFX/actions/runs/7122113476

[image]

ammaree, 2023-12-18:
I did try v5.1 but now already on v5.1.2 not 5.1.1

Are you saying that there is no planned intention to update LVGL to support latest version of v5.1 as well as v5.2.x and latest v5.3?

tobozo, 2023-12-18:
that workflow should probably be updated to test `4.4.6`, `5.0.4` and `5.1.2`, however `5.2.x` are not officially supported

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#support-periods

[image]

> Are you saying that there is no planned intention to update LVGL 

did you mean LovyanGFX? those are two different projects, updating LVGL isn't in LovyanGFX scope

however I can add `4.4.6`, `5.0.4` and `5.1.2` versions to the workflow on the develop branch and start to fix the compilation errors based as per your suggestion

tobozo, 2023-12-18:
apparently no compilation error after updating the workflow with the latest esp-idf versions (`develop` branch), maybe something is weird in your `CMakeLists.txt`? 

https://github.com/lovyan03/LovyanGFX/actions/runs/7251461408/job/19753868753

that's probably unrelated but the [workflow CMakeLists.txt](https://github.com/lovyan03/LovyanGFX/blob/develop/examples/Test/build_test/CMakeLists.txt#L2) looks for a folder named `LovyanGFX` while your local path to LovyanGFX library seems to be named `lovyan`.

ammaree, 2023-12-18:
Apologies, meant LGFX

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#support-periods

image.png (view on web)<https://github.com/lovyan03/LovyanGFX/assets/1893754/29c8dc4f-aac5-4330-b951-78053e436e10>

ammaree, 2023-12-18:
Changing the directory name from "lovyan" to "LovyanGFX" will make no difference. If this was the error none of the source/header files would have been found and we would not have seen the actual compile errors coming from the lovyan component. Have done the rename and proved the theory.

The DAC support has been marked as deprecated since the release of IDF 5.1.0. Although still supported (for a while) it is strongly encouraged to move to the new driver.
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.1/peripherals.html](url)

The "'INTR_CPU_ID_AUTO" related error is fixed in common.cpp with
```cpp
#include "esp_intr_types.h"
buscfg.isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO;
```

The std:uint8_t/uint16_t errors are fixed by adding
```cpp
#include <cstdint>
```

tobozo, 2023-12-18:
this library is also for arduino core and tries to be backwards compatible, so it's not as simple as adding an include, conditional macros may be needed, some dac/i2s functions and properties have been renamed. also thorough testing must follow to prevent the new fixes to break the old versions

meanwhile it is recommended to use one of the following esp-idf **patch** versions along with the `develop` branch of LovyanGFX: `4.4.6`, `5.0.4` or `5.1.2`

ammaree, 2023-12-19:
I 100% agree, code above just the snippets to demonstrate how elementary the fix is.

Unfortunately we are unable to go back to v5.1.x since we have already upgraded our code to support the changes required for v5.2.x

tobozo, 2023-12-19:
I can't think of a conditional macro that will work both for the existing 5.2.x branches (dev and beta1) and the master branch while being past and future proof.

For example this macro doesn't handle the differences between 5.2-dev, 5.2-beta1 and master:

```c
#if defined ( ESP_IDF_VERSION_VAL )
 #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 2, 0) && ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 3, 0)
   #define LGFX_IDF_V5_2 
 #endif
#endif

#if defined LGFX_IDF_V5_2 
  #include "esp_intr_types.h"
  #include <cstdint>
#endif 

```

Ideally the macro should be feature aware instead of version aware, but it doesn't cover possible changes on the master branch:

```c

#if __has_include("esp_intr_types.h")
  #include "esp_intr_types.h"
  #include <cstdint>
  #define LGFX_CPU_AFFINITY_AUTO ESP_INTR_CPU_AFFINITY_AUTO
#elif __has_include("intr_types.h")
  #include "intr_types.h"
  #define LGFX_CPU_AFFINITY_AUTO INTR_CPU_ID_AUTO
#else
  // tech debt here: use version clamping?
  #define LGFX_CPU_AFFINITY_AUTO 0
#endif 

// // also edit common.cpp:
// buscfg.isr_cpu_id = LGFX_CPU_AFFINITY_AUTO;

```

ammaree, 2023-12-20:
Hi @tobozo @lovyan03 

Thanks for the changes, much appreciated.

I can now compile and link for the ESP-WROVER-DEVKIT v4.1 using LGFX_AUTODETECT but cannot get the pre-specified configuration to compile and link. My code is:

```cpp
#define LGFX_AUTODETECT

#ifndef LGFX_AUTODETECT
	#define LGFX_USE_V1
	#define LGFX_ESP_WROVER_KIT
	#define SD_SUPPORTED
#else
	#include <LovyanGFX.hpp>
	#include <LGFX_AUTODETECT.hpp>
#endif
```

Q1: Any suggestion of getting it to work without defining LGFX_AUTODETECT ???

Q2: I also have a MakerFabs ESP32-S3 Parallel16 (480x320) devboard. Is it supported with ESP-IDF v5.2 or later?

Q3: If supported, any suggestion how to get working?

tobozo, 2023-12-20:
> Q1: Any suggestion of getting it to work without defining LGFX_AUTODETECT ???

I assume `ESP-WROVER-DEVKIT v4.1` is [this board](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#get-started-esp-wrover-kit-v4-1-board-front)

you can create your own configuration block instead of using auto detection. see the [settings example](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino)

> Q2: I also have a MakerFabs ESP32-S3 Parallel16 (480x320) devboard. Is it supported with ESP-IDF v5.2 or later?

there are a few Makerfabs config examples in the [lgfx_user](https://github.com/lovyan03/LovyanGFX/tree/master/src/lgfx_user) folder

> Q3: If supported, any suggestion how to get working?

at the moment, ESP-IDF v5.2 is not an officialy supported espressif release, having LovyanGFX working with it in certain conditions is probably accidental :-)

rule of thum with LovyanGFX support: esp-idf master branch and all branches suffixed by `-rc`, `-beta*`, `-dev` aren't stable enough to get any reliable support because things may change, then be reverted, then change again; and no one wants to pay this tech debt on a day to day basis, which is probably the reason why they're outside the service and maintenance period on espressif support graph

esp-idf 5.2.x will be added as soon as espressif associates a tag to the release, it just hasn't happened yet

```log
$ git ls-remote https://github.com/espressif/esp-idf | grep '5\.2'
e49823f10c6b6b8dadbad376002ea2bc531b7b0f        refs/heads/release/v5.2
9690cd19a2f03de5773293c28d701508de061e4e        refs/tags/v5.2-beta1
57bbfd423a1e9533d371a5e0e1e6bcec25bae6fc        refs/tags/v5.2-beta1^{}
a62f09c743a7299cf83f3c524bb6590fad802eea        refs/tags/v5.2-dev
f8d2c0894cf3c12dc231dafebf75d3f1002b312c        refs/tags/v5.2-dev^{}
```

as a reminder, LovyanGFX develop branch has been recently tested with the following combos of devices/esp-idf versions:

[image]

closed 2024-01-26 by github-actions[bot]

tobozo, 2024-02-17:
bump

[esp-idf 5.2 is now officially the latest release](https://github.com/espressif/esp-idf/releases/tag/v5.2).

Some macros needed to be adjusted in LovyanGFX source tree to fix the compilation errors, the changes are available on the [`develop`](https://github.com/lovyan03/LovyanGFX/tree/develop) branch

[image]