lovyan03/LovyanGFX #491: esp32-s3 - Calling LovyanGFX breaks the SPI bus for other devices

LouDou, 2023-12-16:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:   esp32-s3-devkitc-1
- Panel Driver IC:    Panel_ST7735S
- Bus type:           SPI
- LovyanGFX version:  1.1.9
- FrameWork version:   ESP-IDF v5.1.1
- Build Environment:    PlatformIO
- Operating System:   Linux

## Problem Description ( 問題の内容 )

I am sharing the SPI bus between one ST7735s and one AD5238 DAC. I have both the display and DAC working independently. However, as soon as I send anything to the display, the DAC stops working. The DAC is configured using the esp-idf `spi_bus_add_device` API and also sets up the bus using `spi_bus_initialize`.

### Expected Behavior ( 期待される動作 )

Both the DAC and display can be used at the same time on the same SPI bus.

### Actual Behavior ( 実際の動作 )

DAC stops working after the display has been updated.

### Steps to reproduce ( 再現のための前提条件 )

 - Using DAC on its own works perfectly
 - Using the display on its own works perfectly
 - Usng the DAC and then the display, both work
 - Using the display and then the DAC - the display is updated, but the DAC stops responding to any further commands

### Code to reproduce this issue ( 再現させるためのコード )

DAC interface:
```cpp

namespace psc::dac
{
    // configure CS GPIOs - control; output only
    constexpr int cs_gpios[] = {
        DAC_1_CS,
        DAC_2_CS,
    };

    spi_device_handle_t dacs[2];

    spi_transaction_t t{
        .flags = SPI_TRANS_USE_TXDATA,
        .length = 2 * 8,
    };

    inline void writeDac(const size_t idx, const uint16_t data)
    {
        t.tx_data[0] = data >> 8;
        t.tx_data[1] = data & 0xFF;
        /* auto ret = */ spi_device_polling_transmit(dacs[idx], &t);
        // /* auto ret = */ spi_device_transmit(dacs[idx], &t);
        // /* auto ret = */ spi_device_queue_trans(dacs[idx], &t, portMAX_DELAY);
        // ESP_LOGI(TAG, "write %d 0x%02x : 0x%02x : 0x%02x", cs, t.tx_data[0], t.tx_data[1], ret);
    }

    inline void writeDacOutput(const size_t idx, const uint8_t ch, const uint16_t val)
    {
        // Page 16 "DAC Write" / Figure 34: Shift register contents
        // [RW, A2, A1, A0, d12...d0]
        // RW; 0=Write DAC value
        writeDac(idx, (ch << 12) | (val & 0xFFF));
    }

    void setup()
    {
        esp_err_t ret;

        // -----

        // Initialize the SPI bus
        spi_bus_config_t buscfg = {
            .mosi_io_num = SPI_MOSI,
            .miso_io_num = SPI_MISO,
            .sclk_io_num = SPI_SCLK,
        };
        ret = spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO);
        ESP_ERROR_CHECK(ret);

        // -----

        gpio_config_t io_cs_conf = {
            .pin_bit_mask = 0,
            .mode = GPIO_MODE_OUTPUT,
            .pull_up_en = GPIO_PULLUP_DISABLE,
            .pull_down_en = GPIO_PULLDOWN_ENABLE,
            .intr_type = GPIO_INTR_DISABLE,
        };
        for (int i = 0; i < sizeof(cs_gpios) / sizeof(cs_gpios[0]); i++)
        {
            io_cs_conf.pin_bit_mask |= 1ULL << cs_gpios[i];
        }
        gpio_config(&io_cs_conf);

        // -----

        // Attach the DACs to the SPI bus
        spi_device_interface_config_t devcfg0 = {
            .mode = 2,
            .clock_speed_hz = SPI_MASTER_FREQ_8M,
            .spics_io_num = DAC_1_CS,
            .queue_size = 1,
        };
        ret = spi_bus_add_device(SPI2_HOST, &devcfg0, &dacs[0]);
        ESP_ERROR_CHECK(ret);

        spi_device_interface_config_t devcfg1 = {
            .mode = 2,
            .clock_speed_hz = SPI_MASTER_FREQ_8M,
            .spics_io_num = DAC_2_CS,
            .queue_size = 1,
        };
        ret = spi_bus_add_device(SPI2_HOST, &devcfg1, &dacs[1]);
        ESP_ERROR_CHECK(ret);

        // -----

        // initialise CS
        for (auto pin : cs_gpios)
        {
            gpio_set_level((gpio_num_t)pin, 1); // CS /SYNC is inverted
        }
        // configure DACs
        for (size_t idx : {0, 1})
        {
            // Page 18: Gain, BUF and Vdd config
            const uint16_t cfg = (1 << 15) | 0b00'00'11;
            writeDac(idx, cfg);

            // Table 8: /LDAC
            // /LDAC low mode; continuous update using SYNC;
            // /LDAC must be tied high
            const uint16_t lcadmode = (1 << 15) | (1 << 13) | 0x00;
            writeDac(idx, lcadmode);
        }
    }
}
```

SPI and display interface
```cpp

namespace psc::io
{
    class PSCSPI : public lgfx::Bus_SPI
    {
    public:
        PSCSPI()
            : lgfx::Bus_SPI()
        {
            auto cfg = config();

            cfg.spi_host = SPI2_HOST;
            cfg.spi_mode = 0;

            cfg.freq_write = 40'000'000; // for TFT
            cfg.freq_read = 2'500'000;
            cfg.spi_3wire = false;
            cfg.use_lock = true;
            cfg.dma_channel = SPI_DMA_CH_AUTO;

            cfg.pin_sclk = SPI_SCLK;
            cfg.pin_mosi = SPI_MOSI;
            cfg.pin_miso = SPI_MISO;
            cfg.pin_dc = TFT_DC;

            config(cfg);
            init();
        }
    };

    using SPI_sptr = std::shared_ptr<PSCSPI>;
}

//------

namespace psc::tft
{
    class LGFX : public lgfx::LGFX_Device
    {
        psc::io::SPI_sptr _bus_instance;

        lgfx::Panel_ST7735S _panel_instance;
        lgfx::Light_PWM _light_instance;

    public:
        LGFX(psc::io::SPI_sptr bus)
            : _bus_instance(bus)
        {
            {
                _panel_instance.setBus(_bus_instance.get());
            }

            {
                auto cfg = _panel_instance.config();

                cfg.pin_cs = TFT_CS;
                cfg.pin_rst = TFT_RST;
                cfg.pin_busy = TFT_BUSY;

                cfg.panel_width = 128;
                cfg.panel_height = 160;
                cfg.offset_x = 4;
                cfg.offset_y = 0;
                cfg.offset_rotation = 1;
                cfg.dummy_read_pixel = 8;
                cfg.dummy_read_bits = 1;
                cfg.readable = false;
                cfg.invert = false;
                cfg.rgb_order = true;
                cfg.dlen_16bit = false;
                cfg.bus_shared = true;

                _panel_instance.config(cfg);
            }

#if TFT_BL > 0
            {
                auto cfg = _light_instance.config();

                cfg.pin_bl = TFT_BL;
                cfg.invert = false;
                cfg.freq = 44100;
                cfg.pwm_channel = 7;

                _light_instance.config(cfg);
                _panel_instance.setLight(&_light_instance);
            }
#endif

            setPanel(&_panel_instance);
        }
    };

//------

    std::shared_ptr<LGFX> display;

    void setup(psc::io::SPI_sptr bus)
    {
        display = std::make_shared<LGFX>(bus);

        display->init();
#if TFT_BL > 0
        display->light()->setBrightness(128);
#endif
        display->setTextSize(0);
        display->setColorDepth(16);
        display->fillScreen(TFT_BLACK);
    }

    void __preDisplay(const bool clear)
    {
        display->startWrite(true);
        if (clear)
        {
            display->clearDisplay();
        }
        display->setTextDatum(BL_DATUM);
    }

    void __postDisplay()
    {
        display->endWrite();
    }

    void displayConfig(psc::config::Config &cfg)
    {
        __preDisplay(false);

        // More display-> calls in here
    
        __postDisplay(false);
    }
}
```

main
```cpp

    psc::tft::setup(psc::io::SPI());
    psc::dac::setup();

```

DAC is being updated periodically in a timer task using `psc::dac::writeDacOutput(...)`.

Then as soon as `psc::tft::displayConfig(...)` is called, the DAC no longer updates.

LouDou, 2024-01-16:
This issue should not be closed yet please.

lovyan03, 2024-01-16:
spi_bus_initialize はGFXが実行しているのでDACの方には不要ではないでしょうか

LouDou, 2024-01-17:
@lovyan03 if I let this library call `spi_bus_initialize` then the DAC doesn't work. I need to have better control over initialising and using the bus myself for both the DAC and the display.

LouDou, 2024-01-29:
I still haven't got to the bottom of what is going on here. There is a lot of complex code in lgfx's `src/lgfx/v1/platforms/esp32/common.cpp` interface, and it assumes in `_spi_dev_handle` that it owns the SPI bus and that only one device is ever on that bus. I therefore cannot let lgfx manage the bus and the transactions because it is unaware that other devices are on the bus.

This is a shame, I really like the lgfx library for graphics, I just wish it could play nicely with other devices.

LouDou, 2024-02-28:
not stale

lovyan03, 2024-03-01:
検証しやすいように、ビルド可能な もっとシンプルなコードを提供してもらえませんか？

sukesh-ak, 2024-03-27:
You can see the difference in code between Shared SPI bus and otherwise here in `ESP-IDF v5.x`

Exclusive => https://github.com/sukesh-ak/ESP32-TUX/blob/master/main/helpers/helper_storage.hpp

Shared => https://github.com/sukesh-ak/ESP32-TUX/blob/master/main/helpers/helper_storage_shared.hpp

closed 2024-05-03 by github-actions[bot]

mhaberler, 2024-05-03:
not stale.

ainyan03, 2026-08-27:
Found the mechanism, and it is fixed since **1.2.0**.

**What went wrong in 1.1.x:** `Bus_SPI::beginTransaction()` wrote `SPI_PIN_REG` (`SPI_MISC_REG` on ESP32-S3) with only the clock-idle bit and left every hardware-CS *disable* bit at 0. While the panel was being driven, the hardware CS lines that the IDF driver had assigned to your DACs (CS0/CS1, because they were added first) were asserted as well, so the display's command/pixel stream went into both DACs. Whatever 16-bit word they latched last left them in a state where they no longer respond — which matches "DAC then display works, display then DAC doesn't". The fix is afffd0db ("Fixed an issue where the SPI peripheral would conflict with devices that controlled the CS pin"), first released in 1.2.0; current code sets all `CSn_DIS` bits before a panel transaction.

**How it was verified** (ESP32-S3, ESP-IDF 5.1.1 as in your report; the application initialises the bus and adds a mode-2 device with hardware CS first, then LovyanGFX joins the bus). No DAC is needed: the device's hardware-CS pad is read back through the GPIO input path and its edges are counted while the panel is being drawn.

| LovyanGFX | CS edges of the IDF device during panel drawing (550 loops) | during its own transfers |
|---|---|---|
| 1.1.9 | **50600** | 1100 (2 per transfer, as expected) |
| 1.1.16 | **50600** | 1100 |
| 1.2.0 | 0 | 1100 |
| 1.2.28 (IDF 5.1.1 and 5.5.4) | 0 | 1100 |

The exact code used, so we are looking at the same thing:

<details><summary>issue491_cs_edge_test.cpp</summary>

```cpp
// Minimal test for LovyanGFX issue #491.
// Counts edges on the hardware CS line of an ESP-IDF spi_master device (mode 2)
// while LovyanGFX drives a panel on the same SPI2 bus. With a correct driver the
// count taken during panel drawing must stay 0.
//
// Build: ESP-IDF (tested 5.1.1 and 5.5.4), ESP32-S3. No slave device needed:
// the CS pad is read back through the GPIO input path.

#include <LovyanGFX.hpp>
#include <driver/spi_master.h>
#include <driver/gpio.h>
#include <hal/gpio_ll.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>

// ---- pins (edit for your board) ----
#define PIN_SCLK 36
#define PIN_MOSI 37
#define PIN_MISO 35
#define PIN_LCD_CS 3
#define PIN_LCD_DC 35   // shared with MISO on M5Stack CoreS3
#define PIN_DAC_CS 8    // hardware CS of the IDF spi_master device (any free GPIO)
#define TEST_BOARD_CORES3 1

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Bus_SPI _bus;
  lgfx::Panel_ILI9342 _panel;   // any SPI panel works for this measurement
public:
  LGFX(void)
  {
    auto b = _bus.config();
    b.spi_host = SPI2_HOST; b.spi_mode = 0; b.freq_write = 40000000; b.freq_read = 16000000;
    b.use_lock = true; b.dma_channel = SPI_DMA_CH_AUTO;
    b.pin_sclk = PIN_SCLK; b.pin_mosi = PIN_MOSI; b.pin_miso = PIN_MISO; b.pin_dc = PIN_LCD_DC;
    _bus.config(b);
    _panel.setBus(&_bus);
    auto p = _panel.config();
    p.pin_cs = PIN_LCD_CS; p.pin_rst = -1; p.invert = true; p.offset_rotation = 3;
    _panel.config(p);
    setPanel(&_panel);
  }
};

static LGFX lcd;
static spi_device_handle_t dac = nullptr;
static volatile uint32_t cs_edges = 0;

static void IRAM_ATTR cs_isr(void*) { ++cs_edges; }

[... 64 lines cut ...]
    lcd.pushImage(20, 100, 32, 32, img);
    during_lcd += cs_edges - e0;          // must stay 0

    e0 = cs_edges;
    dac_write((uint16_t)n);
    during_dac += cs_edges - e0;          // 2 per transfer (assert + release) is the expected value

    if ((n % 50) == 0) {
      printf("n=%lu  DAC-CS edges during panel drawing=%lu  during DAC transfers=%lu\n",
             (unsigned long)n, (unsigned long)during_lcd, (unsigned long)during_dac);
    }
    vTaskDelay(pdMS_TO_TICKS(20));
  }
}

#ifndef ISSUE491_NO_APP_MAIN
extern "C" void app_main(void) { issue491_test(); }
#endif
```

</details>

Notes on the test: pins and the `board_power_on()` block are for an M5Stack CoreS3 (ILI9342 on SPI2); set `TEST_BOARD_CORES3 0` and adjust the pins/panel for a devkit. Building 1.1.9 on IDF 5.1.1 needed two local touch-ups unrelated to this issue (`comd` field name in `common.cpp`, `<cstdint>` in `Panel_M5HDMI.hpp`).

Please update to 1.2.0 or later (current release is 1.2.28) and let us know if the DACs still misbehave with that.