lovyan03/LovyanGFX #501: Flag to disable SDL

jp-bennett, 2024-01-01:
This is related to my previous work, #484 . I've discovered that when building our binary, if the SDL-devel packages are found, this library tries to include SDL support as well as arduino_default. It would be great to be able to define a build flag like LG_DISABLE_SDL so none of that code fires off.

lovyan03, 2024-01-01:
あなたの環境を知らないので何が問題なのか私にはわからない。
パッケージを含めなければ良いだけではないのですか？

jp-bennett, 2024-01-01:
@lovyan03 It looks like this library is actively looking for the SDL package. The only way I could get working compilation is to uninstall the SDL2-devel package. Admittedly our environment is a bit odd.

I'll try to produce a working patch in the coming days to demonstrate what I mean.

lovyan03, 2024-01-01:
もう一度言いますが、私はあなたの環境がわからない
OSは ?  Windows ? macOS ?
開発環境は ? ArduinoIDE ? VSCode ?

基本的にマイコン用にビルドすることを想定しているので、
VSCode+PlatformIOやArduinoIDEなら何も問題は起きないはずです。
SDLがあることでなぜ困るのか状況がわかりません。

jp-bennett, 2024-01-01:
Compiling on Linux, PlatformIO+VScode, using the Portduino framework to take PlatformIO code and produce a native binary.

Simple to replicate. Make sure the libsdl2-dev package is installed. (and libyaml-cpp-dev libgpiod-dev libbluetooth-dev)
Grab the native-unity branch at https://github.com/meshtastic/firmware/tree/native-unity. In platformIO, select the native target, and build.

lovyan03, 2024-01-01:
あぁ、あなたは SDLではなくPortduino と組み合わせて使いたいのですね？
つまり Nativeビルドを使用するが、 SDLは使わずに Portduinoで動かしたいんですね？

私はその情報をいま初めて聞いた。それが最も重要な前提情報です。

jp-bennett, 2024-01-01:
Sorry to leave out that important detail. Yes, we are doing native builds, and talking to real SPI and I2C hardware. It wouldn't be a problem to also have SDL support, but that currently breaks compilation.

lovyan03, 2024-01-01:
I just found out about Portduino and thought it would be useful.

In that case, if you use the Native build, it would be convenient if you could run LovyanGFX on both Portduino and SDL.
In my opinion, it is necessary to be able to use both at the same time depending on how the user code is written, rather than relying on compile switches.

For now, I'll look into Portduino.

lovyan03, 2024-01-01:
Portduinoを動かす方法が良くわからない。
前提条件がいろいろあるように思うが環境を整えるのが非常に面倒に思います

jp-bennett, 2024-01-01:
It's a bit of a pain to use. https://github.com/meshtastic/platform-native is the intended way to do it. Though the easiest starting point for testing is Meshtastic.

lovyan03, 2024-01-02:
私はWindowsとmacOSを使っていますが、そもそもPortduinoがLinuxにしか対応していないように見えます。

lovyan03, 2024-01-02:
I updated the develop branch. This will exclude SDL from being built if the define constant ARDUINO exists.

closed 2024-01-17 by jp-bennett